esphome/issues #1486: ble_presence using Android App, almost there

rdehuyss, 2020-09-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, esp32-poe

**ESPHome version (latest production, beta, dev branch)**

1.15.1

**Affected component:**

ble_presence

**Description of problem:**
I have the beacon simulator app running on my android (very stable, almost not consuming battery). I use it to emit IBeacon BLE events with a fixed UUID. I see esphome receiving a packet with the UUID in it, but the presence detection does not trigger.

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:
  - platform: ble_presence
    service_uuid: '50B04BBB-7B26-06AC-7441-92140A6A9E2b'
    name: "Ronald BLE GSM"
```

**Logs (if applicable):**

```
[23:00:34][VV][esp32_ble_tracker:326]: Parse Result:
[23:00:34][VV][esp32_ble_tracker:343]:   Address: 46:28:A0:63:3A:D0 (RANDOM)
[23:00:34][VV][esp32_ble_tracker:345]:   RSSI: -86
[23:00:34][VV][esp32_ble_tracker:346]:   Name: ''
[23:00:34][VV][esp32_ble_tracker:360]:   Manufacturer data: BE.AC.2B.9E.6A.0A.14.92.41.74.AC.06.26.7B.BB.4B.B0.50.00.00.00.00.BB.00 (24)
[23:00:34][VV][esp32_ble_tracker:368]: Adv data: 1B.FF.FF.FF.BE.AC.2B.9E.6A.0A.14.92.41.74.AC.06.26.7B.BB.4B.B0.50.00.00.00.00.BB.00 (28)

```

**Additional information and things you've tried:**
I tried to look into this but have very little c++ skills and no knowledge about the BLE stack. What I did find out is:
- the 1th byte (1B) is the length of the packet
- the 2nd byte (FF) is the type of packet, in this case ESP_BLE_AD_MANUFACTURER_SPECIFIC_TYPE
- the 3th until the 6th byte (FF.FF.BE.AC) - don't really know
- the 7th until the 22nd byte (2B.9E.6A.0A.14.92.41.74.AC.06.26.7B.BB.4B.B0.50) is the **UUID in reverse**!

**So, the data is all there, ESPHome just doesn't recognize it yet.** My guess is that it is because the app is emitting using a manufacturer specific adv_type packet and not the expected ESP_BLE_AD_TYPE_128SRV_* type.

If this would work, we would have super simple and effectively working presence detection.

Including @OttoWinter, @dermotduffy, @lumpusz, @zinefer and @glmnet - all the people who commented on https://github.com/esphome/esphome/pull/937

Lumpusz, 2020-09-18:
Hi,

I did a quick test with the setup I originally used for the development. It seems to work just fine for me, although I use a different app on my phone to advertise the signal, and the service_uuid. Here is my setup:

ESPHome config:

```
esphome:
  name: ble_test
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

esp32_ble_tracker:

sensor:
  - platform: ble_rssi
    service_uuid: '11AA11AA'
    name: "Android phone"

logger:

api:

ota:
```

nrfConnect scrrenshots:

[image]
[image]

ESPHome logs:
```
[08:10:44][C][esp32_ble_tracker:530]: BLE Tracker:
[08:10:44][C][esp32_ble_tracker:531]:   Scan Duration: 300 s
[08:10:44][C][esp32_ble_tracker:532]:   Scan Interval: 320.0 ms
[08:10:44][C][esp32_ble_tracker:533]:   Scan Window: 30.0 ms
[08:10:44][C][esp32_ble_tracker:534]:   Scan Type: ACTIVE
[08:10:44][C][ble_rssi:011]: BLE RSSI Sensor 'Android phone'
[08:10:44][C][ble_rssi:011]:   Unit of Measurement: 'dB'
[08:10:44][C][ble_rssi:011]:   Accuracy Decimals: 0
[08:10:44][C][ble_rssi:011]:   Icon: 'mdi:signal-distance-variant'
[08:10:44][C][ota:029]: Over-The-Air Updates:
[08:10:44][C][ota:030]:   Address: ble_test.local:3232
[08:10:44][C][api:095]: API Server:
[08:10:44][C][api:096]:   Address: ble_test.local:6053
[08:11:00][D][sensor:092]: 'Android phone': Sending state -55.00000 dB with 0 decimals of accuracy
[08:11:03][D][sensor:092]: 'Android phone': Sending state -55.00000 dB with 0 decimals of accuracy
[08:11:08][D][sensor:092]: 'Android phone': Sending state -61.00000 dB with 0 decimals of accuracy
[08:11:11][D][sensor:092]: 'Android phone': Sending state -61.00000 dB with 0 decimals of accuracy
[08:11:14][D][sensor:092]: 'Android phone': Sending state -61.00000 dB with 0 decimals of accuracy
[08:11:16][D][sensor:092]: 'Android phone': Sending state -56.00000 dB with 0 decimals of accuracy

```

rdehuyss, 2020-09-18:
Thanks for the tip of this app, I had not seen it before.

The beacon simulator app I referenced cannot create UUID with less than 128bit. I'll try this app tonight!

Thanks for the feedback,
Ronald

Lumpusz, 2020-09-18:
> Thanks for the tip of this app, I had not seen it before.
> 
> The beacon simulator app I referenced cannot create UUID with less than 128bit. I'll try this app tonight!
> 
> Thanks for the feedback,
> Ronald

The code should be able to work with 16, 32, and 128 bit UUIDs. Back in the day I tested all 3 of the using nrfConnect.

rdehuyss, 2020-09-18:
> The code should be able to work with 16, 32, and 128 bit UUIDs. Back in the day I tested all 3 of the using nrfConnect.

I think it is more related to the app sending an ESP_BLE_AD_MANUFACTURER_SPECIFIC_TYPE BLE event. I will test this evening for sure. Thanks for the feedback!

ashmckenzie, 2020-10-19:
I can confirm using nRF Connect with a 32 bit UUID works with ESPhome :+1:  Thanks @Lumpusz 🙇

closed 2021-03-02 by stale[bot]

fokcuk, 2021-06-28:
any updates on this? nRF connect generated beacons of any length are detected by ESPHome no problem.
But beacon simulator app and native HA companion app beacons are not - like they are different kind of beacons