esphome/issues #1487: MQTT Config Validation Fails When MQTT Component Defined in a Package

DeanCording, 2020-09-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Ubuntu/Mosquitto

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**ESPHome version (latest production, beta, dev branch)**

1.15.1
**Affected component:**

https://esphome.io/guides/configuration-types.html#packages

**Description of problem:**
When config for MQTT client is placed in a package the config validation fails.

**Problem-relevant YAML-configuration entries:**
This config works:
```yaml
esphome:
  name: test
  platform: ESP8266
  board: esp12e
  build_path: ./build/test

wifi:
  ssid: "Home"
  password: "xxxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"

captive_portal:

debug:

# Enable logging
logger:
  level: debug
  
ota:

mqtt:
  broker: xxxxxxxxx
  username: esphome
  password: esphomepass
  discovery: false

time:
   - platform: sntp
     servers: xxxxxxxx
     id: sntp_time

status_led:
  pin:
    number: GPIO2
    inverted: true

text_sensor:
  - platform: version
    name: "ESPHome Version"
  - platform: wifi_info
    ip_address:
      name: ESP IP Address
    ssid:
      name: ESP Connected SSID
    bssid:
      name: ESP Connected BSSID
    mac_address:
      name: ESP Mac Wifi Address
      
sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 10s
    expire_after: 2min

```

Placing this configuration in a package causes the config validation to fail.

base.yaml
```yaml
esphome:
  name: ${device_name}
  platform: ESP8266
  board: esp12e
  build_path: ./build/${device_name}

wifi:
  ssid: "Home"
  password: "coconuts"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}

captive_portal:

debug:

# Enable logging
logger:
  level: debug

ota:

mqtt:
  broker: iot64.cording.id.au
  username: esphome
  password: esphomepass
  discovery: false
  
time:
   - platform: sntp
     servers: ntp.cording.id.au
     id: sntp_time

status_led:
  pin:
    number: GPIO2
    inverted: true
    
text_sensor:
  - platform: version
    name: "ESPHome Version"
  - platform: wifi_info
    ip_address:
      name: ESP IP Address
    ssid:
      name: ESP Connected SSID
    bssid:
      name: ESP Connected BSSID
    mac_address:
      name: ESP Mac Wifi Address

sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 10s
    expire_after: 2min
```
test.yaml
```yaml
substitutions:
  device_name: test

packages:
  config: !include config/base.yaml
```

**Logs (if applicable):**

```
$ esphome test.yaml run
INFO Reading configuration test.yaml...
INFO Detected timezone 'AEST' with UTC offset 10
Failed config

sensor.wifi_signal: [source config/base.yaml:55]
  platform: wifi_signal
  name: WiFi Signal Sensor
  update_interval: 60s
  
  This option requires component mqtt.
  expire_after: 2min [source config/base.yaml:58]
```

**Additional information and things you've tried:**

DeanCording, 2020-09-18:
If I remove the expire_after setting, the config validates and compiles, but the sensor states are not published on MQTT.

akomelj, 2020-09-30:
I've encountered a similar issue with version 1.15.2 (I haven't tried the 1.15.* versions before).

Using `mqtt` in `packages` does not configure MQTT components correctly (`mqtt_sensor`, `mqtt_text_sensor`) so no sensor data is published to MQTT. Manual publishing via automations works though.

This configuration does not publish data to MQTT:
```
# in common configuration yaml; included from device yaml
packages:
  mqtt: !include common-mqtt.inc.yaml
```

This configuration directly in device yaml does publish data:
```
# in device yaml
<<: !include common/common-mqtt.inc.yaml
```

I can see that in the first case (w/ packages) MQTT components are not configured for sensors as no MQTT component configuration data is dumped for the sensors during the boot process.

Overriding a single key from packages mqtt configuration in device yaml does not help either.

I've also compared generated configurations (`esphome config`) and they are identical up to the order of the keys. With `packages` configuration, `mqtt` key is listed somewhere in the middle of the configuration yaml; when including `mqtt` configuration directly in device yaml, it si configured at the very bottom of the configuration file.

akomelj, 2020-09-30:
To expand a bit further: the problem appears to be related to `OnlyWith` implementation in `config_validation.py`.

MQTT wrapper components for sensors/switches/etc appear to be generated only when `OnlyWith()` implementation finds key `mqtt` in `CORE.raw_config`. I assume that `CORE.raw_config` is the raw configuration file where (default) `mqtt` configuration may reside in `packages` only; consequently `OnlyWith()` does not find it.

Will see if I can elaborate this any further or prepare a pull request with a fix. I'm not really a Python guy so this will be more like shooting blind. :-P

akomelj, 2020-09-30:
Ok, the following contraption appears to fix the problem with `OnlyWith` component check. Does this look OK for a pull request or does any of the maintainers want to implement it differently?

Basically, the original check that verified if the component key existed in the raw configuration is replaced with a check that verifies whether the component key exists in the raw configuration or - if `packages` key exists in the raw configuration - any of the packages contains component configuration.

```python
    @property
    def default(self):
        # pylint: disable=unsupported-membership-test
        if (self._component in CORE.raw_config or
                (CONF_PACKAGES in CORE.raw_config and
                    self._component in
                        {list(x.keys())[0] for x in CORE.raw_config[CONF_PACKAGES].values()})):
            return self._default
        return vol.UNDEFINED
```

DeanCording, 2020-10-06:
Confirm that issue is fixed in ESPHome version 1.15.2

DeanCording, 2020-10-08:
Mistakenly closed issue before PR has been merged.

R4isen, 2020-11-11:
Hi, I have included follow-up fix related to mqtt options validations when mqtt is defined in the packages.

DeanCording, 2021-03-12:
Waiting for release containing patch before closing issue.

bbrezuica, 2021-04-01:
Hello,

any idea on when will the fix for this issue be released? This really starts to get on my nerves. I've had sensors down or stuck and I was still seeing values in HA and had no clue they were old/wrong.

Thank you.

j-k-olofsson, 2021-05-25:
I'm on version 1.18.0 and the problem still exists. Any idea on which release will inculde the patch?

R4isen, 2021-05-25:
Well I fetched latest release from repo and cherry picked those changes on top. I'm not waiting for final resolution 😉

foormea, 2021-05-29:
I confirm, 1.18.0, having my `mqtt` config in my package results in `mqtt` messages not going through (apart from log and status, but no sensor data). Moving the `mqtt` config into my "main" `yaml` solves the problem.

paulmonigatti, 2021-09-17:
An interim fix was implemented in esphome/esphome#2210, and a final fix in esphome/esphome#2193. Should all be working now!

DeanCording, 2022-01-02:
Finally had a chance to check out the fix.  All good, issue resolved.

closed 2022-01-02 by DeanCording