esphome/issues #1488: TuyaMCU: not found

spiro-c, 2020-09-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**
I was trying  TuyaMCU with my wifi battery door sensor and i cant get it working if i use configuration from 
https://esphome-configs.io/devices/mirabella-door-window-sensor/ i have no problem it connect to mqtt and send state

**Problem-relevant YAML-configuration entries:**
```yaml
# Basic Config
# https://mirabellagenio.net.au/door-%26-window-sensor
esphome:
  name: mirabella_genio_door_window_sensor
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
#  manual_ip:
#    static_ip: 192.168.10.230
#    gateway: 192.168.10.1
#    subnet: 255.255.255.0
  fast_connect: true

ota:

logger:
  level: debug
  baud_rate: 0

# Enable Home Assistant API
api:

# Make sure you can upload new firmware OTA

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

# Register the Tuya MCU connection
tuya:

```

**Logs (if applicable):**

```
INFO Connecting to mirabella_genio_door_window_sensor.local:6053 (192.168.10.43)
INFO Successfully connected to mirabella_genio_door_window_sensor.local
[14:42:40][I][app:105]: ESPHome version 1.16.0-dev compiled on Sep 16 2020, 16:30:29
[14:42:40][C][wifi:443]: WiFi:
[14:42:40][C][wifi:303]:   SSID: [redacted]
[14:42:40][C][wifi:304]:   IP Address: 192.168.10.43
[14:42:40][C][wifi:306]:   BSSID: [redacted]
[14:42:40][C][wifi:307]:   Hostname: 'mirabella_genio_door_window_sensor'
[14:42:40][C][wifi:311]:   Signal strength: -53 dB ▂▄▆█
[14:42:40][C][wifi:315]:   Channel: 11
[14:42:40][C][wifi:316]:   Subnet: 255.255.255.0
[14:42:40][C][wifi:317]:   Gateway: 192.168.10.1
[14:42:40][C][wifi:318]:   DNS1: 192.168.10.5
[14:42:40][C][wifi:319]:   DNS2: (IP unset)
[14:42:40][C][uart_esp8266:075]: UART Bus:
[14:42:40][C][uart_esp8266:077]:   TX Pin: GPIO1
[14:42:40][C][uart_esp8266:080]:   RX Pin: GPIO3
[14:42:40][C][uart_esp8266:081]:   RX Buffer Size: 256
[14:42:40][C][uart_esp8266:083]:   Baud Rate: 9600 baud
[14:42:40][C][uart_esp8266:084]:   Data Bits: 8
[14:42:40][C][uart_esp8266:085]:   Parity: NONE
[14:42:40][C][uart_esp8266:086]:   Stop bits: 1
[14:42:40][C][uart_esp8266:088]:   Using hardware serial interface.
[14:42:40][C][logger:185]: Logger:
[14:42:40][C][logger:186]:   Level: DEBUG
[14:42:40][C][logger:187]:   Log Baud Rate: 0
[14:42:40][C][logger:188]:   Hardware UART: UART0
[14:42:40][C][ota:029]: Over-The-Air Updates:
[14:42:40][C][ota:030]:   Address: mirabella_genio_door_window_sensor.local:8266
[14:42:40][C][api:095]: API Server:
[14:42:40][C][api:096]:   Address: mirabella_genio_door_window_sensor.local:6053
[14:42:40][C][tuya:023]: Tuya:
[14:42:40][C][tuya:026]:   Configuration will be reported when setup is complete. Current init_state: 0
[14:42:40][C][tuya:027]:   If no further output is received, confirm that this is a supported Tuya device.

```

**Additional information and things you've tried:**

glmnet, 2020-09-18:
Cc @jesserockz

jesserockz, 2020-09-18:
Is this the type of device that boots when the door is opened?
I have a feeling this is going to be hard to debug because the data will be sent before the logger can connect via wifi.
You could try `level: VERY_VERBOSE`  though and paste the logs from that?

spiro-c, 2020-09-19:
This is the log until i remove battery 
```
INFO Connecting to mirabella_genio_door_window_sensor.local:6053 (192.168.10.43)
INFO Successfully connected to mirabella_genio_door_window_sensor.local
[18:54:47][I][app:105]: ESPHome version 1.16.0-dev compiled on Sep 19 2020, 18:48:26
[18:54:47][C][wifi:443]: WiFi:
[18:54:47][C][wifi:303]:   SSID: [redacted]
[18:54:47][C][wifi:304]:   IP Address: 192.168.10.43
[18:54:47][C][wifi:306]:   BSSID: [redacted]
[18:54:47][C][wifi:307]:   Hostname: 'mirabella_genio_door_window_sensor'
[18:54:47][C][wifi:311]:   Signal strength: -65 dB ▂▄▆█
[18:54:47][V][wifi:313]:   Priority: 0.0
[18:54:47][C][wifi:315]:   Channel: 11
[18:54:47][C][wifi:316]:   Subnet: 255.255.255.0
[18:54:47][C][wifi:317]:   Gateway: 192.168.10.1
[18:54:47][C][wifi:318]:   DNS1: 192.168.10.5
[18:54:47][C][wifi:319]:   DNS2: (IP unset)
[18:54:47][C][uart_esp8266:075]: UART Bus:
[18:54:47][C][uart_esp8266:077]:   TX Pin: GPIO1
[18:54:47][C][uart_esp8266:080]:   RX Pin: GPIO3
[18:54:47][C][uart_esp8266:081]:   RX Buffer Size: 256
[18:54:47][C][uart_esp8266:083]:   Baud Rate: 9600 baud
[18:54:47][C][uart_esp8266:084]:   Data Bits: 8
[18:54:47][C][uart_esp8266:085]:   Parity: NONE
[18:54:47][C][uart_esp8266:086]:   Stop bits: 1
[18:54:47][C][uart_esp8266:088]:   Using hardware serial interface.
[18:54:47][C][logger:185]: Logger:
[18:54:47][C][logger:186]:   Level: VERY_VERBOSE
[18:54:47][C][logger:187]:   Log Baud Rate: 0
[18:54:47][C][logger:188]:   Hardware UART: UART0
[18:54:47][C][ota:029]: Over-The-Air Updates:
[18:54:47][C][ota:030]:   Address: mirabella_genio_door_window_sensor.local:8266
[18:54:47][C][api:095]: API Server:
[18:54:47][C][api:096]:   Address: mirabella_genio_door_window_sensor.local:6053
[18:54:47][C][tuya:023]: Tuya:
[18:54:47][C][tuya:026]:   Configuration will be reported when setup is complete. Current init_state: 0
[18:54:47][C][tuya:027]:   If no further output is received, confirm that this is a supported Tuya device.
[18:54:47][VV][api.service:192]: on_hello_request: HelloRequest {
  client_info: 'Home Assistant 0.115.0'
}
[18:54:47][V][api.connection:588]: Hello from client: 'Home Assistant 0.115.0 (192.168.xxx.xxx)'
[18:54:47][VV][api.service:012]: send_hello_response: HelloResponse {
  api_version_major: 1
  api_version_minor: 3
  server_info: 'mirabella_genio_door_window_sensor (esphome v1.16.0-dev)'
}
[18:54:47][VV][api.service:199]: on_connect_request: ConnectRequest {
  password: ''
}
[18:54:47][D][api.connection:604]: Client 'Home Assistant 0.115.0 (192.168.10.240)' connected successfully!
[18:54:47][VV][api.service:016]: send_connect_response: ConnectResponse {
  invalid_password: NO
}
[18:54:47][VV][api.service:234]: on_device_info_request: DeviceInfoRequest {}
[18:54:47][VV][api.service:036]: send_device_info_response: DeviceInfoResponse {
  uses_password: NO
  name: 'mirabella_genio_door_window_sensor'
  mac_address: 'xx:xx:xx:xx:xx:xx'
  esphome_version: '1.16.0-dev'
  compilation_time: 'Sep 19 2020, 18:48:26'
  model: 'PLATFORMIO_ESP01_1M'
  has_deep_sleep: NO
}
[18:54:47][VV][api.service:241]: on_list_entities_request: ListEntitiesRequest {}
[18:54:47][VV][api.service:040]: send_list_entities_done_response: ListEntitiesDoneResponse {}
[18:54:47][VV][api.service:248]: on_subscribe_states_request: SubscribeStatesRequest {}
[18:54:47][VV][api.service:298]: on_subscribe_homeassistant_services_request: SubscribeHomeassistantServicesRequest {}
[18:54:47][VV][api.service:319]: on_subscribe_home_assistant_states_request: SubscribeHomeAssistantStatesRequest {}
[18:54:48][VV][scheduler:131]: Running interval 'heartbeat' with interval=1000 last_execution=16078 (now=17078)
[18:54:48][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=0
[... 50 lines cut ...]
[18:54:53][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[18:54:53][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[18:54:53][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[18:54:53][VV][uart_esp8266:101]:     Wrote 0b11111111 (0xFF)
[18:54:54][VV][scheduler:131]: Running interval 'heartbeat' with interval=1000 last_execution=22078 (now=23081)
[18:54:54][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=0
[18:54:54][VV][uart_esp8266:111]:     Wrote 0b01010101 (0x55)
[18:54:54][VV][uart_esp8266:111]:     Wrote 0b10101010 (0xAA)
[18:54:54][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[18:54:54][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[18:54:54][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[18:54:54][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[18:54:54][VV][uart_esp8266:101]:     Wrote 0b11111111 (0xFF)
[18:54:55][VV][scheduler:131]: Running interval 'heartbeat' with interval=1000 last_execution=23078 (now=24085)
[18:54:55][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=0
[18:54:55][VV][uart_esp8266:111]:     Wrote 0b01010101 (0x55)
[18:54:55][VV][uart_esp8266:111]:     Wrote 0b10101010 (0xAA)
[18:54:55][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[18:54:55][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[18:54:55][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[18:54:55][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[18:54:55][VV][uart_esp8266:101]:     Wrote 0b11111111 (0xFF)
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to mirabella_genio_door_window_sensor.local:6053 (192.168.10.43)
```

jesserockz, 2020-09-19:
Ok, comparing the code from the built in `tuya` and that custom header, I think the protocol is slightly different as the heartbeat in the header is different, so the mcu never responds to the `tuya` one.

Can you flash the other header and capture the logs in at least `VERBOSE` mode so I can try to see the specific data coming in?

spiro-c, 2020-09-19:
here it is
```
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from mirabella_genio_door_window_sensor/debug
INFO Connected to MQTT broker!
[20:50:12][I][app:142]: Rebooting safely...
[20:50:20][I][mqtt:202]: MQTT Connected!
[20:50:21][V][mqtt:372]: Publish(topic='mirabella_genio_door_window_sensor/sensor/doow_window_battery/state' payload='74' retain=1)
[20:50:21][I][app:059]: setup() finished successfully!
[20:50:21][I][app:105]: ESPHome version 1.16.0-dev compiled on Sep 19 2020, 20:49:32
[20:50:21][C][wifi:443]: WiFi:
[20:50:21][C][wifi:303]:   SSID: [redacted]
[20:50:21][C][wifi:304]:   IP Address: 192.168.10.43
[20:50:21][C][wifi:306]:   BSSID: [redacted]
[20:50:21][C][wifi:307]:   Hostname: 'mirabella_genio_door_window_sensor'
[20:50:21][C][wifi:311]:   Signal strength: -50 dB ▂▄▆█
[20:50:21][C][wifi:315]:   Channel: 11
[20:50:21][C][wifi:316]:   Subnet: 255.255.255.0
[20:50:21][C][wifi:317]:   Gateway: 192.168.10.1
[20:50:21][C][wifi:318]:   DNS1: 192.168.10.5
[20:50:21][C][wifi:319]:   DNS2: (IP unset)
[20:50:21][VV][scheduler:131]: Running interval 'update' with interval=1000 last_execution=9021 (now=10063)
[20:50:21][D][adc:056]: 'Doow Window Battery': Got voltage=3.36V
[20:50:21][V][sensor:013]: 'Doow Window Battery': Received new state 3.361328
[20:50:21][VV][sensor.filter:013]: Filter(0x3fff14a4)::input(3.361328)
[20:50:21][VV][sensor.filter:020]: Filter(0x3fff14a4)::output(70.166016) -> SENSOR
[20:50:21][D][sensor:092]: 'Doow Window Battery': Sending state 70.16602 % with 0 decimals of accuracy
[20:50:21][V][mqtt:372]: Publish(topic='mirabella_genio_door_window_sensor/sensor/doow_window_battery/state' payload='70' retain=1)
[20:50:21][C][uart_esp8266:075]: UART Bus:
[20:50:21][C][uart_esp8266:077]:   TX Pin: GPIO1
[20:50:21][C][uart_esp8266:080]:   RX Pin: GPIO3
[20:50:21][C][uart_esp8266:081]:   RX Buffer Size: 256
[20:50:21][C][uart_esp8266:083]:   Baud Rate: 9600 baud
[20:50:21][C][uart_esp8266:084]:   Data Bits: 8
[20:50:21][C][uart_esp8266:085]:   Parity: NONE
[20:50:21][C][uart_esp8266:086]:   Stop bits: 1
[20:50:21][C][uart_esp8266:088]:   Using hardware serial interface.
[20:50:21][C][template.binary_sensor:018]: Template Binary Sensor 'Door Window Sensor'
[20:50:21][C][template.binary_sensor:018]:   Device Class: 'door'
[20:50:21][C][logger:185]: Logger:
[20:50:21][C][logger:186]:   Level: VERY_VERBOSE
[20:50:21][C][logger:187]:   Log Baud Rate: 115200
[20:50:21][C][logger:188]:   Hardware UART: UART1
[20:50:21][C][adc:026]: ADC Sensor 'Doow Window Battery'
[20:50:21][C][adc:026]:   Unit of Measurement: '%'
[20:50:21][C][adc:026]:   Accuracy Decimals: 0
[20:50:21][C][adc:026]:   Icon: 'mdi:flash'
[20:50:21][V][adc:026]:   Unique ID: 'a4cf12a42ed5-adc'
[20:50:21][C][adc:029]:   Pin: VCC
[20:50:21][C][adc:051]:   Update Interval: 1.0s
[20:50:21][C][ota:029]: Over-The-Air Updates:
[20:50:21][C][ota:030]:   Address: mirabella_genio_door_window_sensor.local:8266
[20:50:21][C][mqtt:051]: MQTT:
[20:50:21][C][mqtt:053]:   Server Address: 192.168.10.6:1883 (192.168.10.6)
[20:50:21][C][mqtt:054]:   Username: [redacted]
[20:50:21][C][mqtt:055]:   Client ID: [redacted]
[20:50:21][C][mqtt:060]:   Topic Prefix: 'mirabella_genio_door_window_sensor'
[20:50:21][C][mqtt:062]:   Log Topic: 'mirabella_genio_door_window_sensor/debug'
[20:50:21][C][sb1:242]: SB1 UART:
[20:50:21][C][sb1:243]:   Boot Mode: NORMAL
[20:50:21][C][sb1:244]:   Product Info: {"p":"3ibf4tshedrzkijl","v":"1.0.6"}
[20:50:21][C][wifi_signal.sensor:009]: WiFi Signal 'Door WiFi Signal'
[20:50:21][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[20:50:21][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[20:50:21][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[20:50:21][V][wifi_signal.sensor:009]:   Unique ID: 'a4cf12a42ed5-wifisignal'
[20:50:21][C][mqtt.sensor:024]: MQTT Sensor 'Door WiFi Signal':
[20:50:21][C][mqtt.sensor:028]:   State Topic: 'mirabella_genio_door_window_sensor/sensor/door_wifi_signal/state'
[20:50:21][C][mqtt.binary_sensor:018]: MQTT Binary Sensor 'Door Window Sensor':
[... 1134 lines cut ...]
[20:51:12][C][sb1:244]:   Product Info: {"p":"3ibf4tshedrzkijl","v":"1.0.6"}
[20:51:12][C][wifi_signal.sensor:009]: WiFi Signal 'Door WiFi Signal'
[20:51:12][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[20:51:12][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[20:51:12][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[20:51:12][V][wifi_signal.sensor:009]:   Unique ID: 'a4cf12a42ed5-wifisignal'
[20:51:12][C][mqtt.sensor:024]: MQTT Sensor 'Door WiFi Signal':
[20:51:12][C][mqtt.sensor:028]:   State Topic: 'mirabella_genio_door_window_sensor/sensor/door_wifi_signal/state'
[20:51:12][C][mqtt.binary_sensor:018]: MQTT Binary Sensor 'Door Window Sensor':
[20:51:12][C][mqtt.binary_sensor:019]:   State Topic: 'mirabella_genio_door_window_sensor/binary_sensor/door_window_sensor/state'
[20:51:12][C][mqtt.sensor:024]: MQTT Sensor 'Doow Window Battery':
[20:51:12][C][mqtt.sensor:028]:   State Topic: 'mirabella_genio_door_window_sensor/sensor/doow_window_battery/state'
[20:51:13][VV][scheduler:131]: Running interval 'update' with interval=1000 last_execution=4832 (now=5832)
[20:51:13][D][adc:056]: 'Doow Window Battery': Got voltage=3.35V
[20:51:13][V][sensor:013]: 'Doow Window Battery': Received new state 3.352539
[20:51:13][VV][sensor.filter:013]: Filter(0x3fff14a4)::input(3.352539)
[20:51:13][VV][sensor.filter:020]: Filter(0x3fff14a4)::output(69.067383) -> SENSOR
[... cut ...]
[20:51:13][V][mqtt:372]: Publish(topic='mirabella_genio_door_window_sensor/sensor/doow_window_battery/state' payload='69' retain=1)
[20:51:13][I][sb1:375]: Rebooting: SB1_STATE_RUNNING_NORMAL
[20:51:13][I][app:142]: Rebooting safely...
[20:51:13][VV][preferences:051]: SAVE 32: 0=0x00000000 1=0x0DEFE4E3 (Type=233825507, CRC=0x0DEFE4E3)
[20:51:13][VV][preferences:051]: SAVE 34: 0=0x00000000 1=0x84834245 (Type=2223194693, CRC=0x84834245)
[20:51:13][D][sb1:392]: SB1 UART shutting down; next boot mode NORMAL
[20:51:13][VV][uart_esp8266:182]:     Flushing...```

jesserockz, 2020-09-19:
Ok cool, I will try and take a look into this and see if I can work out the exact differences for the battery devices to make it compatible.

jesserockz, 2020-09-24:
@spiro-c I have had a look into this and compared the `tuya` code to the `sb1_uart.h` 
It is quite different with the protocol flow and data sent. So I suggest to continue that file. For it to become built in it would need to be implemented as a new sensor platform (maybe `tuya_battery`) in the future.

spiro-c, 2020-09-27:
@jesserockz  ok thanks i was just try to see if it work

closed 2021-02-08 by stale[bot]