esphome/issues #1490: HA addon dosen't use web_server port parameter

Mak69, 2020-09-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
HASS.IO

**ESP (ESP32/ESP8266, Board/Sonoff):**
Any

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**
Home Assistant Addon v1.15.1 dosen't use the parameter web_server: port when trying to open device/node web server interface
For example:
My node.yaml device configuration contains:
(`
web_server:
port: 8080
`)
When I try to open web server interface from the add-on web-ui the browser opens the page node.local/ instead of node.loacl:8080/

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

napieraj, 2020-10-01:
are you sure your YAML formatting is correct?
```yaml
web_server:
  port: 8080
```

Mak69, 2020-12-10:
Yes, I am.

```
web_server:
  port: 8080
  auth:
    username: ****
    password: ****
```

If I write the esp_hostname.local:8080 directly in the browser I can connect
From the addon it tryes to connect to esp_hostname.local/