esphome/issues #1491: Waveshare 2.7inch e-Paper HAT shows nothing

damanti-me, 2020-09-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io, HassOS 4.13
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32_devkitC_V4
**ESPHome version (latest production, beta, dev branch)**

1.15.1 console latest
**Affected component:**

https://esphome.io/components/display/waveshare_epaper.html
2.7inch Waveshare HAT (b/w)
https://www.waveshare.com/wiki/2.7inch_e-Paper_HAT

**Description of problem:**
- General Error : Displays shows nothing, doesnt even flicker.
- compiling without an error

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: waveshare_display1
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "xxx"
  password: "xxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Waveshare Display1"
    password: "xxx"

captive_portal:

# Enable logging
logger:
  level: debug

# Enable Home Assistant API
api:

ota:
font:
  - file: "arial.ttf"
    id: my_font
    size: 20

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23

display:
  - platform: waveshare_epaper
    cs_pin: GPIO05
    dc_pin: GPIO17
    busy_pin: GPIO04
    reset_pin: GPIO16
    model: 2.70in
    update_interval: 20s
    lambda: |-
      it.print(0, 10, id(my_font), "Hello World!");

```

**Logs (if applicable):**

[... 184 lines cut ...]
[18:16:41][C][waveshare_epaper:451]:   Rotations: 0 °
[18:16:41][C][waveshare_epaper:451]:   Dimensions: 176px x 264px
[18:16:41][C][waveshare_epaper:452]:   Model: 2.7in
[18:16:41][C][waveshare_epaper:453]:   Reset Pin: GPIO16 (Mode: OUTPUT)
[18:16:41][C][waveshare_epaper:454]:   DC Pin: GPIO17 (Mode: OUTPUT)
[18:16:41][C][waveshare_epaper:455]:   Busy Pin: GPIO4 (Mode: INPUT)
[18:16:41][C][waveshare_epaper:456]:   Update Interval: 20.0s
[18:16:41][C][captive_portal:169]: Captive Portal:
[18:16:41][C][ota:029]: Over-The-Air Updates:
[18:16:41][C][ota:030]:   Address: waveshare_display1.local:3232
[18:16:41][C][api:095]: API Server:
[18:16:41][C][api:096]:   Address: waveshare_display1.local:6053
```

**Additional information and things you've tried:**

- [ ]  compiled examplie sketch with Arduino IDE with the same connection - workes

- [ ]  compiled expample with python with a raspberry pi according to the description of the manufactor - runs like a charm
- [ ] mesured the voltage of the display and the board
- [ ]  Tried different GPIOS according to https://components101.com/sites/default/files/component_pin/ESP32-Pinout.png
- [ ] different Voltage (5V)
- [ ] tried same setup with ESP12e - no success
- [ ] tried simple line command `it.line(0, 0, 50, 50);` for lambda
- [ ] checked cable connection with multimeter

damanti-me, 2020-09-19:
Short add: I tried the exact same approach with the 2.9 inch Display from waveshare and it works like a charm. Seems that the HAT Rev 2.1 version of the 2.7 inch Display is not supported. Is a feature request now the way?

maikkoe, 2020-09-28:
okay, this will probably be my problem too #1478

damanti-me, 2020-10-23:
> This will probably be my problem too #1478

hmm seems like you are running into memory issues with the ESP8266. I also used an ESP32 here. Seems to be different

t9zx, 2021-02-14:
I have the 2.9in Rev2.1 version and tried the approach shown in esphome/esphome#1481 but my display still doesn't show anything. I found this docu of the 2.9in Rev2.1 version here: https://www.waveshare.com/w/upload/7/79/2.9inch-e-paper-v2-specification.pdf and as much as I can tell there are more differences than just the delay of the reset duration in the command table compared to https://www.waveshare.com/w/upload/e/e6/2.9inch_e-Paper_Datasheet.pdf (e.g. commands of the v1 do not exist in v2 anymore).

sfingerling, 2021-03-07:
looks that we could fix it: 
- LUT are wrong, has to be copied from arduino example
- The reset function needs to be changed in order to wake up the display before receving the power on command: 
 void reset_() {
    if (this->reset_pin_ != nullptr) {
      this->reset_pin_->digital_write(true);
      delay(200);  // NOLINT
      this->reset_pin_->digital_write(false);
      delay(10);  // NOLINT
	  this->reset_pin_->digital_write(true);
      delay(200);  // NOLINT
    }

Proper code will follow after clean it up

eecue, 2021-03-26:
Ah glad I found this ticket! I'm experiencing the same thing with ePaper Diver HAT Rev2.1. Confirmed it works fine using RPi example code.   I applied the patch from  #1478 and it fixed my issue.

damanti-me, 2021-05-06:
@eecue  can you describe what patch you mean? What did you do to make it work?

eecue, 2021-05-06:
@damanti-me I changed the code to what @sfingerling showed above... but unfortunately it causes the whole screen to flash black/clear a few times before updating every time.  @sfingerling any idea how to fix that or have you made any further progress on this?

dougle03, 2021-08-19:
> whole screen to flash black/clear a few times before updating every time

I'm getting this too. Did you find a solution?

`spi:
  clk_pin: D1
  mosi_pin: D2
  miso_pin: D7 #not connected
display:
  - platform: waveshare_epaper
    cs_pin: D3
    dc_pin: D4
    busy_pin: D5
    reset_pin: D6
    model: 2.90in
    id: mydisplay
    update_interval: 2s
    full_update_every: 30
    rotation: 90°
    #brightness: 100%
    lambda: |-
      //Header
      it.strftime(0, 0, id(font4), TextAlign::TOP_LEFT, "%a %H:%M", id(esptime).now());
      it.print(0, 50, id(font3), "TESTING");`

eecue, 2021-08-19:
Nope still flashes.

closed 2021-12-12 by jesserockz