esphome/issues #1492: Tuya MCU not initializing since 1.15.0

silicode19, 2020-09-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.): ** Running from Docker: esphome/esphome:latest

**ESP (ESP32/ESP8266, Board/Sonoff): Feit Smart WiFi Dimmer - Intertek 5014010 (ESP8266)**

**ESPHome version (latest production, beta, dev branch) 1.15.1**

**Affected component: Tuya MCU**

**Description of problem: All Web toggle requests and Home Assistant requests do not toggle light.  Only physical button works **

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ${device_name}
  comment: ${device_description}
  platform: ESP8266
  board: esp8285

wifi:
  ssid: !secret ssid_id
  password: !secret wifi_password
  fast_connect: true
  power_save_mode: light

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendly_name} Fallback"
    password: !secret fallback_password

captive_portal:

# Enable Over the Air updates
ota:
  password: !secret main_password

# Enable Home Assistant API
api:
  password: !secret main_password

# Make sure logging is not using the serial port
logger:
  baud_rate: 0
  level: NONE

# My dimmer used the hardware serial port on the alternate pins
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

# Register the Tuya MCU connection
tuya:

# Make the light
light:
  - platform: "tuya"
    id: $device_name
    name: ${friendly_name}
    dimmer_datapoint: 2
    switch_datapoint: 1
    min_value: 10
    max_value: 1000

```

**Logs (if applicable):**

```
[09:46:56][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=0
[09:46:57][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=0
[09:46:58][V][api.connection:693]: Error: Disconnecting Home Assistant 0.115.1 (192.168.x.x)
[09:46:58][D][api:067]: Disconnecting Home Assistant 0.115.1 (192.168.x.x)
[09:46:58][V][api.connection:588]: Hello from client: 'Home Assistant 0.115.1 (192.168.x.x)'
[09:46:58][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=0
[09:46:58][D][api.connection:604]: Client 'Home Assistant 0.115.1 (192.168.x.x)' connected successfully!
[09:46:59][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=0
[09:47:00][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=0
[09:47:01][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=0
...
[09:49:21][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=0
[09:49:22][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=0
[09:49:23][D][light:264]: 'Office Light' Setting:
[09:49:23][D][light:273]:   State: OFF
[09:49:23][V][tuya:275]: Datapoint 1 set to 0
[09:49:23][V][tuya:261]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[01.01.00.01.00 (5)] INIT_STATE=0
[09:49:23][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=0
[09:49:24][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=0
...
[09:57:37][D][light:264]: 'Office Light' Setting:
[09:57:38][V][tuya:275]: Datapoint 2 set to 1000
[09:57:38][V][tuya:261]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[02.02.00.04.00.00.03.E8 (8)] INIT_STATE=0
[09:57:38][V][tuya:275]: Datapoint 1 set to 1
[09:57:38][V][tuya:261]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[01.01.00.01.01 (5)] INIT_STATE=0

```

**Additional information and things you've tried:**

Jenda987, 2020-09-19:
I must agree, there is something wrong with TUYA MCU. If I put 'tuya:' into YAML then device will not even boot (no network access). And I must flash firmware without 'tuya:' over serial (because network didn't work) to make device working again.
I checked tasmota firmware with tuya and tasmota is working with tuya MCU as expected (so it isn't hardware issue).
It's hard to say what is happening because without network access I do not have access to any kind of log (my device looks like dead until I flash with some other firmware).

grrraham, 2020-10-28:
So I have Moes Dimmable (wall) light switches.
They have large rectangular buttons.
The older style with the rounded corners on the buttons work fine with 0-255 (default) scaling.
The newer style with the more square corners on the button, requires scaling, and HomeAssistant / the button has issues.
  After a power cycle, the dimmer wall switch will work perfectly at the device.
  If you play with it from home assistant and set the brightness to minimum, ESPHome will let the Brightness go below the 
   minimum setpoint and this breaks the switch such that the local brightness control will no longer work.

Can the Tuya Light integration be updated so that the min_value and max_value get scaled between 0-100% in Home Assistant?

closed 2021-03-11 by stale[bot]