esphome/issues #1496: WiFi connection issue since Version 1.15.0

Chris591, 2020-09-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):** Sonoff Basic

**ESPHome version (latest production, beta, dev branch)** v1.15.1

**Affected component:** WiFi

**Description of problem:** WiFi can't connect

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
substitutions:
  device_name: test_licht64

esphome:
  name: ${device_name}
  platform: ESP8266
  board: esp8285

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
#  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "${device_name} Fallback Hotspot"
#    password: !secret wifi_fallback_password
#
#captive_portal:

# Enable logging
logger:
#  level: VERBOSE

# Enable Web server
web_server:
  port: 80

# Enable Home Assistant API
api:

# Enable Home Assistant API.
ota:

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: ${device_name}_Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: ${device_name}_WiFi_Signal
    update_interval: 60s

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "${device_name}_Button"
    on_press:
      - switch.toggle: relay
  - platform: gpio
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
      inverted: True
    name: "${device_name}_Taster"
    filters:
      # Small filter, to debounce the button press.
      - delayed_on: 100ms
#      - delayed_off: 1s
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: "${device_name}_Switch"
    pin: GPIO12
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[12:26:39][I][wifi:194]: WiFi Connecting to 'ABC'...
[12:26:45][W][wifi_esp8266:392]: Event: Disconnected ssid='ABC' reason='Probe Request Unsuccessful'
[12:26:45][W][wifi:485]: Error while connecting to network.
[12:26:45][W][wifi:522]: Restarting WiFi adapter...
... repeating these messages all the time ...
```

**Additional information and things you've tried:**

If I enable verbose logging I don't see the issue anymore!

pengelen, 2020-09-21:
Looks like the same issue mentioned in #1472 and #1460

Aleksahek, 2020-09-25:
I have the same problem. how to fix?

pengelen, 2020-09-28:
> I have the same problem. how to fix?

Go back to version 1.14.x.

napieraj, 2020-10-01:
in the case of #1460 upgrading to 1.15.2 solved the issue, can you confirm this?

Chris591, 2020-10-01:
For me it didn't solved it. Will check the logs again if it shows different things now.

Holen Sie sich Outlook für Android<https://aka.ms/ghei36>

Chris591, 2020-10-05:
if I use VERY_VERBOSE logging and remove the first binary sensor gpio

during the compile I see this warning:
In file included from /data/test_licht64/.piolibdeps/test_licht64/ESPAsyncWebServer-esphome/src/SPIFFSEditor.cpp:1:0:
/data/test_licht64/.piolibdeps/test_licht64/ESPAsyncWebServer-esphome/src/SPIFFSEditor.h:16:101: warning: 'SPIFFS' is deprecated (declared at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h:269): SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]
     SPIFFSEditor(const String& username=String(), const String& password=String(), const fs::FS& fs=SPIFFS);

I get this logging output:
_[15:25:33]{l[15:25:33]SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
[15:25:33][I][logger:166]: Log initialized
[15:25:33][C][status_led:014]: Setting up Status LED...
[15:25:33][VV][preferences:038]: LOAD 32: valid=NO, 0=0x5A9B0740 1=0x19D2D243 (Type=233825507, CRC=0x6A755A43)
[15:25:33][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[15:25:33][VV][preferences:051]: SAVE 32: 0=0x00000001 1=0x42F4583F (Type=233825507, CRC=0x42F4583F)
[15:25:33][I][app:029]: Running through setup()...
[15:25:33][V][app:030]: Sorting components by setup priority...
[15:25:33][VV][scheduler:056]: set_interval(name='update', interval=60000, offset=3991)
[15:25:33][D][binary_sensor:034]: 'test_licht64_Taster': Sending initial state OFF
[15:25:33][C][switch.gpio:011]: Setting up GPIO Switch 'test_licht64_Switch'...
[15:25:33][VV][preferences:038]: LOAD 34: valid=NO, 0=0x91C38715 1=0x6E8BB3DC (Type=1408867281, CRC=0x0DADDE47)
[15:25:33][D][switch:025]: 'test_licht64_Switch' Turning OFF.
[15:25:33][VV][preferences:051]: SAVE 34: 0=0x00000000 1=0x53F99BD1 (Type=1408867281, CRC=0x53F99BD1)
[15:25:33][D][switch:045]: 'test_licht64_Switch': Sending state OFF
[15:25:33][D][switch:025]: 'test_licht64_Switch' Turning OFF.
[15:25:33][VV][scheduler:056]: set_interval(name='', interval=900000, offset=161669)
[15:25:33][C][wifi:033]: Setting up WiFi...
[15:25:33][V][wifi_esp8266:051]: Disabling AP.
[15:25:33]bcn 0
[15:25:33]del if1
[15:25:33]usl
[15:25:33]mode : null
[15:25:33][V][wifi_esp8266:041]: Enabling STA.
[15:25:33]mod[V][wifi_esp8266:443]: Event: Changed Mode old=AP new=OFF
[15:25:33]wifi evt: 8
[15:25:33]e : sta(b4:e6:2d:3a:cf:3b)
[15:25:33]add if0
[15:25:33][V][wifi_esp8266:443]: Event: Changed Mode old=OFF new=STA
[15:25:33]wifi evt: 8
[15:25:33]sleep disable
[15:25:33][D][wifi:324]: Starting scan...
[15:25:33][VV][scheduler:131]: Running interval '' with interval=900000 last_execution=4293905804 (now=222)
[15:25:33][VV][scheduler:131]: Running interval 'update' with interval=60000 last_execution=4294903423 (now=222)
[15:25:33][V][sensor:013]: 'test_licht64_Uptime': Received new state 0.241000
[15:25:33][D][sensor:092]: 'test_licht64_Uptime': Sending state 0.24100 s with 0 decimals of accuracy
[15:25:39]scandone
[15:25:39][D][wifi:339]: Found networks:
[15:25:39][D][wifi:341]:   No network found!
[15:25:44][D][wifi:324]: Starting scan...
[15:25:50]scandone
[15:25:50][D][wifi:339]: Found networks:
[15:25:50][D][wifi:341]:   No network found!
[15:25:55][D][wifi:324]: Starting scan...
[15:26:01]scandone
[15:26:01][D][wifi:339]: Found networks:
[15:26:01][D][wifi:341]:   No network found!
[15:26:06][D][wifi:324]: Starting scan..._

Mirodin, 2020-11-11:
I can confirm that this problem still persist for ESPHome version 1.15.3 on my Sonoff basic. Device works fine when I tested with Tasmota and ESPEasy. Are there any news/progress on this?

closed 2021-03-26 by stale[bot]