esphome/issues #1497: Shelly 2.5: i2c NACK issue with ADE7953

Savjee, 2020-09-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
HassOS with the ESPHome add-on

**ESP (ESP32/ESP8266, Board/Sonoff):**
Shelly 2.5 (ESP8266, 1MB)

**ESPHome version (latest production, beta, dev branch)**
`ESPHome version 1.15.1 compiled on Sep 20 2020, 15:35:25`

**Affected component:**
ADE7953 Power Sensor: https://esphome.io/components/sensor/ade7953.html?highlight=ade795

**Description of problem:**
I'm using a configuration for the Shelly 2.5 from this website: https://esphome-configs.io/devices/shelly-25/

Trying to get the power sensor to work, without success. On boot, ESPHome finds no i2c device:

```
[15:35:58][C][i2c:028]: I2C Bus:
[15:35:58][C][i2c:029]:   SDA Pin: GPIO12
[15:35:58][C][i2c:030]:   SCL Pin: GPIO14
[15:35:58][C][i2c:031]:   Frequency: 50000 Hz
[15:35:58][I][i2c:033]: Scanning i2c bus for active devices...
[15:35:58][I][i2c:049]: Found no i2c devices!
```

After about a minute, these `NACK` messages start to show up:

```
[15:40:39][W][i2c:070]: Received NACK on transmit of address 0x38
[15:40:39][W][i2c:070]: Received NACK on transmit of address 0x38
[15:40:39][W][i2c:070]: Received NACK on transmit of address 0x38
[15:40:39][W][i2c:070]: Received NACK on transmit of address 0x38
[15:40:39][W][i2c:070]: Received NACK on transmit of address 0x38
```

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: ade7953
    voltage:
      name: ${devicename} voltage
    current_a:
      name: ${channel_2} current
      internal: true
    current_b:
      name: ${channel_1} current
      internal: true
    active_power_a:
      name: ${channel_2} power
      id: power_channel_2
      filters:
        - multiply: 1
      on_value_range:
        - above: ${max_power}
          then:
            - light.turn_off: lightid2
            - homeassistant.service:
                service: persistent_notification.create
                data:
                  title: Message from ${devicename}
                data_template:
                  message: Switch turned off because power exceeded ${max_power}W
    active_power_b:
      name: ${channel_1} power
      id: power_channel_1
      filters:
        - multiply: -1
      on_value_range:
        - above: ${max_power}
          then:
            - light.turn_off: lightid1
            - homeassistant.service:
                service: persistent_notification.create
                data:
                  title: Message from ${devicename}
                data_template:
                  message: Switch turned off because power exceeded ${max_power}W
    update_interval: 30s
```

**Logs (if applicable):**

This is the problematic one:
```
[15:40:39][W][i2c:070]: Received NACK on transmit of address 0x38
```

**Additional information and things you've tried:**

Things I've tried so far:
* Checked the GPIO numbers & pinouts. They seem correct (and match with Tasmota)
* Tried all i2c bus frequencies
* Restarted the Shelly with the [Restart switch](https://esphome.io/components/switch/restart.html?highlight=restart) 

Unfortunately, I can't easily power-cycle the device because it's in a wall.

I have two Shelly 2.5 units that have this issue. I guess the sensor is going into sleep mode and doesn't recover from that.

dzikus, 2020-09-21:
I'm using four shelly2.5 with ESPHome, none of them is showing "NACK". I'm using 1.14.5, 1.15, 1.15.1 and 1.15.2 on them.

Try to power cycle them for example thru security switch:
[image]

I tried to reproduce your bug without success:
```
Showing Logs For: wykusz_prawe.yaml
INFO Reading configuration /config/esphome/wykusz_prawe.yaml...
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 28 March 02:00:00 to 31 October 03:00:00
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 28 March 02:00:00 to 31 October 03:00:00
INFO Starting log output from shelly25_wykusz_prawy.lan using esphome API
INFO Connecting to shelly25_wykusz_prawy.lan:6053 (172.x.y.104)
INFO Successfully connected to shelly25_wykusz_prawy.lan
[15:44:35][I][app:105]: ESPHome version 1.15.1 compiled on Sep 16 2020, 20:21:08
[15:44:35][C][wifi:443]: WiFi:
[15:44:35][C][wifi:303]:   SSID: [redacted]
[15:44:35][C][wifi:304]:   IP Address: 172.x.y.104
[15:44:35][C][wifi:306]:   BSSID: [redacted]
[15:44:35][C][wifi:307]:   Hostname: 'shelly25_wykusz_prawy'
[15:44:35][C][wifi:311]:   Signal strength: -67 dB ▂▄▆█
[15:44:35][C][wifi:315]:   Channel: 11
[15:44:35][C][wifi:316]:   Subnet: 255.255.255.0
[15:44:35][C][wifi:317]:   Gateway: 172.x.y.254
[15:44:35][C][wifi:318]:   DNS1: 172.x.y.254
[15:44:35][C][wifi:319]:   DNS2: (IP unset)
[15:44:35][D][sensor:092]: 'Wykusz prawy Moc 2': Sending state -0.02597 W with 1 decimals of accuracy
[15:44:35][D][sensor:092]: 'Wykusz prawy zużycie dzienne 2': Sending state 0.00014 kW with 3 decimals of accuracy
[15:44:35][D][sensor:092]: 'Wykusz prawy Moc 1': Sending state -0.01948 W with 1 decimals of accuracy
[15:44:35][D][sensor:092]: 'Wykusz prawy zużycie dzienne 1': Sending state 0.00169 kW with 3 decimals of accuracy
[15:44:35][D][sensor:092]: 'Wykusz prawy prąd 2': Sending state 0.01773 A with 2 decimals of accuracy
[15:44:35][D][sensor:092]: 'Wykusz prawy prąd 1': Sending state 0.01773 A with 2 decimals of accuracy
[15:44:35][D][sensor:092]: 'Wykusz prawy napięcie': Sending state 247.39569 V with 1 decimals of accuracy
[15:44:35][C][i2c:028]: I2C Bus:
[15:44:35][C][i2c:029]:   SDA Pin: GPIO12
[15:44:35][C][i2c:030]:   SCL Pin: GPIO14
[15:44:35][C][i2c:031]:   Frequency: 50000 Hz
[15:44:35][I][i2c:033]: Scanning i2c bus for active devices...
[15:44:35][I][i2c:040]: Found i2c device at address 0x38
[15:44:35][C][uptime.sensor:030]: Uptime Sensor 'Wykusz prawy Uptime'
[15:44:35][C][uptime.sensor:030]:   Unit of Measurement: 's'
[15:44:35][C][uptime.sensor:030]:   Accuracy Decimals: 0
[15:44:35][C][uptime.sensor:030]:   Icon: 'mdi:timer-outline'
[15:44:35][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Wykusz prawy Switch 1'
[15:44:37][C][gpio.binary_sensor:016]:   Pin: GPIO13 (Mode: INPUT)
[15:44:37][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Wykusz prawy Switch 2'
[15:44:37][C][gpio.binary_sensor:016]:   Pin: GPIO5 (Mode: INPUT)
[15:44:37][C][switch.gpio:042]: GPIO Switch 'Wykusz prawy Relay 1'
[15:44:37][C][switch.gpio:043]:   Pin: GPIO4 (Mode: OUTPUT)
[15:44:37][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[15:44:37][C][switch.gpio:061]:   Interlocks:
[15:44:37][C][switch.gpio:065]:     Wykusz prawy Relay 2
[15:44:37][C][switch.gpio:042]: GPIO Switch 'Wykusz prawy Relay 2'
[15:44:37][C][switch.gpio:043]:   Pin: GPIO15 (Mode: OUTPUT)
[15:44:37][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[15:44:37][C][switch.gpio:061]:   Interlocks:
[15:44:37][C][switch.gpio:065]:     Wykusz prawy Relay 1
[15:44:37][C][template.switch:058]: Template Switch 'Wykusz prawy zablokuj przyciski'
[15:44:37][C][template.switch:059]:   Restore State: NO
[15:45:05][D][sensor:092]: 'Wykusz prawy Moc 2': Sending state 0.09740 W with 1 decimals of accuracy
[15:45:05][D][sensor:092]: 'Wykusz prawy zużycie dzienne 2': Sending state 0.00014 kW with 3 decimals of accuracy
[15:45:05][D][sensor:092]: 'Wykusz prawy Moc 1': Sending state 0.19481 W with 1 decimals of accuracy
[15:45:05][D][sensor:092]: 'Wykusz prawy zużycie dzienne 1': Sending state 0.00169 kW with 3 decimals of accuracy
[15:45:05][D][sensor:092]: 'Wykusz prawy prąd 2': Sending state 0.01773 A with 2 decimals of accuracy
[15:45:05][D][sensor:092]: 'Wykusz prawy prąd 1': Sending state 0.01773 A with 2 decimals of accuracy
[15:45:05][D][sensor:092]: 'Wykusz prawy napięcie': Sending state 246.74443 V with 1 decimals of accuracy
[15:45:07][D][adc:056]: 'temp_analog_reading': Got voltage=0.21V
[15:45:07][D][sensor:092]: 'temp_analog_reading': Sending state 0.21289 V with 2 decimals of accuracy
[15:45:07][D][resistance:037]: 'temp_resistance_reading' - Resistance 2206.8Ω
[15:45:07][D][sensor:092]: 'temp_resistance_reading': Sending state 2206.75708 Ω with 1 decimals of accuracy
[... 47 lines cut ...]
[15:47:35][D][sensor:092]: 'Wykusz prawy zużycie dzienne 1': Sending state 0.00170 kW with 3 decimals of accuracy
[15:47:35][D][sensor:092]: 'Wykusz prawy prąd 2': Sending state 0.01773 A with 2 decimals of accuracy
[15:47:35][D][sensor:092]: 'Wykusz prawy prąd 1': Sending state 0.01773 A with 2 decimals of accuracy
[15:47:35][D][sensor:092]: 'Wykusz prawy napięcie': Sending state 246.07777 V with 1 decimals of accuracy
[15:48:05][D][sensor:092]: 'Wykusz prawy Moc 2': Sending state 0.05195 W with 1 decimals of accuracy
[15:48:05][D][sensor:092]: 'Wykusz prawy zużycie dzienne 2': Sending state 0.00014 kW with 3 decimals of accuracy
[... cut ...]
[15:48:35][D][sensor:092]: 'Wykusz prawy Moc 1': Sending state 0.38961 W with 1 decimals of accuracy
[15:48:35][D][sensor:092]: 'Wykusz prawy zużycie dzienne 1': Sending state 0.00170 kW with 3 decimals of accuracy
[15:48:35][D][sensor:092]: 'Wykusz prawy prąd 2': Sending state 0.01773 A with 2 decimals of accuracy
[15:48:35][D][sensor:092]: 'Wykusz prawy prąd 1': Sending state 0.01773 A with 2 decimals of accuracy
[15:48:35][D][sensor:092]: 'Wykusz prawy napięcie': Sending state 247.74292 V with 1 decimals of accuracy
```

Savjee, 2020-09-22:
Thanks for the reply. Toggling the circuit-breakers did indeed fix the problem.
Now I'm wondering: why does it happen, and how often will this be required? 

I'm assuming this is a one-time issue because I flashed ESPHome over-the-air instead of using a USB interface (which cuts power afterward)?

napieraj, 2020-10-01:
closing this as it appears the issue is resolved, feel free to reopen or create a new issue if you have further problems

closed 2020-10-01 by napieraj