esphome/issues #1500: Xiami BLE lost ability to get battery from CCG1 (specific) model sending combined data packets

Alex9779, 2020-09-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

v1.15.2

**Affected component:**

https://esphome.io/components/sensor/xiaomi_ble.html

**Description of problem:**
After https://github.com/esphome/esphome/pull/1027 the component lost the ability to get battery level from certain CGG1 devices.
Before the parsing of messages went in a loop because some CGG1 models send the battery level in the same advertisement as temperature and humidity. Took me a while to figure out so. When I recognized I didn't have the time for research and now the PR is merged and discussion closed...
@ahpohl did you miss that? You just killed the loop...
See #799 with a sample of packet data...

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Alex9779, 2020-09-21:
I just tailored a fix and a PR based on how it worked earlier...

Misiu, 2020-09-21:
@Alex9779 is there a way to test this with Home Assistant?
I have CCG1 and I never got the battery level. Hopefully, your PR will fix that 🤞

Alex9779, 2020-09-21:
Well theoretically yes, but I have no idea how to merge an open PR in the DEV addon.
Much simpler would be to follow the guide to setup a development environment here: https://esphome.io/guides/contributing.html#setting-up-development-environment
But you need a machine with python installed and pip and stuff. The easiest is some kind of Linux machine.
Then you can pull my PR into your local repository and assuming you setup your dev env correctly using "esphome" after that uses it and you can generate a firmware and upload it...

closed 2020-10-13 by glmnet