esphome/issues #1501: frequent abort's in operator new

andreacampi, 2020-09-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome installed on MacOS X 101.15.6 via Home-brew.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ATOM Lite ESP32 Development Kit (https://shop.pimoroni.com/products/atom-lite-esp32-development-kit)

**ESPHome version (latest production, beta, dev branch)**

1.15.1

**Affected component:**

Possibly https://esphome.io/components/prometheus.html

**Description of problem:**
esphome crashes and reboots every 3-5 minutes with an abort. I don't get any readable message, but looking at source code it comes from here: https://code.woboq.org/gcc/libstdc++-v3/libsupc++/new_op.cc.html#54

```c
new_handler handler = std::get_new_handler ();
if (! handler)
    _GLIBCXX_THROW_OR_ABORT(bad_alloc());
```

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: greenhouse
  platform: ESP32
  board: pico32
  on_boot:
    then:
      - light.turn_on:
          id: led
          brightness: 20%
          red: 0%
          green: 0%
          blue: 100%

interval:
  - interval: 5s
    then:
      - if:
          condition:
            not:
              - mqtt.connected:
              - wifi.connected:
          then:
            - light.turn_on:
                id: led
                brightness: 100%
                red: 100%
                green: 0%
                blue: 0%

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Greenhouse Fallback Hotspot"
    password: !secret wifi_ap_password

captive_portal:

# Enable logging
logger:
  logs:
[... 157 lines cut ...]
WARNING Decoded 0x400f223d: cbuf::resize(unsigned int) at /Users/acampi/.platformio/packages/framework-arduinoespressif32/cores/esp32/cbuf.cpp:117
WARNING Decoded 0x400f2281: cbuf::resizeAdd(unsigned int) at /Users/acampi/.platformio/packages/framework-arduinoespressif32/cores/esp32/cbuf.cpp:117
WARNING Decoded 0x400eba88: AsyncResponseStream::write(unsigned char const*, unsigned int) at /Users/acampi/home-automation/esphome/greenhouse/.piolibdeps/greenhouse/ESPAsyncWebServer-esphome/src/WebResponses.cpp:594
WARNING Decoded 0x400ebaac: non-virtual thunk to AsyncResponseStream::write(unsigned char const*, unsigned int)
WARNING Decoded 0x400f106e: Print::write(char const*) at /Users/acampi/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:234
WARNING Decoded 0x400f1081: Print::print(__FlashStringHelper const*)
WARNING Decoded 0x400db4ba: esphome::prometheus::PrometheusHandler::sensor_row_(AsyncResponseStream*, esphome::sensor::Sensor*) at /Users/acampi/home-automation/esphome/greenhouse/src/esphome/components/prometheus/prometheus_handler.cpp:70
WARNING Decoded 0x400dba43: esphome::prometheus::PrometheusHandler::handleRequest(AsyncWebServerRequest*) at /Users/acampi/home-automation/esphome/greenhouse/src/esphome/components/prometheus/prometheus_handler.cpp:13
WARNING Decoded 0x400eb02d: AsyncWebServerRequest::_parseLine() at /Users/acampi/home-automation/esphome/greenhouse/.piolibdeps/greenhouse/ESPAsyncWebServer-esphome/src/WebRequest.cpp:797
WARNING Decoded 0x400eb0f1: AsyncWebServerRequest::_onData(void*, unsigned int) at /Users/acampi/home-automation/esphome/greenhouse/.piolibdeps/greenhouse/ESPAsyncWebServer-esphome/src/WebRequest.cpp:797
WARNING Decoded 0x400eb359: std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#8}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at /Users/acampi/home-automation/esphome/greenhouse/.piolibdeps/greenhouse/ESPAsyncWebServer-esphome/src/WebRequest.cpp:797
 (inlined by) _M_invoke at /Users/acampi/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
WARNING Decoded 0x401d2301: std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at /Users/acampi/home-automation/esphome/greenhouse/.piolibdeps/greenhouse/AsyncTCP-esphome/src/AsyncTCP.cpp:1040
 (inlined by) AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at /Users/acampi/home-automation/esphome/greenhouse/.piolibdeps/greenhouse/AsyncTCP-esphome/src/AsyncTCP.cpp:935
WARNING Decoded 0x401d237d: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at /Users/acampi/home-automation/esphome/greenhouse/.piolibdeps/greenhouse/AsyncTCP-esphome/src/AsyncTCP.cpp:1040
WARNING Decoded 0x401d2a4e: _async_service_task(void*) at /Users/acampi/home-automation/esphome/greenhouse/.piolibdeps/greenhouse/AsyncTCP-esphome/src/AsyncTCP.cpp:1040
 (inlined by) _async_service_task at /Users/acampi/home-automation/esphome/greenhouse/.piolibdeps/greenhouse/AsyncTCP-esphome/src/AsyncTCP.cpp:194
WARNING Decoded 0x4008ec91: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[14:13:40]
[14:13:40]Rebooting...
```

**Additional information and things you've tried:**

I am starting to remove pieces of configuration to see if I can come up with a simpler repro.

andreacampi, 2020-09-21:
I disabled only the Prometheus component and now I get crashes every 7-10 minutes:

```
[14:40:09][D][esp-idf:000]: E (529769) phy_init: failed to allocate memory for RF calibration data
[14:40:09]
[14:40:09]abort() was called at PC 0x400f4f0d on core 0
[14:40:09]
[14:40:09]Backtrace: 0x40092548:0x3ffe9e90 0x40092779:0x3ffe9eb0 0x400f4f0d:0x3ffe9ed0 0x40113f38:0x3ffe9f00 0x4011403b:0x3ffe9f30 0x4011435a:0x3ffe9f60 0x4011060e:0x3ffe9f90 0x40095697:0x3ffe9fb0 0x4008ec91:0x3ffe9ff0
WARNING Found stack trace! Trying to decode it
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d esphome/greenhouse -t idedata
WARNING Decoded 0x40092548: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
WARNING Decoded 0x40092779: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
WARNING Decoded 0x400f4f0d: esp_phy_load_cal_and_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/phy_init.c:579 (discriminator 2)
WARNING Decoded 0x40113f38: wifi_rf_phy_enable
WARNING Decoded 0x4011403b: wifi_hw_start
WARNING Decoded 0x4011435a: wifi_start_process
WARNING Decoded 0x4011060e: ieee80211_ioctl_process
WARNING Decoded 0x40095697: ppTask
WARNING Decoded 0x4008ec91: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

This looks like https://github.com/esphome/issues/issues/1215 . Given that the original issue is also during memory allocation, I'm kinda wondering if it's just the same root cause, i.e. excessive memory utilization in the BLE stack.

napieraj, 2020-10-01:
have you tried removing or commenting out all BLE components from your configuration to verify?

andreacampi, 2020-10-01:
I did and without it it is stable.

closed 2021-02-12 by stale[bot]