esphome/issues #1502: Sensorc ntc higher tempature by 10 degrees then in reality

dzikus, 2020-09-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

Shelly 2.5

**ESPHome version (latest production, beta, dev branch)**

1.15.1

**Affected component:**

ntc, resistance, adc

**Description of problem:**

On 1.14.5 temperature is compatible with original Shelly firmware (indications are very similar), starting since 1.15 temperature is higher by ~10 degerees C/K.
These two are in the same wall, very close to each other (2cm gap between):
[image]
[image]

Here is history graph of another one ntc temperature also in the same wall before and after upgrade to 1.15:
[image]

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: shelly25_wykusz_prawy
  longdevicename: Wykusz prawy
  pin_led1: GPIO0
  pin_button1: GPIO2
  pin_relay1: GPIO15
  pin_switch2n: GPIO5
  pin_sda: GPIO12
  pin_switch1n: GPIO13
  pin_scl: GPIO14
  pin_relay2: GPIO4
  pin_ade7953: GPIO16
  pin_temp: A0
  open_duration: 26s
  close_duration: 25s
#  open_duration: 27s
#  close_duration: 26s
  max_power: "200"
  max_temp: "70.0"

esphome:
  name: "${devicename}"
  platform: ESP8266
  board: modwifi

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret domain
  ap:
    ssid: "esp-${devicename}"
    password: !secret wifi_password

captive_portal:

logger:

api:

ota:

web_server:
  port: 80

time:
  - platform: sntp
    id: sntp_time
    timezone: Europe/Warsaw
    servers:
      - !secret home_ntp
#      - time.google.com
  - platform: homeassistant

i2c:
  sda: ${pin_sda}
  scl: ${pin_scl}

sensor:
  - platform: ade7953
    voltage:
      name: ${longdevicename} napięcie
    current_a:
      name: ${longdevicename} prąd 2
      internal: true
    current_b:
      name: ${longdevicename} prąd 1
      internal: true
    active_power_a:
      name: ${longdevicename} Moc 2
      id: power_channel_2
      filters:
        - multiply: 1
      on_value_range:
        - above: ${max_power}
          then:
            - switch.turn_off: relay1
            - switch.turn_off: relay2
            - homeassistant.service:
                service: persistent_notification.create
                data:
                  title: Wiadomość od ${longdevicename}
                data_template:
                  message: ${longdevicename} A pobiera za dużo mocy, ponad ${max_power}W
    active_power_b:
      name: ${longdevicename} Moc 1
      id: power_channel_1
      filters:
        - multiply: -1
      on_value_range:
        - above: ${max_power}
          then:
            - switch.turn_off: relay1
            - switch.turn_off: relay2
            - homeassistant.service:
                service: persistent_notification.create
                data:
                  title: Wiadomość od ${longdevicename}
                data_template:
                  message: ${longdevicename} B pobiera za dużo mocy, ponad ${max_power}W
    update_interval: 30s
  - platform: total_daily_energy
    name: ${longdevicename} zużycie dzienne 1
    power_id: power_channel_1
    filters:
      - multiply: 0.001
    unit_of_measurement: kW
  - platform: total_daily_energy
    name: ${longdevicename} zużycie dzienne 2
    power_id: power_channel_2
    filters:
      - multiply: 0.001
    unit_of_measurement: kW
  - platform: wifi_signal
    name: "${longdevicename} WiFi Signal"
    update_interval: 300s
  - platform: uptime
    name: "${longdevicename} Uptime"
    update_interval: 300s
  - platform: ntc
    sensor: temp_resistance_reading
    name: ${longdevicename} temperatura
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    icon: "mdi:thermometer"
    calibration:
      b_constant: 3350
      reference_resistance: 10kOhm
      reference_temperature: 298.15K
    on_value_range:
      - above: ${max_temp}
        then:
          - switch.turn_off: relay1
          - switch.turn_off: relay2
          - homeassistant.service:
              service: persistent_notification.create
              data:
                title: Wiadomość od ${longdevicename}
              data_template:
                message: ${longdevicename} wyłączony ponieważ temperatura maksymalna ${max_temp} została przekroczona
  - platform: resistance
    id: temp_resistance_reading
    sensor: temp_analog_reading
    configuration: DOWNSTREAM
    resistor: 32kOhm
  - platform: adc
    id: temp_analog_reading
    pin: A0

text_sensor:
  - platform: version
    name: "${longdevicename} ESPHome Version"
    internal: true
  - platform: wifi_info
    ip_address:
      name: "${longdevicename} IP Address"
      internal: true
    ssid:
      name: "${longdevicename} Connected SSID"
      internal: true
    bssid:
      name: "${longdevicename} Connected BSSID"
      internal: true

binary_sensor:
  - platform: gpio
    pin: ${pin_switch1n}
    name: "${longdevicename} Switch 1"
    internal: true
    id: switch1
  - platform: gpio
    pin: ${pin_switch2n}
    name: "${longdevicename} Switch 2"
    internal: true
    id: switch2

switch:
  - platform: gpio
    pin: ${pin_relay1}
    name: "${longdevicename} Relay 1"
    internal: true
    id: relay1
    interlock: &interlock_group [relay1, relay2]
  - platform: gpio
    pin: ${pin_relay2}
    name: "${longdevicename} Relay 2"
    internal: true
    id: relay2
    interlock: *interlock_group
  - platform: template
    id: block_control
    name: "${longdevicename} zablokuj przyciski"
    optimistic: true
  - platform: template
    name: "Move UP"
    internal: true
    lambda: |-
      if (id(switch1).state && !id(block_control).state) {
        return true;
      } else {
        return false;
      }
    on_turn_on:
      then:
        cover.open: roller
    on_turn_off:
      then:
        cover.stop: roller
  - platform: template
    name: "Move DOWN"
    internal: true
    lambda: |-
      if (id(switch2).state && !id(block_control).state) {
        return true;
      } else {
        return false;
      }
    on_turn_on:
      then:
        cover.close: roller
    on_turn_off:
      then:
        cover.stop: roller
  - platform: restart
    name: "${longdevicename} Restart"
  - platform: shutdown
    name: "${longdevicename} Shutdown"
    internal: true

cover:
  - platform: time_based
    name: "${longdevicename}"
    device_class: shutter
    id: roller
    open_action:
      - switch.turn_on: relay2
    open_duration: ${open_duration}
    close_action:
      - switch.turn_on: relay1
    close_duration: ${close_duration}
    stop_action:
      - switch.turn_off: relay1
      - switch.turn_off: relay2
```

dzikus, 2020-09-25:
I did some tests and problem appear on 1.15.0b2 release, on 1.15.0b1 temperature isn't too high.

dzikus, 2020-09-25:
Found it: it is dependent on "arduino_version" that was bumped on 1.15.0b2 release. On 1.15.2 with arduino_version: 2.4.2 it is working as expected, on 2.5.2 it doesn't compile because of ntc, on 2.6.3 it is working as expected. Problem appears on arduino_version: 2.7.2

christophebaraer, 2020-12-09:
Thanks @dzikus for the analysis. I'm also observing an increase of heat but couldn't quantify it. 

Looking around, it appeared the same problem occurred on Tasmota but has since been addressed https://github.com/arendst/Tasmota/issues/7991#issuecomment-698252659

Can this be the same root cause on ESPhome ?

Lovely-Maisonette, 2020-12-16:
Tasmota had a problem with the floating IRQ pin. Here, the configuration above also is leaving the pin floating.
I know that is a problem in itself, not sure if only when using a specific Arduino version. 

To fix the floating pin add this to your yaml
binary_sensor:
  # Prevent "floating" pin (less current/heat)
  - platform: gpio
    id: ADE7953
    pin:
      number: GPIO16
      mode: INPUT_PULLUP

This will lower the temperatures. The question is if the temps return to the original values (meaning the new Arduino version keeps the pin floating, while the old one did not !?!) or they go down to a different level (meaning that additionally to the floating pin in all versions, there is a problem with the temp reading in the new version)

dzikus, 2020-12-17:
> Are you sure the temperature is wrong and not actually running hotter (or maybe both?)

Are you kidding? You don’t even read my whole analysis or saw screenshots of my graphs. I have already shown what the reason is.

Lovely-Maisonette, 2020-12-18:
I would kindly ask you show more respect here, especially when someone is trying to help you!
And please, read my answer again.

I presume the temp reading is NOT WRONG, but in fact running hotter. You would need an external temp sensor to verify that.
The explanation could be that the one running cooler as no floating pin. Did you even try my solution?!? Try and only then report back your results without bashing.

christophebaraer, 2020-12-18:
Thank you @Lovely-Maisonette for the help. It did work for me and I see the temperatures on the Shelly 2.5 drop from about 80+°C to 70°C when both relays are active. 

One question from a neophyte, is the Shelly actually running hot with the pin floating or is it just the temperature being reported inaccurately? 

Has someone been able to compare the temperature reported when using ESPHome vs the stock firmware ?

Lovely-Maisonette, 2020-12-18:
There might be one or two issues here. 

1) Floating pin (not being configured as input or output) , it acts as a load, creating additional heat
If you temp went down with the PULLUP, the pin was floating. The 10C temp reduction is real!

2) Is there additionally a reading problem? I do not know. @dzikus says he saw temp differences by simply using a different Arduino version. This may have two explanations
a) Somehow older Arduino versions do the PULLUP by default implicitly
b) there is a reading problem
c) both.

@ christophebaraer 
Could you try the following configurations and report the temperatures to confirm?
1. Old Arduino, no PULLUP
2. Old Arduino, PULLUP
3. New  Arduino, no PULLUP
4. New  Arduino, PULLUP

My guess is that 1,2 and 4 will be identical 70°C, and 3 80°C
You probably tested 3 and 4 already

You can try the @dzikus reported OK version (2.6.3) with

```
esphome:
  # ...
  arduino_version: 2.6.3
```

blackjid, 2021-01-17:
I have tested this on my shelly 25.

- framework-arduinoespressif8266 3.20704.0 (2.7.4)   no PULLUP `~72c`
- framework-arduinoespressif8266 3.20704.0 (2.7.4)   PULLUP `~62c`
- framework-arduinoespressif8266 3.20603.200130 (2.6.3)    no PULLUP `~62c`
- framework-arduinoespressif8266 3.20603.200130 (2.6.3)    PULLUP `~62c`

All the readings were with the relays off...

Looks consistent with @dzikus readings...

After reading [this](https://savjee.be/2020/11/shelly25-esphome-potential-fire-hazard-fix/) I wonder how can I get the shellys 25 to run at lower temps??, In the post the author says he runs them at around `50c`
Any ideas??

Lovely-Maisonette, 2021-01-17:
@blackjid, dzikus said he saw only a 10C increase (in relation to the original shelly firmware) 
Now we can tell this was a real 10C increase do to increase current. The PULLUP brings the current (and therefore the temperature) to the shelly non esphome values.

That would mean that your 62C are normal, and if someone has lower temp readings is because they have it better aired, different walling or enclosure, etc. I for instance get 75C reading with PULLUP, but my installation is very tight, and my walls do not absorb and dissipate much heat. 

The shelly2.5 is rated for 90C Do not forget to add a section to your yaml to switch both relays off if temp goes for any reason above that number. The temp us caused by the shunt resistor, to calculate the power usage. If you want a shelly that runs cooler, you need one without PM.

blackjid, 2021-01-17:
Thanks! @Lovely-Maisonette 

Yes, I have the section to protect for higher temps..

So I shouln't be worried if running at 78c right? with the relays on and with the load I want to power... (a couple of lifx bulbs and a google home...)

christophebaraer, 2021-01-17:
@blackjid @Lovely-Maisonette 

I ended up moving back to the original Shelly firmware and I still get temperatures around 72C. 

I contacted the guys at Shelly who confirmed this was the nominal operating temperature.

blackjid, 2021-01-17:
Thanks for the info....

closed 2021-06-02 by stale[bot]