esphome/issues #1503: Problem to connect to WiFi when set scan parameters on ble tracker

JoseMlCuevas, 2020-09-21:
Operating system
HassOS 4.13
Supervisor
Version
245
**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32
Please provide details about which ESP you're using below.
--> heltec_wifi_kit_32

**ESPHome version (latest production, beta, dev branch)**
latest production
ESPHome version like v1.14 or 1.15-dev
1.15.2

**Affected component:**
https://esphome.io/components/esp32_ble_tracker.html

**Description of problem:**
When I add the "scan_parameters" to the esp32_ble_tracker, the device doesn't connect to wifi, I tested API nad MQTT but nothing happens, I delete the options of scan_parameters then it works fine.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: plant_monitor
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  power_save_mode: none
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_pass
  - ssid: !secret home_ssid
    password: !secret home_pass

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Plant Monitor Fallback Hotspot"
    password: "7YfGQlbGmP9d"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
    
web_server:
  port: 80

status_led:
  pin: GPIO2

esp32_ble_tracker:
  scan_parameters:
    interval: 200ms
    window: 200ms
    active: false

binary_sensor:
  - platform: status
    name: "ESP32s BLE Tracker Status"
  - platform: ble_presence
    mac_address: XX:XX:XX:XX:XX:XX:
    name: "iBeacon"
sensor:
  - platform: xiaomi_hhccjcy01
    mac_address: 'XX:XX:XX:XX:XX:'
    temperature:
      name: "Plant Temperature"
    moisture:
      name: "Plant Moisture"
    illuminance:
      name: "Plant Illuminance"
    conductivity:
      name: "Plant Soil Conductivity"
    battery_level:
      name: "Plant Monitor Battery Level"

PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Alex9779, 2020-11-04:
You can't use the same number for interval and window as this takes up all your radio time by the BLE stack. So it is no surprise oyu don't find wifi networks or can connect because there is no time for the wifi stack to use the radio.

As far as I can tell all of my devices work best with a scan_window of 120ms with default scan_interval of 320ms.
That is for multiple Xiaomi CGG1, LYWSDGQ and Ruuvi tags...
With just all the default values I don't get advertisements from some of them, seems to depend on their specific firmware.

I wanted to update the docs some time ago with some hints but forget about it...

JoseMlCuevas, 2020-11-06:
Thank you for your answer and your time, I appreciate a lot this information. I was reading the article:

 

http://dev.ti.com/tirex/content/simplelink_academy_cc2640r2sdk_1_12_01_16/modules/ble_scan_adv_basic/ble_scan_adv_basic.html#scanning

 

there use the same time on the example

In this task, we will scan continuously with the following scan parameters:

*	Scan interval: 150 milliseconds(ms)
*	Scan window: 150 milliseconds(ms)
*	Scan duration: 5 seconds(s)
*	Scan type: Passive

 

 

I’m looking the way to the device detect instantly the iBeacon, I mean if I turn off the iBeacon the ESP32 recognized immediately  and if I turn on it detect immediately.

I try this:

 

esp32_ble_tracker:

  scan_parameters:

    interval: 320ms

    window: 120ms

    duration: 60s

    active: true

 

but it is not instant, if you have some information to help.

 

Thanks

Have a nice day.

closed 2021-03-20 by stale[bot]