esphome/issues #1504: Dashboard Accessibility Issues - Status Bar Color

nilsmau, 2020-09-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

n.a.
**ESP (ESP32/ESP8266, Board/Sonoff):**

n.a.
**ESPHome version (latest production, beta, dev branch)**

v1.15 onwards 
**Affected component:**

ESPHome Dashboard

**Description of problem:**
It is nearly impossible for people with red-green colour blindness to differentiate by the status bar whether a device is online or not. The red and green are too similar. In the old dashboard, the animated dot helped. This is gone now.
Please change respective classes in esphome.css "card status_offline" and "card status_online" - in addition, I would recommend to use a second indicator (symbol, letter) to show device states.

**Problem-relevant YAML-configuration entries:**
```yaml
```

**Logs (if applicable):**

```
```

**Additional information and things you've tried:**

randybb, 2020-09-21:
Related to https://github.com/esphome/feature-requests/issues/890

nilsmau, 2021-02-03:
Any news?
https://github.com/esphome/feature-requests/issues/1033

jesserockz, 2021-07-20:
This is fixed in esphome/dashboard#44 and will be released on Wednesday :tada:

closed 2021-07-20 by jesserockz