esphome/issues #1506: How correct value of total_daily_energy manually

ricain59, 2020-09-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**
wemos d1 

**ESPHome version (latest production, beta, dev branch)**
 v1.15.2

**Affected component:**
total_daily_energy

**Description of problem:**
My problem is a energy_total_daily up to 110kw without reason. It's possible correct this value manually?
[image]

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

jesserockz, 2020-09-28:
There will most likely be an explanation for the jump. Can you please paste the relevant YAML-configuration?
This includes the `total_daily_energy` and the power sensor it is using.

ricain59, 2020-09-28:
Of course, this is my yaml:

`sensor:
  - platform: pzemac
    current:
      name: "PZEMV3 Current"
    voltage:
      name: "PZEMV3 Voltage"
    power:
      name: "PZEMV3 Power"
      id: pzem_power
    frequency:
      name: "PZEMV3 Frequency"
    power_factor:
      name: "PZEMV3 Power Factor"
    update_interval: 10s
    
  - platform: total_daily_energy
    name: "PZEMV3 Total Daily Energy"
    power_id: pzem_power
    filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
    unit_of_measurement: kWh
    icon: mdi:counter

  - platform: integration
    name: "PZEMV3 Energy Meter" 
    sensor: pzem_power
    time_unit: h
    filters:
     # Multiplication factor from W to kW is 0.001
      - lambda: return x * (0.001);
    unit_of_measurement: kWh
    icon: mdi:counter
    restore: true`

jesserockz, 2020-09-28:
Thanks,

- Is the same thing happening for your `integration` sensor? Or just the `total_daily_energy`?
- Does it happen consistently around the same time each day?
- Can you get logs from ESPHome at the time when it happens?

ricain59, 2020-09-28:
Hi,

- Just total_daily_energy 
- happen random days and rarely
- Where i can find the logs? i have syslog server but i think esphome not permitted connect to syslog

This is pzemv3 + wemos d1 + hilink power supply

Thank you.

jesserockz, 2020-09-28:
Ok if its random days and times then it will be hard to catch. The logs are not stored anywhere specific, just pushed out via uart and api if you connect via the esphome dashboard or cli.

Both `total_daily_energy` and `integration` subscribe to the state the exact same way and do the exact same calculations (in this instance), so I am not sure where it could be going wrong.

ricain59, 2020-09-29:
Correct, both subscribe the state.
Yesterday on fail current on home, went up again 111kw. It's strange.
Any way correct this value manually? on tasmota it's possible via the console and on esphome not found way to do this.

jesserockz, 2020-09-29:
No in the current code you can't. If you are up to the task you could create a reset/set function and action for it? Otherwise put in a feature-request and reference this issue too.

Sorry I couldn't be more help at this point in time.

closed 2021-02-10 by stale[bot]