esphome/issues #1507: Thermostat Climate will not set to Target Temperature below 10 C

MagnaPilot, 2020-09-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASS.IO
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, HELTEC HTIT-WB32 and NodeMCU 1.1

**ESPHome version (latest production, beta, dev branch)**

1.16-dev
**Affected component:**

https://next.esphome.io/components/climate/thermostat.html

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:

  - platform: gpio
    pin: 
      number: 0
      mode: INPUT_PULLUP
    internal: true
    name: up_button
    filters:
      - delayed_on: 500ms
    on_press:
      then:
        - climate.control:
            id: refridgerator_thermostat
            mode: 'COOL'
            target_temperature: !lambda return (id(refridgerator_thermostat).target_temperature + 1.0);
        - component.update: my_display
    
  - platform: gpio
    pin: 
      number: 17
      mode: INPUT_PULLUP
    internal: true
    name: down_button
    filters:
      - delayed_on: 500ms
    on_press:
      then:
        - climate.control:
            id: refridgerator_thermostat
            mode: 'COOL'
            target_temperature: !lambda return (id(refridgerator_thermostat).target_temperature - 1.0);
        - component.update: my_display
        
  - platform: gpio
    pin: 
      number: 12
      mode: INPUT_PULLUP
    internal: true
    name: power_button
    filters:
      - delayed_on: 500ms
    on_press:
      then:
        - climate.control:
            id: refridgerator_thermostat
            mode: 'COOL'
            target_temperature: 1
        - component.update: my_display

climate:
  - platform: thermostat
    sensor: refridgerator_sensor_10k
    id: refridgerator_thermostat
    name: rv_refridgerator_thermostat
    hysteresis: 1.0
    default_target_temperature_high: 1 °C
    cool_action:
      - switch.turn_on: tankheater1
    idle_action:
      - switch.turn_off: tankheater1
```

**Logs (if applicable):**

```
[D][ntc:026]: 'Refridgerator Temperature' - Temperature: 24.9°C
[D][sensor:092]: 'Refridgerator Temperature': Sending state 24.88594 °C with 1 decimals of accuracy
[D][climate:262]: 'rv_refridgerator_thermostat' - Sending state:
[D][climate:265]:   Mode: COOL
[D][climate:267]:   Action: COOLING
[D][climate:276]:   Current Temperature: 24.89°C
[D][climate:282]:   Target Temperature: 10.00°C
[D][binary_sensor:036]: 'up_button': Sending state OFF
[D][binary_sensor:036]: 'up_button': Sending state ON
[D][climate:010]: 'rv_refridgerator_thermostat' - Setting
[D][climate:014]:   Mode: COOL
[D][climate:025]:   Target Temperature: 11.00
[D][climate:262]: 'rv_refridgerator_thermostat' - Sending state:
[D][climate:265]:   Mode: COOL
[D][climate:267]:   Action: COOLING
[D][climate:276]:   Current Temperature: 24.89°C
[D][climate:282]:   Target Temperature: 11.00°C
[D][binary_sensor:036]: 'power_button': Sending state OFF
[D][binary_sensor:036]: 'power_button': Sending state ON
[D][climate:010]: 'rv_refridgerator_thermostat' - Setting
[D][climate:014]:   Mode: COOL
[D][climate:025]:   Target Temperature: 1.00
[D][climate:262]: 'rv_refridgerator_thermostat' - Sending state:
[D][climate:265]:   Mode: COOL
[D][climate:267]:   Action: COOLING
[D][climate:276]:   Current Temperature: 24.89°C
[D][climate:282]:   Target Temperature: 10.00°C
[D][binary_sensor:036]: 'down_button': Sending state OFF
[D][binary_sensor:036]: 'down_button': Sending state ON
[D][climate:010]: 'rv_refridgerator_thermostat' - Setting
[D][climate:014]:   Mode: COOL
[D][climate:025]:   Target Temperature: 9.00
[D][climate:262]: 'rv_refridgerator_thermostat' - Sending state:
[D][climate:265]:   Mode: COOL
[D][climate:267]:   Action: COOLING
[D][climate:276]:   Current Temperature: 24.89°C
[D][climate:282]:   Target Temperature: 10.00°C

```

**Additional information and things you've tried:**

I have attempted setting both the target and target_high temperatures, no changes have been found.  Even with the default set target of 1C, the thermostat still defaults to 10C upon boot.  I have run this code on two different branded ESP32 modules.  I did not include the sensor code, but did include the button code for reference on how I was requesting the value change.  If my code is the cause of the issue, I apologize, I've spent a few hours trying to understand this issue and have had no luck on the internet finding similar problems.

MagnaPilot, 2020-09-28:
updated to put description in the correct location

kbx81, 2020-10-02:
@MagnaPilot have you tried setting `min_temperature` and `max_temperature` as documented here?: https://next.esphome.io/components/climate/index.html#base-climate-configuration

MagnaPilot, 2020-10-02:
> 
> 
> @MagnaPilot have you tried setting `min_temperature` and `max_temperature` as documented here?: https://next.esphome.io/components/climate/index.html#base-climate-configuration

I adjusted the code to give it a shot.  You were correct.  Once I put the Visual Min, Max, and Step in the Thermostat configuration, I was then able to change the target Temperature.  I hadn't considered it, as it's listed as a Visual and Optional setting, and this ESP won't be connecting to a Home Assistant at first.  I'm not sure if I am correct to assume this is a bug, or an error in the documentation, though I'm not sure why the visual setting would dictate the thermostat operation either.  Thank you for the suggestion, this does get around the issue if anything.

napieraj, 2020-10-02:
the `visual` parameter is indeed optional and should not affect the operation of the climate device:
> these do not affect operation and are solely for controlling how the climate device shows up in the frontend

kbx81, 2020-10-02:
What I may do then is a PR to break `visual` `min_temperature` and `max_temperature` out as separate configuration entities. What I'm unclear on is why you would ever want to set the temperature on the thermostat device to a value outside of the range of what can be selected in the Home Assistant UI. This is not possible on any thermostat that I know of; such was was my thought process when creating this component. Can you explain your use case/why you would want to do this?

The behavior you were seeing is not a bug -- there is validation code to ensure a value outside of the min/max is not allowed. This is required to ensure buggy code elsewhere cannot set an invalid/undesired value on the component. For example, what if a buggy automation sets your thermostat to -4000000 degrees and as a result your heat never switches on...and you're on vacation for three weeks in the middle of winter? (This is a thing that actually happened during development!) The validation code is arguably "safety" code that ensures reasonably behavior in all cases.

MagnaPilot, 2020-10-02:
> 
> 
> What I may do then is a PR to break `visual` `min_temperature` and `max_temperature` out as separate configuration entities. What I'm unclear on is why you would ever want to set the temperature on the thermostat device to a value outside of the range of what can be selected in the Home Assistant UI (and, imho, makes for an inconsistent user experience). This is not possible on any thermostat that I know of; such was was my thought process when creating this component. Can you explain your use case/why you would want to do this?
> 
> The behavior you were seeing is not a bug -- there is validation code to ensure a value outside of the min/max is not allowed. This is required to ensure buggy code elsewhere cannot set an invalid/undesired value on the component. For example, what if a buggy automation sets your thermostat to -4000000 degrees and as a result your heat never switches on...and you're on vacation for three weeks in the middle of winter? (This is a thing that actually happened during development!) The validation code is arguably "safety" code that ensures reasonable behavior in all cases.

@kbx81, I don't disagree that there should be a min and max safety boundary to assure that Thermostat cannot go outside a safe range, but if that were the case, I would also recommend that they not be part of an optional visual configuration, as it isn't currently obvious that this setting has an unseen effect on the target_temperature.  If that is indeed the case, and there is a hard safety limit as set by the min and max, I would think it should be required variable directly for thermostat to operate as you mentioned, and not just an optional visual item.  I would also think that if someone were to be using this as a Internal only thermostat, would the min and max variable as currently configured even work at all?  I'd recommend there be a safety_mintemp and safety_maxtemp required variable for thermostat or climate in general, and the visual min/max be configured as a visual gui configuration only as the manual currently describes.  

My use case in this instance is as a controller for a refrigerator in an RV, that will not initially have a Home Assistant instance to constantly report to, and once the RV does have it's own mobile HA instance, it may not be a reliable always on device due to the mobile nature of RVs.  

Whichever way you decided to go about it, I appreciate your looking into it either way.

kbx81, 2020-10-02:
Happy to help! It seems to me that the best thing to do in this case is to update the documentation to call out that `visual` `min_temperature` and `max_temperature` define the usable upper and lower boundaries for this particular component. At this point, adding additional required configuration parameters would result in a breaking change and...well...it seems we try to not do that. It sounds as if the documentation had called this out, you would have noticed, acted accordingly, and we probably wouldn't be here right now. 😄

Semi-related/other thoughts: I still don't see why one would want to set a value outside the range available on Home Assistant...which leads me wonder...what really is the point of `visual` `min_temperature` and `max_temperature`? 😆 In other words, why are they just "visual"? It occurs to me that (as you suggested) we should probably have a required `min_temperature` and `max_temperature` and then these are used to limit the set point _and_ define the boundaries used in the UI. I don't know of a commercial thermostat/climate device that, "if you just do this one special magic trick", allows you to choose a set point that's outside of the range it supports. I realize your specific use case may not have a UI, but, regardless, there still need to be limits defined so that the set point can be validated. I think what I'm trying to say is that, if the _actual_ range the device can be set to is different from the range the user sees, that just makes for a confusing user experience (and, as such, would contradict the direction HA is trying to go right now). Anyway, that's my two cents. 😄

closed 2021-02-13 by stale[bot]