esphome/issues #1508: Xiaomi LYWSD02 - last firmware update, changes format, add battery?

nikito7, 2020-09-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

lywsd02

**Description of problem:**
Last xiaomi firmware update breaks this
- no humidity
- 3???.? degress

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
[23:10:10][D][xiaomi_ble:273]: Got Xiaomi LYWSD02 (E7:2E:**:E5):
[23:10:10][D][xiaomi_ble:282]:   Battery Level: 25%
[23:17:14][D][xiaomi_ble:273]: Got Xiaomi LYWSD02 (E7:2E:**:E5):
[23:17:14][D][xiaomi_ble:276]:   Temperature: 3093.9°C
[23:21:07][I][app:105]: ESPHome version 1.16.0-dev compiled on Sep 23 2020, 23:20:03

```

**Additional information and things you've tried:**

nikito7, 2020-09-24:
After some study....

Just use xiaomi_cgg1 after firmware x.x.x.0085

Now we have battery level

I only have seen servicedata type: 0A, 06, 04.
No 0D!

Alex9779, 2020-10-26:
Which version of esphome are you using?
There has been a recent change in the dev which found its way to 0.15.3 fixing a regression and in #1580 there seems not problem getting temp and humidity with dev branch at that time where that fix was already in there...

KixMan28, 2020-11-14:
@nikito7 Thanks for the hint. I was looking forward to this battery level since I started using these LYWSD02 thermometers.

nikito7, 2020-11-14:
Fixed in Dev

...
  - platform: xiaomi_lywsd02
    mac_address: 
    temperature:
      name: "LYWSD02 Temperature"
    humidity:
      name: "LYWSD02 Humidity"
      accuracy_decimals: 0
    battery_level:
      name: "LYWSD02 Battery"
...

closed 2020-11-14 by nikito7

KixMan28, 2020-11-14:
@nikito7  If I use `xiaomi_lywsd02` and the batterz_level in the ESPHome config, I get an error:

```
 [battery_level] is an invalid option for [sensor.xiaomi_lywsd02]. Please check the indentation.
```
I have the latest ESPhome integration version 1.15.3.

Anz idea whz I get this error? 
With xiaomi_cgg1 it works fine.

nikito7, 2020-11-14:
1.16.0-dev

KixMan28, 2020-11-14:
When will be this feature added to the stable version?

nikito7, 2020-11-14:
Use cgg1 for now

KixMan28, 2020-11-14:
This is what I am doing. I am lookig forward to the newer version. 

Thank you for your support.