esphome/issues #1510: i cant upload in the app with python envirenment

aliakbar-bsr, 2020-09-25:
i was using esphome for a while and a few month ago it compiled but didnt connected to the wifi and i installed beta version and it solved 
i installed it again today but when i push upload button it shows this messege
{
PackageManager: Installing tool-esptool @ <2
Warning! Package Mirror: The 'sha1' sum 'e15fba5accf70d9c1574c47055ee9aedc0c3fe1c' of downloaded file 'tool-esptool-linux_armv6l-1.413.0.tar.gz' is not equal to remote 'bdfd4890e232df1876df63c8e676b12119e89c9f'
Looking for another mirror...
Error: Could not install 'tool-esptool' with version requirements '<2' for your system 'linux_armv7l'.

Please try this solution -> http://bit.ly/faq-package-manager
}
what do i have to do ??
i love your realy good job on esphome and i will be really happy with your answers for solving it

closed 2020-09-25 by close-issue-app[bot]

keesj, 2021-01-01:
Have a look here. it appears the mirror hosts the wrong file and if you download the correct file from https://sourceforge.net/projects/platformio-storage/files/packages/ and put it in $HOME/.platformio/packages/  the compilation can continue
https://github.com/platformio/platform-espressif8266/issues/230

keesj, 2021-01-02:
After performing pio upgrade and platformio upgrade the problem disappeared