esphome/issues #1511: Neopixelbus hangs ESP32Cam

Fusseldieb, 2020-09-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESP32Cam

**ESPHome version (latest production, beta, dev branch)**

1.15

**Affected component:**

neopixelbus

**Description of problem:**
Earlier I could use Neopixelbus and the ESP32Cam library at the same time and it would work just fine (This was in 1.14 I believe). Now in 1.15 it just hangs. When I flash it with Neopixelbus and the ESP32Cam enabled, it uploads and dies. It doesn't restart on its own anymore and I need to rip it from the location and reflash it.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: cozinha_led_teto
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "Oliver"
  password: "2440121oo7"
  manual_ip:
    static_ip: 192.168.2.59
    gateway: 192.168.2.1
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

light:
  - platform: neopixelbus
    pin: GPIO13
    num_leds: 50
    name: "Kitchen LED"
    id: kitchen_led

esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

  name: Kitchen Camera
  resolution: 640x480
  jpeg_quality: 20
  max_framerate: 3fps
  contrast: 1
  vertical_flip: false

```

**Logs (if applicable):**

```
[02:08:01]entry 0x400806ac
[02:08:02][I][logger:166]: Log initialized
[02:08:02][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[02:08:02][I][app:029]: Running through setup()...
[02:08:03][C][wifi:033]: Setting up WiFi...
[02:08:03][D][wifi:324]: Starting scan...
< Hangs forever, never reboots or prints anything >
```

**Additional information and things you've tried:**
I tried to use FastLED, but the lightstrip flickers like crazy, but this was alreadz an issue with older ESPHome version, that's why I chose Pixelbus, but now it's also broken.

BarrettLowe, 2020-09-29:
I have this same problem but without the camera. As soon as I put neopixelbus in, the whole thing hangs.

BarrettLowe, 2020-09-29:
I found #1481 which solved it for me. FYI

napieraj, 2020-10-01:
@Fusseldieb is this still an issue or resolved using the abovementioned link (to #1481)?

Fusseldieb, 2020-10-01:
> @Fusseldieb is this still an issue or resolved using the abovementioned link (to #1481)?

Switching to an older library of Neopixelbus like the mentioned post says fixes it. It's back online.

napieraj, 2020-10-01:
in that case I understand it the same issue and will close this to concentrate discussion in one place over the neopixel dependency issue #1481 - if you have further remarks feel free to reopen or create a new issue

closed 2020-10-01 by napieraj