esphome/issues #1513: No up-to-date i386 build

ehmke, 2020-09-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

esphome 

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

jesserockz, 2020-09-27:
i386 images were removed due to them not working recently.
https://github.com/esphome/esphome-docker-base/issues/5

napieraj, 2020-10-01:
closing this issue, the images have been removed, for more information and further discussion see #1386

closed 2020-10-01 by napieraj