esphome/issues #1515: Configuration colors not visible to include files

GHF, 2020-09-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

macOS 10.15.7 using Homebrew pip3

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 (TTGO T-Display board)

**ESPHome version (latest production, beta, dev branch)**

1.15.2 (through pip)

**Affected component:**

https://esphome.io/guides/configuration-types.html#color

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

jesserockz, 2020-09-28:
Yes, the header is included so that you can use variables and functions define inside it, but the other direction does not work.
If you want to use the color (or any other components), you will most likely need to pass it as one of the parameters to your defined function.
 
If you have many, you could create another function which sets pointers to them in the custom header file to be used elsewhere in the file.

GHF, 2020-10-08:
> If you want to use the color (or any other components)

I opened this bug because _only_ colors seem to be inaccessible from the custom header files, while other components _are_ accessible. For example, I could use `id(<font>)` and `id(<sensor>)` just fine in https://github.com/GHF/iaq-node/blob/dev/display.h

FWIW, the workaround you provided is reasonable (and I ended up just putting colors in the header itself), but this particular behavior unique to Color seemed unintentional.

jesserockz, 2020-10-08:
Ah ok, I was unaware that you could get other components via `id()` too, which does make this a bug then.

closed 2021-02-20 by stale[bot]