esphome/issues #1516: Custom UART text sensor is not presented to home-assistant.

iwwwwwwi, 2020-09-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

v1.16-dev

**Affected component:**

https://esphome.io/components/text_sensor/uart.html

**Description of problem:**
I'm trying to implement something like this via esphome: https://www.thingiverse.com/thing:4531305. Basically what I need is to send a string via uart and capture a response. I hooked up the modified KY033 module to my esp and flashed it with esphome. After adding this ESP to home-assistant via ESPHome integration I don't see any text_sensor on this ESP. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: dev1
  platform: ESP32
  board: esp32doit-devkit-v1
  includes:
    - uart_read_line_sensor.h
wifi:
  ssid: !secret ssid
  password: !secret wifi_password
  fast_connect: true
  manual_ip:
    static_ip: !secret dev_ip
    gateway: !secret gw_ip
    subnet: 255.255.255.0

# Enable logging
logger:
  level: VERBOSE #makes uart stream available in esphome logstream
  baud_rate: 0 #disable logging over uart

api:
  password: !secret api_pass
ota:
  password: !secret ota_pass

uart:
  id: uart_bus
  tx_pin: GPIO26
  rx_pin: GPIO27
  baud_rate: 300

text_sensor:
- platform: custom
  lambda: |-
    auto my_custom_sensor = new UartReadLineSensor(id(uart_bus));
    App.register_component(my_custom_sensor);
    return {my_custom_sensor};
  text_sensors:
    id: "uart_readline"
- platform: version
  name: "test_text"
  on_value:
    then:
      - lambda: |-
          ESP_LOGD("main", "The current version is %s", x.c_str());
          
switch:
  - platform: template
    name: "Switch"
    lambda: |-
      if (id(uart_readline).state == "*POW=ON#") {
        return true;
      } else if(id(uart_readline).state == "*POW=OFF#") {
        return false;
      } else {
        return {};
      }

binary_sensor:
  - platform: homeassistant
    name: "Input Boolean From Home Assistant"
    entity_id: input_boolean.uart_trigger
    on_press:
      then:
      - uart.write: '\r*pow=on#\r'
    on_release:
      then:
      - uart.write: '\r*pow=off#\r'

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[02:04:41][I][app:105]: ESPHome version 1.16.0-dev compiled on Sep 27 2020, 01:43:10
[02:04:41][C][wifi:443]: WiFi:
[02:04:41][C][wifi:303]:   SSID: [redacted]
[02:04:41][C][wifi:304]:   IP Address: [redacted]
[02:04:41][C][wifi:306]:   BSSID: [redacted]
[02:04:41][C][wifi:307]:   Hostname: 'dev1'
[02:04:41][C][wifi:311]:   Signal strength: -48 dB ▂▄▆█
[02:04:41][C][wifi:315]:   Channel: 6
[02:04:41][C][wifi:316]:   Subnet: 255.255.255.0
[02:04:41][C][wifi:317]:   Gateway: [redacted]
[02:04:41][C][wifi:318]:   DNS1: 0.0.0.0
[02:04:41][C][wifi:319]:   DNS2: 0.0.0.0
[02:04:41][C][uart_esp32:088]: UART Bus:
[02:04:41][C][uart_esp32:090]:   TX Pin: GPIO26
[02:04:41][C][uart_esp32:093]:   RX Pin: GPIO27
[02:04:41][C][uart_esp32:094]:   RX Buffer Size: 256
[02:04:41][C][uart_esp32:096]:   Baud Rate: 300 baud
[02:04:41][C][uart_esp32:097]:   Data Bits: 8
[02:04:41][C][uart_esp32:098]:   Parity: NONE
[02:04:41][C][uart_esp32:099]:   Stop bits: 1
[02:04:41][C][template.switch:058]: Template Switch 'Switch'
[02:04:41][C][template.switch:059]:   Restore State: NO
[02:04:41][C][template.switch:060]:   Optimistic: NO
[02:04:41][C][logger:185]: Logger:
[02:04:41][C][logger:186]:   Level: VERBOSE
[02:04:41][C][logger:187]:   Log Baud Rate: 0
[02:04:41][C][logger:188]:   Hardware UART: UART0
[02:04:41][C][version.text_sensor:021]: Version Text Sensor 'test_text'
[02:04:41][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[02:04:41][V][version.text_sensor:021]:   Unique ID: '3c71bf591730-version'
[02:04:41][C][ota:029]: Over-The-Air Updates:
[02:04:41][C][ota:030]:   Address: [redacted]
[02:04:41][C][ota:032]:   Using Password.
[02:04:41][C][api:095]: API Server:
[02:04:41][C][api:096]:   Address: [redacted]
[02:04:41][C][homeassistant.binary_sensor:032]: Homeassistant Binary Sensor 'Input Boolean From Home Assistant'
[02:04:41][C][homeassistant.binary_sensor:033]:   Entity ID: 'input_boolean.uart_trigger'
[02:05:02][D][homeassistant.binary_sensor:021]: 'input_boolean.uart_trigger': Got state ON
[02:05:02][D][binary_sensor:036]: 'Input Boolean From Home Assistant': Sending state ON
[02:05:11][D][homeassistant.binary_sensor:021]: 'input_boolean.uart_trigger': Got state OFF
[02:05:11][D][binary_sensor:036]: 'Input Boolean From Home Assistant': Sending state OFF
```

**Additional information and things you've tried:**
I added an additional text_sensor (using version platform) and it popped up in hass just fine.
I've also created a hass triggered binary sensor that triggers writes to uart to test if the example switch works. There is a reflective surface close to KY033 sensor and I see that the LED on it also flashes while I toggle my binary sensor in hass but the template binary switch from example doc never changes its state.
Besides standard init logs, there are no uart related logs in the log output while connecting to this esp from ESPHome WebUI.
Serial console doesn't show any uart related logs.

jesserockz, 2020-09-28:
Hi @iwwwwwwi. You need to add a `name` for components to show up in Home Assistant:
```
text_sensor:
- platform: custom
  lambda: |-
    auto my_custom_sensor = new UartReadLineSensor(id(uart_bus));
    App.register_component(my_custom_sensor);
    return {my_custom_sensor};
  text_sensors:
    id: "uart_readline"
    name: "your name here"
```

iwwwwwwi, 2020-09-28:
@jesserockz /facepalm - didn't realise that it's that simple :) . This however only partially fixes my situation - I see the sensor in Home Assistant now but it doesn't seem to update (constantly show "Unknown"). 
For debugging I directly connected GPIO26 with GPIO27 to ensure it's not related to my sensor, so now UART TX is directly connected to RX. Still nothing happens with neither the text_sensor nor with the template switch based on that text sensor while i trigger uart_write.

iwwwwwwi, 2020-09-28:
stupid me... it was a matter of single vs double quotes in my uart.write. The template based switch also didn't work because of case of the string. Thx again @jesserockz.

napieraj, 2020-10-01:
I am interpreting this as a resolved and closing the issue. If you have any further comments please reopen the issue or create a new one

closed 2020-10-01 by napieraj

AlenHodnik, 2021-06-08:
Hello.

I'm having same problem and my text sensor is still "unknown" even with changes made according to these posts. Any clue what could be the problem? 

Thanks. A.H.

mortenmoulder, 2021-08-17:
Having the same issue as @AlenHodnik - my code is identical to the OP, except I also added the `name: "your name here"` (and replaced the name). It just says "Unknown" in HA.

jesserockz, 2021-08-17:
> Having the same issue as @AlenHodnik - my code is identical to the OP, except I also added the `name: "your name here"` (and replaced the name). It just says "Unknown" in HA.

@AlenHodnik @mortenmoulder  Please open a new issue, but also know that custom code/sensors are not really supported.