esphome/issues #1519: PCD 8544 Display all pixels black

fr3nd, 2020-09-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

Version: 1.15.2

**Affected component:**

https://esphome.io/components/display/pcd8544.html

**Description of problem:**
I've connected the display to the board like this: http://acoptex.com/uploads/esp32nokia_1.jpg
Following [this tutorial](http://acoptex.com/project/1299/basics-project-072z-esp32-development-board-with-nokia-5110-graphical-lcd-module-at-acoptexcom/) the display works properly with backlight, but when trying to configure it with ESPHome, I get all the pixels in black.

However, if I disconnect the GND pin the screen shows the data properly but the backlight doesn't work.

**Problem-relevant YAML-configuration entries:**
```yaml
spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23

font:
  - file: 'pixelmix.ttf'
    id: font1
    size: 8

display:
  - platform: pcd8544
    cs_pin: GPIO5
    dc_pin: GPIO19
    reset_pin: GPIO14
    update_interval: 5s
    lambda: |-
      it.print(0, 0, id(font1), "Hello World!");

switch:
  - platform: gpio
    pin: 27
    name: "Nokia screen backlight"
```

**Logs (if applicable):**

Nothing relevant in the logs

**Additional information and things you've tried:**

rradar, 2020-10-02:
> but when trying to configure it with ESPHome, I get all the pixels in black.

Actually I had the exact same experience. I did try further after reproducing this on two different esp's and went ahead buying a second pcd 8544 display. :see_no_evil:

rradar, 2020-10-11:
It seems to be a problem with the contrast of the display a user in the esphome discord mentioned

Cyax, 2020-10-18:
Hello, is there any news about this problem or temporary solution ?

rradar, 2020-10-22:
@fr3nd wrote:
> Hello, is there any news about this problem or temporary solution ?

@Cyax wrote:
> However, if I disconnect the GND pin the screen shows the data properly but the backlight doesn't work.

rradar, 2020-10-24:
@fr3nd wrote:
> Hello, is there any news about this problem or temporary solution ?

@Cyax wrote:
> However, if I **disconnect the GND pin the screen shows the data properly** but the backlight doesn't work.

closed 2020-11-02 by jesserockz

zella, 2021-07-09:
Same issue. After text drawing display become black