esphome/issues #1520: Reading ADS1115 timed out

clumsyCoder00, 2020-09-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 mini and Wemos D1 mini lite

**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.15.0

**Affected component:**

https://esphome.io/components/sensor/ads1115.html

**Description of problem:**
After startup, logs flooded with "Reading ADS1115 timed out." Value for one or two of the four analog inputs is not returned. Restarting device, greater update_interval value have no effect.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
i2c:
  sda: 4
  scl: 5

ads1115:
  address: 0x48
  continuous_mode: true

sensor:
  - platform: ads1115
    multiplexer: 'A0_GND'
    update_interval: 2s
    gain: 6.144
    name: "ADS1115 Channel A0"
  - platform: ads1115
    multiplexer: 'A1_GND'
    update_interval: 2s
    gain: 6.144
    name: "ADS1115 Channel A1"
  - platform: ads1115
    multiplexer: 'A2_GND'
    update_interval: 2s
    gain: 6.144
    name: "ADS1115 Channel A2"
  - platform: ads1115
    multiplexer: 'A3_GND'
    update_interval: 2s
    gain: 6.144
    name: "ADS1115 Channel A3"
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[15:20:58][W][ads1115:116]: Reading ADS1115 timed out
[15:20:58][D][ads1115:163]: 'ADS1115 Channel A0': Got Voltage=0.714000V
[15:20:58][D][sensor:092]: 'ADS1115 Channel A0': Sending state 0.71400 V with 3 decimals of accuracy
[15:20:58][W][ads1115:116]: Reading ADS1115 timed out
[15:20:58][W][ads1115:116]: Reading ADS1115 timed out
[15:20:58][W][ads1115:116]: Reading ADS1115 timed out
[15:20:58][W][ads1115:116]: Reading ADS1115 timed out
[15:20:58][D][ads1115:163]: 'ADS1115 Channel A1': Got Voltage=0.726000V
[15:20:58][D][sensor:092]: 'ADS1115 Channel A1': Sending state 0.72600 V with 3 decimals of accuracy
[15:20:58][W][ads1115:116]: Reading ADS1115 timed out
[15:20:59][W][ads1115:116]: Reading ADS1115 timed out
[15:20:59][W][ads1115:116]: Reading ADS1115 timed out
[15:20:59][W][ads1115:116]: Reading ADS1115 timed out
[15:20:59][D][ads1115:163]: 'ADS1115 Channel A3': Got Voltage=0.717000V
[15:20:59][D][sensor:092]: 'ADS1115 Channel A3': Sending state 0.71700 V with 3 decimals of accuracy
[15:20:59][W][ads1115:116]: Reading ADS1115 timed out
[15:20:59][W][ads1115:116]: Reading ADS1115 timed out
[15:20:59][W][ads1115:116]: Reading ADS1115 timed out
[15:20:59][W][ads1115:116]: Reading ADS1115 timed out
[15:20:59][W][ads1115:116]: Reading ADS1115 timed out
[15:20:59][W][ads1115:116]: Reading ADS1115 timed out
[15:21:00][W][ads1115:116]: Reading ADS1115 timed out
```

**Additional information and things you've tried:**
Removing continuous_mode: true from the configuration resolves the issue, even when update_interval is decreased to 0.5s.

napieraj, 2020-10-02:
try continuous mode with only one channel, other users have reported issues using continuous mode with multiple channels at the same time

> I have 2 setup as differential inputs but cannot get them to be used at the same time. If I comment out either of the ADS1115 sensors the other will work. Not changing any wiring just commenting out code.

> v

napieraj, 2020-10-03:
related: #892

clumsyCoder00, 2020-11-19:
Thank you for getting back to me. You are correct, multiple inputs are my issue as well.

closed 2020-11-19 by clumsyCoder00