esphome/issues #1525: MAX7219 matrix cannot display more than 9 characters

jptrsn, 2020-10-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io running in docker on Ubuntu Server 20.02

**ESP (ESP32/ESP8266, Board/Sonoff):**

Tested on both Wemos D1 Mini (8266) and ESP32-Devkit-C (32)
**ESPHome version (latest production, beta, dev branch)**

v1.15.2
**Affected component:**

MAX7219 Digit Display
https://esphome.io/components/display/max7219digit.html

**Description of problem:**
I am attempting to display a value from a text_sensor onto a Max7219 digit display, if/when a new value is received. When not showing the text value, it should display the current time.

All of this functionality works as expected, but only up to 9 characters long. Once I reach 10 characters, the ESP will crash and reboot. Once it boots and conects to Home Assistant, if the sensor text hasn't changed, it will crash again and again.

**Problem-relevant YAML-configuration entries:**
```yaml

globals:
  - id: newMessage
    type: bool
    initial_value: '0'

spi:
  clk_pin: D5
  mosi_pin: D7

text_sensor:
  - platform: homeassistant
    entity_id: sensor.current_track
    id: track
    on_value:
      then:
        - lambda: |-
            id(newMessage) = 1;
        - delay: 5000ms
        - lambda: |-
            id(newMessage) = 0;

time:
  - platform: sntp
    id: sntp_time
    timezone: "America/Toronto"
    servers:
      - 0.ca.pool.ntp.org
      - 1.ca.pool.ntp.org
      - 2.ca.pool.ntp.org

display:
  - platform: max7219digit
    id: digit
    cs_pin: D8
    num_chips: 5
    intensity: 1
    scroll_speed: 100ms
    lambda: |-
      if (id(newMessage) == 1) {
        it.print(20, 0, id(pixel), TextAlign::TOP_CENTER, id(track).state.c_str());
        //it.printf(20, 0, id(pixel), TextAlign::TOP_CENTER, "%s", id(track).state.c_str())
      } else {
        it.strftime(20, -1, id(pixel), TextAlign::TOP_CENTER, "%H:%M", id(sntp_time).now());
      }
      
font:
  - file: "pixelmix.ttf"
    id: pixel
    size: 8
    glyphs: "0123456789:"

  - file: 'arial_narrow_7.ttf'
    id: arial
    size: 10

```

**Logs (if applicable):**

```
[11:37:39][C][logger:185]: Logger:
[11:37:39][C][logger:186]:   Level: DEBUG
[11:37:39][C][logger:187]:   Log Baud Rate: 115200
[11:37:39][C][logger:188]:   Hardware UART: UART0
[11:37:39][C][homeassistant.text_sensor:011]: Homeassistant Text Sensor 'track'
[11:37:39][C][homeassistant.text_sensor:012]:   Entity ID: 'sensor.current_track'
[11:37:39][C][max7219DIGIT:046]: MAX7219DIGIT:
[11:37:39][C][max7219DIGIT:047]:   Number of Chips: 5
[11:37:39][C][max7219DIGIT:048]:   Intensity: 1
[11:37:39][C][max7219DIGIT:049]:   Scroll Mode: 0
[11:37:39][C][max7219DIGIT:050]:   Scroll Speed: 100
[11:37:39][C][max7219DIGIT:051]:   Scroll Dwell: 1000
[11:37:39][C][max7219DIGIT:052]:   Scroll Delay: 1000
[11:37:39][C][max7219DIGIT:053]:   CS Pin: GPIO12 (Mode: OUTPUT)
[11:37:39][C][max7219DIGIT:054]:   Update Interval: 0.5s
[11:37:39][C][captive_portal:169]: Captive Portal:
[11:37:39][C][ota:029]: Over-The-Air Updates:
[11:37:39][C][ota:030]:   Address: max_display_32.local:3232
[11:37:39][C][api:095]: API Server:
[11:37:39][C][api:096]:   Address: max_display_32.local:6053
[11:37:39][C][sntp:039]: SNTP Time:
[11:37:39][C][sntp:040]:   Server 1: '0.ca.pool.ntp.org'
[11:37:39][C][sntp:041]:   Server 2: '1.ca.pool.ntp.org'
[11:37:39][C][sntp:042]:   Server 3: '2.ca.pool.ntp.org'
[11:37:39][C][sntp:043]:   Timezone: 'EST5EDT4,M3.2.0/2,M11.1.0/2'
[11:37:39][D][api.connection:604]: Client 'Home Assistant 0.115.6 (192.168.1.13)' connected successfully!
[11:37:40][D][homeassistant.text_sensor:016]: 'sensor.current_track': Got state '1234567890'
[11:37:40][D][text_sensor:015]: 'track': Sending state '1234567890'
[11:37:40]abort() was called at PC 0x40152a5f on core 1
[11:37:40]
[11:37:40]Backtrace: 0x4008c8b0:0x3ffb1c80 0x4008cae1:0x3ffb1ca0 0x40152a5f:0x3ffb1cc0 0x40152aa6:0x3ffb1ce0 0x40151b37:0x3ffb1d00 0x40151c4e:0x3ffb1d20 0x400d7686:0x3ffb1d40 0x400d76d7:0x3ffb1d70 0x400d77bb:0x3ffb1d90 0x400d780f:0x3ffb1db0 0x400d6764:0x3ffb1dd0 0x400d6a07:0x3ffb1df0 0x400d6a43:0x3ffb1e60 0x400de2a3:0x3ffb1e90 0x400d7849:0x3ffb1ef0 0x401604c9:0x3ffb1f10 0x400810d7:0x3ffb1f30 0x400dbb11:0x3ffb1f60 0x400de4ca:0x3ffb1f90 0x400eabf9:0x3ffb1fb0 0x40088ff9:0x3ffb1fd0
[11:37:40]
[11:37:40]Rebooting...
[11:37:40]ets Jun  8 2016 00:22:57
[11:37:40]
[11:37:40]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[11:37:40]configsip: 0, SPIWP:0xee
[11:37:40]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[11:37:40]mode:DIO, clock div:2
[11:37:40]load:0x3fff0018,len:4
[11:37:40]load:0x3fff001c,len:1044
[11:37:40]load:0x40078000,len:8896
[11:37:40]load:0x40080400,len:5828
[11:37:40]entry 0x400806ac
```

**Additional information and things you've tried:**
I've tried both Wemos D1 Mini and an ESP32 to check if that would help. I have modified how the lambda function is rendering the text (e.g. `it.print` and `it.printf`), different fonts, etc. I'm not sure what's causing the issue, hopefully the logs will help.

napieraj, 2020-10-01:
cc @rspaargaren

rspaargaren, 2020-10-01:
Let me check upon your code example if I see the same behaviour and of course can fix it!🤞

closed 2021-02-12 by stale[bot]