esphome/issues #1526: light.on_turn_off / light .on_turn_on firing inappropriately or binary light not working correctly

justlikeef, 2020-10-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP12-F

**ESPHome version (latest production, beta, dev branch)**
dev

**Affected component:**
light

**Description of problem:**
If the light is on, calling turn_on with a brightness of 100% or 10% causes the locatorled to turn on

**Problem-relevant YAML-configuration entries:**
```yaml
        - light.turn_on:
            id: light1
            brightness: 100%

        - light.turn_on:
            id: light1
            brightness: 10%

light:
  - platform: binary
    name: "Locator LED"
    output: locatorledpin
    id: locatorled
    internal: true
  - platform: monochromatic
    id: light1
    output: dimmer1
    name: Dimmerized Light
    on_turn_on:
      - light.turn_off: locatorled
    on_turn_off:
      - light.turn_on: locatorled

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[17:43:48][D][light:264]: 'Dimmerized Light' Setting:
[17:43:48][D][light:277]:   Brightness: 100%
[17:43:48][D][light:303]:   Transition Length: 1.0s
[17:43:48][D][light:264]: 'Locator LED' Setting:
[17:43:48][D][light:273]:   State: OFF
[17:43:55][D][light:264]: 'Dimmerized Light' Setting:
[17:43:55][D][light:277]:   Brightness: 10%
[17:43:55][D][light:303]:   Transition Length: 1.0s
[locatorled is actually on]

**Additional information and things you've tried:**

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

j2mc, 2020-10-02:
I'm also experiencing erratic behavior with the on_turn_on triggers.  Using latest production, it seems on_turn_on gets triggered at the following event from when it is supposed to be called. eg:
Turn light on, then change brightness or turn light off. On_turn_on seems to always trigger on that second action no matter what it is.

LuboSta, 2020-10-02:
having same issue with  5 of mine sonoff switches ( T1 and T2 ) there was no issue up til latest update ( jum from ver 1.14.x to 1.15.3   so not sure where in 1.15 it started ).  I had no extra config before. I  had tried to add filters   - delayed_on: 10ms   - delayed_off: 10ms on binary sensor , but doesnt help random "on" ( seams random off is not happening ) still happening.

LuboSta, 2020-10-02:
correction ... it also randomly turn off :)

MaartenKleijwegt, 2020-10-05:
I might have duplicated this issue with #1534 

I'm also seeing very erratic behaviour in the on_turn_on logic. In my case the on_turn_on and on_turn_off seem reversed. 

Maybe not related but I've also noticed when replacing on_turn_on with a script to call that if I checked if the light was on at the beginning of the script, but after I changed the light, that it would report the state before the state change. After inserting a delay of 2s before the check inside the script. this was fixed. Could there be a change in the code that shifted the time of the value being changed and the actions being called?

MaartenKleijwegt, 2020-10-05:
Could this also be related to #1521?

justlikeef, 2020-10-08:
Binary light gets out of sync with what is reported through mqtt to HA.  This seems to be a binary light issue.

closed 2020-10-11 by glmnet