esphome/issues #1528: ESP32Cam 160° Lens - `horizontal_mirror` doesn't work

Fusseldieb, 2020-10-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant on Docker using Ubuntu (irrelevant)

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32Cam + 160° Lens

**ESPHome version (latest production, beta, dev branch)**

1.15

**Affected component:**

esp32_camera

**Description of problem:**
If I use the 160° camera variant found on AliExpress, the horizontal_mirror doesn't work, only vertical_flip.

**Problem-relevant YAML-configuration entries:**
```yaml
horizontal_mirror: true

```

**Logs (if applicable):**

```
-
```

**Additional information and things you've tried:**

Fusseldieb, 2021-01-31:
Was this fixed at some point?

Fusseldieb, 2021-06-02:
Don't stale

nepozs, 2021-06-14:
It looks like "160° Lens" version simply has different hardware camera, so it can have different capability.

NopeNix, 2022-06-16:
+1

my esp32-cam is capable of flipping (checked with an example scetch) but the 
```yml
horizontal_mirror: true
```
does also not work for me. 

i'm using the [following ](https://www.az-delivery.de/en/products/esp32-cam-modul-esp32-wifi-bluetooth-modul-inklusive-kamera)esp32-cam

adhityan26, 2022-06-27:
any updates regarding this issue?
also using  ```horizontal_mirror: true```
bot the image is still using default orientation

DooMMasteR, 2024-08-08:
`true` is the default, if you want it flipped you need to use 
```yaml
horizontal_mirror: false
```
@adhityan26 @NopeNix @Fusseldieb