esphome/issues #1529: Shelly 2.5 with ADE7653

djwmarcx, 2020-10-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

Shelly 2.5

**ESPHome version (latest production, beta, dev branch)**

v1.15.2

**Affected component:**

ade7953

**Description of problem:**
The device is getting hot. The internal NTC is measuring 76ºC. It's also hot to the touch.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

jesserockz, 2020-10-04:
A possible solution to this could be to add an Optional GPIO input pin to the `ade7953` config and code which is for the IRQ pin.

djwmarcx, 2020-10-05:
Great. I'll check it!

closed 2020-11-15 by jesserockz