esphome/issues #1530: Unable to compile the code

hussamaldean, 2020-10-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HomeAssistant on Raspberry Pi 3B

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-WROOM and ESP8266-nodeMC
**ESPHome version (latest production, beta, dev branch)**

V.1.15.2 (latest)
**Affected component:**

compiling the code

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32test
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "TP-Link_9DFF"
  password: "58054093"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32Test Fallback Hotspot"
    password: "vAu0Uq1tGPCP"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/esp32test.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/esp32test
Processing esp32test (board: esp-wrover-kit; framework: arduino; platform: espressif32@1.12.4)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Looking for AsyncTCP-esphome library in registry
APIRequestError: [API] Invalid response: b'<HTML></HTML>':
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/main.py", line 168:
    env.SConscript("$BUILD_SCRIPT")
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/root/.platformio/platforms/espressif32/builder/main.py", line 223:
    target_elf = env.BuildProgram()
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/platformio.py", line 62:
    env.ProcessProjectDeps()
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/platformio.py", line 131:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/piolib.py", line 1036:
    project.install_dependencies()
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/piolib.py", line 868:
    lm.install(uri)
  File "/usr/local/lib/python3.6/dist-packages/platformio/managers/lib.py", line 272:
    interactive=interactive,
  File "/usr/local/lib/python3.6/dist-packages/platformio/managers/lib.py", line 165:
    "/v2/lib/search", dict(query=" ".join(query)), cache_valid="1h"
  File "/usr/local/lib/python3.6/dist-packages/platformio/util.py", line 345:
    result = _get_api_result(url, params, data)
  File "/usr/local/lib/python3.6/dist-packages/platformio/util.py", line 81:
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/platformio/util.py", line 319:
    raise exception.APIRequestError("Invalid response: %s" % r.text.encode("utf-8"))
========================= [FAILED] Took 44.85 seconds =========================
```

**Additional information and things you've tried:**

It was working just fine then suddenly this happen. I tried to reinstall HA and everything else and still same issue

[image]

jesserockz, 2020-10-05:
This looks like either a networking issue or a server side issue on platformio.

hussamaldean, 2020-10-06:
> This looks like either a networking issue or a server side issue on platformio.
Hi,
It worked once and it didn't work after that 
what I did is i restarted HA and worked, now it doesn't 
I tried to reset the HA and raspberry pi and same results

closed 2021-02-17 by stale[bot]