esphome/issues #1531: RGBW color interlock

notherealmarco, 2020-10-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
esp8266 nodemcu

**ESPHome version (latest production, beta, dev branch)**
1.15.2

**Affected component:**
RGBW Light

**Description of problem:**
The new RGBW update isn't working good for me. I used to get a variable temperature light by mixing the white (which is a warm white) with the RGB colors. With the new update, i cannot do anymore things like: white led turned on to 100% and RGB leds to 30%. The new color interlock feature isn't helpful as it doesn't allow white and rgb to be mixed together.
I had to roll back to 1.14.

**Problem-relevant YAML-configuration entries:**
```yaml
light:
  - platform: rgbw
    name: "mrc_lampada"
    red: pin_1
    green: pin_2
    blue: pin_5
    white: pin_7
    gamma_correct: 2.8
    default_transition_length: 2s
```

**Logs (if applicable):**

```
not needed

```

**Additional information and things you've tried:**

napieraj, 2020-10-04:
duplicate of #1475, closing

tduffy83, 2020-10-08:
This is not a duplicate of 1475.  This person needs to set color_interlock to false and they'll get expected behavior.

napieraj, 2020-10-10:
> This is not a duplicate of 1475.  This person needs to set color_interlock to false and they'll get expected behavior.

@notherealmarco can you confirm? if the above does not provide the expected behavior, please reopen the issue

nofuturekid, 2020-10-10:
At least I can confirm, because it also affects all of my RGBW strips.
In #1475#issuecomment-706395151 @margau already prepared a fix, I think. It was that "color_interlock = false" seems to no be honored.

So it's an issue and should be re-opened.

closed 2020-10-15 by glmnet

notherealmarco, 2020-10-25:
Issue hasn't been fixed in 0.115.3
I'm still not available to control RGB and White indipendently.

nofuturekid, 2020-10-25:
For me it is working. So maybe you should paste the relevant configs.

peq123, 2020-10-26:
@notherealmarco hi there, I wrote the code for the interlock. Could you post your config so I could have a look

notherealmarco, 2020-10-29:
My config is
`light:
  - platform: rgbw
    name: "mrc_lampada"
    red: pin_1
    green: pin_2
    blue: pin_5
    white: pin_7
    color_interlock: false
    gamma_correct: 2.8
    default_transition_length: 2s
    effects:
      - random:
      - flicker:`

I figured out that the issue may be in Home Assistant only, not in the ESPHome firmware.
Calling the light.turn_on service with:
`entity_id: light.mrc_lampada_2
rgb_color: [0,0,0]
white_value: 255
brightness: 255`
This sets all the channels to 255 (even if RGB color is 0,0,0).

If I send the same command from MQTT (`{"state":"ON", "color":{"r":0,"g":0,"b":0},"white_value":255,"brightness":255}`) it works good, only the white channel turns on.

tduffy83, 2020-12-01:
I can confirm color_interlock: false works as intended with 1.15.3.  The controls are still weird when color_interlock: true per #1475.

regoras, 2021-01-26:
Echoing @tduffy83 that color_interlock: false works in 1.15.3 for effects such as random and flicker to work. Unfortunately with color_interlock: false, white temperatures get thrown off, as it looks like RGB LEDs are still on in white mode.