esphome/issues #1535: Teckin SP20 relay not responsive to button or api. Relay toggles when hwl8012 changes modes

theiam79, 2020-10-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker on Windows 10

**ESP (ESP32/ESP8266, Board/Sonoff):**

Teckin SP20 (ESP8266 )

**ESPHome version (latest production, beta, dev branch)**

v1.15.2

**Affected component:**

https://esphome.io/components/switch/gpio.html

**Description of problem:**
Plug seems to respond to button presses or relay toggles from HA through the API, but the relay doesn't actually turn on or off. On top of that, when hlw8012 changes modes (ie `Changing mode to CURRENT mode`) the relay toggles.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: ${device_name}
  comment: ${device_description}
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_not_ssid
  password: !secret wifi_not_pass
  ap:
    ssid: ${friendly_name}_AP
    password: !secret wifi_not_pass

web_server:
  port: 80

debug:

logger:
  level: VERBOSE

# Enable Home Assistant API
api:
  password: !secret esp_api_pass

ota:
  password: !secret esp_ota_pass

time:
  - platform: homeassistant
    id: homeassistant_time

sensor:
  # Power measuring sensor
  - platform: hlw8012
    sel_pin:
      number: GPIO12
      inverted: True
    cf_pin: GPIO5
    cf1_pin: GPIO14
    current:
      name: ${friendly_name} Current
    power:
      name: ${friendly_name} Power
      id: power
      filters:
        # Some calibration
        - multiply: 0.4545
[... 189 lines cut ...]
[17:55:25][V][sensor:013]: 'SP20 1 Total Daily Energy': Received new state 0.461456
[17:55:25][D][sensor:092]: 'SP20 1 Total Daily Energy': Sending state 0.00046 kWh with 3 decimals of accuracy
[17:55:28][D][hlw8012:071]: Got power=176.2W, voltage=122.2V
[17:55:28][V][sensor:013]: 'SP20 1 Voltage': Received new state 122.160477
[17:55:28][D][sensor:092]: 'SP20 1 Voltage': Sending state 122.16048 V with 1 decimals of accuracy
[17:55:28][V][sensor:013]: 'SP20 1 Power': Received new state 176.162582
[17:55:28][D][sensor:092]: 'SP20 1 Power': Sending state 80.06590 W with 1 decimals of accuracy
[17:55:28][V][sensor:013]: 'SP20 1 Total Daily Energy': Received new state 0.528177
[17:55:28][D][sensor:092]: 'SP20 1 Total Daily Energy': Sending state 0.00053 kWh with 3 decimals of accuracy
[17:55:31][D][hlw8012:071]: Got power=177.5W, voltage=122.2V
[17:55:31][V][sensor:013]: 'SP20 1 Voltage': Received new state 122.160477
[17:55:31][D][sensor:092]: 'SP20 1 Voltage': Sending state 122.16048 V with 1 decimals of accuracy
[17:55:31][V][sensor:013]: 'SP20 1 Power': Received new state 177.497131
[17:55:31][D][sensor:092]: 'SP20 1 Power': Sending state 80.67245 W with 1 decimals of accuracy
[17:55:31][V][sensor:013]: 'SP20 1 Total Daily Energy': Received new state 0.595404
[17:55:31][D][sensor:092]: 'SP20 1 Total Daily Energy': Sending state 0.00060 kWh with 3 decimals of accuracy
[17:55:31][V][hlw8012:084]: Changing mode to CURRENT mode
[17:55:34][V][sensor:013]: 'SP20 1 Power': Received new state 6.672825
[17:55:34][D][sensor:092]: 'SP20 1 Power': Sending state 3.03280 W with 1 decimals of accuracy
[17:55:34][V][sensor:013]: 'SP20 1 Total Daily Energy': Received new state 0.597916
[17:55:34][D][sensor:092]: 'SP20 1 Total Daily Energy': Sending state 0.00060 kWh with 3 decimals of accuracy
```

**Additional information and things you've tried:**
Tried a few different configs but they all have the same issue. If I remove the sensor config the relay stops toggling since there aren't any mode changes, but the button/api still don't actually toggle the relay.

MaartenKleijwegt, 2020-10-05:
https://frenck.dev/calibrating-an-esphome-flashed-power-plug/ Frenck has similar plugs in which GPIO14 and GPIO 4 are switched. Maybe they changed pin layout? If what you say is true and the relay switches with mode changes, maybe this one has the relay tied to gpio12?

MaartenKleijwegt, 2020-10-05:
To add to my comment:
    someone has added some comments to the blakadder page for this socket: https://templates.blakadder.com/teckin_SP20.html
To summarize:

   he bought a similar looking device exhibiting the same behaviour as yours. He found a similar looking model named the Wage
https://templates.blakadder.com/waga_CHCZ02MB.html

short of it your pin layout is probably:
   GPIO #	Component
GPIO00	Led2i
GPIO01	None
GPIO02	None
GPIO03	HLWBL SELi
GPIO04	None
GPIO05	BL0937 CF
GPIO09	None
GPIO10	None
GPIO12	Relay1
GPIO13	Button1
GPIO14	HLWBL CF1
GPIO15	Led1i
GPIO16	None
FLAG	None

theiam79, 2020-10-05:
I had a hunch it was a different pinout, but I wasn't sure how to go about determining it without opening the plug up. The pinout you linked in your second response worked like a charm and now everything works as expected. Thank you very much!

closed 2020-10-05 by theiam79