esphome/issues #1536: Sensor "on_value_range" actions not triggered when temperature is above threshold. It is only triggered when temperature moves from below to above thresold

SensorsIot, 2020-10-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant add-on in a container

**ESP (ESP32/ESP8266, Board/Sonoff):**
Wemos Mini ESP8266

**ESPHome version (latest production, beta, dev branch)**
1.15.2

**Affected component:**
 Automation on_value_range

**Description of problem:**
I want to measure the temperature of an amplifier and switch it off if it is higher than e.g. 40 degrees. This works fine if the amplifier heats up from below 40 degrees. If I (unintended) switch it on while above 40 degrees it is no more protected because ESPhome does not switch it off anymore.
The doc says: "In the first example, this range is defined as “any value above or including 65.0”

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

esphome:
  name: qo_100
  platform: ESP8266
  board: nodemcuv2
  
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
      inverted: True
    id: button_1
    on_press:
      then:
        - light.toggle: power_1

  - platform: status
    name: "My LS Status"

output:
  - platform: gpio
    pin: D7
    id: relay_1

light:
  - platform: binary
    name: "QO-100 Main"
    id: power_1
    output: relay_1
    
i2c:
  sda: D2
  scl: D1
  scan: True
  id: bus_a
    
sensor:
  - platform: hdc1080
    temperature:
      name: "QO-100 PA"
      on_value_range:
        - above: 40.0
          then:
            - light.turn_off: power_1
    humidity:
      name: "QO-100 Box"
    update_interval: 10s

**Logs (if applicable):**
<!--
Please copy the debug log here. If possible, also connect to the ESP over USB and copy those logs into the backticks.
-->
[18:40:09][D][sensor:092]: 'QO-100 PA': Sending state 44.37255 °C with 1 decimals of accuracy
[18:40:09][D][sensor:092]: 'QO-100 Box': Sending state 48.65723 % with 0 decimals of accuracy
[18:40:09][D][hdc1080:054]: Got temperature=44.4°C humidity=48.7%
[18:40:19][D][sensor:092]: 'QO-100 PA': Sending state 44.34234 °C with 1 decimals of accuracy
[18:40:19][D][sensor:092]: 'QO-100 Box': Sending state 48.65723 % with 0 decimals of accuracy
[18:40:19][D][hdc1080:054]: Got temperature=44.3°C humidity=48.7%
```
PASTE DEBUG LOG HERE

**Additional information and things you've tried:**

I found a similar issue which was closed without a solution.

glmnet, 2020-10-06:
This is expected behavior.
Per the `on_value_range` [docs ](https://esphome.io/components/sensor/index.html#on-value-range)

> This trigger will only trigger when the new value is inside the range and the previous value was outside the range

for your particular request I would replace the `light.toggle` action with a construction which checks the temperature before even trying to turn on the device.

Hope this helps

SensorsIot, 2020-10-06:
I added the feature in the switch (if somebody else has the same problem):
switch:
  - platform: output
    name: main
    id: main
    output: relay_1
    on_turn_on:
      - if:
          condition:
            sensor.in_range:
              id: pa
              above: 40.0
          then:
            - switch.turn_off: main

closed 2020-10-06 by SensorsIot