esphome/issues #1538: Actions after delay don't seem to be working

docmattman, 2020-10-06:
**Operating environment/Installation: Docker**

**Wimos D1 Mini**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**

I have a script that I'm calling to turn on a relay, then delay for `1s`, then turn it off.  The turn on works fine.  But many times, it doesn't turn off.  So it seems like the turn off command after the delay doesn't work consistently.  Sometimes it works fine, but often times, it just turns on and never turns off.

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:
  - platform: gpio
    id: ${deviceid}_btn
    pin:
      number: 12
      inverted: True
      mode: INPUT_PULLUP
    name: ${devicelabel} Button
    on_state:
      then:
      - if:
          condition:
            binary_sensor.is_on: ${deviceid}_btn
          then:
            - script.execute: launch

  - platform: template
    id: ${deviceid}_working_sensor
    name: ${devicelabel} Working
    device_class: moving
    lambda: |-
      if (id(is_working)) {
        return 1;
      } else {
        return 0;
      }
    
script:
  - id: launch
    then:
      - lambda: |-
          id(is_working) = 1;
          id(${deviceid}_working_sensor).publish_state(id(is_working));
      - switch.turn_on: ${deviceid}_switch_relay
      - delay: 1s
      - switch.turn_off: ${deviceid}_switch_relay

```

**Logs (if applicable):**

```

```

**Additional information and things you've tried:**

Not sure what else to try.  It works properly sometimes.  But most of the time, the `- switch.turn_off: ${deviceid}_switch_relay` doesn't seem to be run and the relay stays on.

glmnet, 2020-10-07:
Please provide all required information

docmattman, 2020-10-08:
Updated the issue to fix the info/template.

danobot, 2020-10-17:
I am facing an issue where `delay` of `60s` works fine but `15min` causes the device to reset.

works fine repeatedly:
```
interval:
  - interval: 6h
    then:
      - light.turn_on: uv
      - delay: 60s
      - light.turn_off: uv
```

Resets:
```
interval:
  - interval: 6h
    then:
      - light.turn_on: uv
      - delay: 15min
      - light.turn_off: uv
```

glmnet, 2020-10-17:
@docmattman please provide the full yaml for this issue. Also this issue must include the logs, you can use online logs now and that will tell us more 
Also you can add log actions to your code so you know if the action is running or not (which will indicate more precisely where the issue is)

@danobot This could be related to the scheduler issue I fixed a few days ago. Could you help testing the dev branch already contains the fix.

docmattman, 2020-10-18:
@glmnet What log settings do I need to turn on to get the info you need?  Here is the full yaml file I'm using:

```
substitutions:
  deviceid: halloween_candy
  devicelabel: "Halloween Candy"
  
esphome:
  name: ${deviceid}
  platform: ESP8266
  board: esp01_1m
  
wifi:
  ssid: "XXXXXXXXXXX"
  password: "XXXXXXXXXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${devicelabel} Hotspot"
    password: "abcde12345"

# Enable Home Assistant API
# api:

captive_portal:

# Enable logging
logger: 
  # level: ERROR
  # logs:
  #   switch: ERROR
  #   sensor: ERROR
    
mqtt:
  broker: thingtank  

ota:

# web_server:
#   port: 80
        
globals:
  - id: is_working
    type: float
    restore_value: no
    initial_value: '0'

switch:
  - platform: restart
    id: ${deviceid}_switch_restart
    name: "${devicelabel}: Restart switch"
    icon: "mdi:restart"
    
  - platform: gpio
    id: ${deviceid}_switch_relay
    name: "${devicelabel} Relay"
    pin: 13
    

binary_sensor:
  - platform: gpio
    id: ${deviceid}_btn
    pin:
      number: 12
      inverted: True
      mode: INPUT_PULLUP
    name: ${devicelabel} Button
    on_state:
      then:
      - if:
          condition:
            binary_sensor.is_on: ${deviceid}_btn
          then:
            - script.execute: launch

  - platform: template
    id: ${deviceid}_working_sensor
    name: ${devicelabel} Working
    device_class: moving
    lambda: |-
      if (id(is_working)) {
        return 1;
      } else {
        return 0;
      }
    
script:
  - id: launch
    then:
      - lambda: |-
          id(is_working) = 1;
          id(${deviceid}_working_sensor).publish_state(id(is_working));
      - logger.log:
            format: "Turning on relay %.1f"
            args: [ '""' ]
      - switch.turn_on: ${deviceid}_switch_relay
      - delay: 1s
      - logger.log:
            format: "Turning off relay %.1f"
            args: [ '"!"' ]
      - script.execute: stop_belt

  - id: stop_belt
    then:
      - lambda: |-
          id(is_working) = 0;
          id(${deviceid}_working_sensor).publish_state(id(is_working));
      - switch.turn_off: ${deviceid}_switch_relay
```

glmnet, 2020-10-18:
Set log level verbose and copy logs here, when issue happens of course.

danobot, 2020-10-19:
@glmnet You can remove my comments as they turned out to be unrelated to this issue. Reboots were caused by `reboot_timeout` in `api` component.

closed 2021-03-02 by stale[bot]