esphome/issues #1542: ESP32 : GPIO Switch does not work after using LEDC on that GPIO (via OTA)

Guyohms, 2020-10-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant 0.116.1 (Hass.io)

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 (ESP-WROOM-32)

**ESPHome version (latest production, beta, dev branch)**

1.15.2

**Affected component:**

https://esphome.io/components/output/ledc.html
https://esphome.io/components/switch/gpio.html

**Description of problem:**
If a GPIO is configured as a LEDC output, you cannot use it after as a GPIO switch unless you reflash the ESP via USB (erasing the whole flash).
The switch is created, shows in Home Assistant can be activated as usual but the GPIO always stays LOW.

**Problem-relevant YAML-configuration entries:**
```yaml

# First upload
output:
  - platform: ledc
    pin: GPIO25
    id: cyl_EN
    frequency: 500Hz

# Second upload
switch:
  - platform: gpio
    name: "Output ENABLE"
    id: cyl_ENABLE
    pin:
      number: GPIO25

```

**Logs (if applicable):**

No error related in log

**Additional information and things you've tried:**

I tried cleaning the build files, recompiling and uploading via OTA : Did not work
I uploaded back with the LEDC output: It worked
I uploaded again with the switch component : Did not work
I flashed it through esphome flasher : It worked right away.

My guess is that the LEDC component sets something in the hardware configuration registers related to this GPIO that is not cleared on a OTA upload leaving the GPIO in "PWM" mode with 0% duty cycle.

stanislav-brabec, 2020-11-24:
See #619 for more.

It seems that the damage could be permanent. You can try to upload fixed sketch. Maybe it will fix the internal hardware registers, and camera will start to work again.

closed 2021-04-07 by stale[bot]

lucius-the, 2021-10-03:
`My guess is that the LEDC component sets something in the hardware configuration registers related to this GPIO that is not cleared on a OTA upload leaving the GPIO in "PWM" mode with 0% duty cycle.`

Had the same behavior, using a small LED, light component, output was gpio, I then tried playing around with ledc (which worked nicely), but then it wouldn't work again with gpio. Until I found out I had to flash via USB to resolve it.
This saved me from pulling my hair, thank you !