esphome/issues #1545: Unexpected warning of automation

hktkzyx, 2020-10-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Item|Detail
--|--
arch | aarch64
chassis | embedded
dev | false
docker | true
docker_version | 19.03.11
hassio | true
host_os | HassOS 4.13
installation_type | Home Assistant OS
os_name | Linux
os_version | 4.19.127-v8
python_version | 3.8.5
supervisor | 247
timezone | Asia/Shanghai
version | 0.116.2
virtualenv | false

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP12-F

**ESPHome version (latest production, beta, dev branch)**

v1.15.2

**Description of problem:**
I would like to use an automation to subscribe a topic from mqtt and pulish a topic to activate the deep sleep mode of esp8266 to save power. The sensor measures the temperature and humidity and wake up 10s every 5min. Since the mode of automation is single, the action is expected to be finished in a very short time. However, I get the "Already running" warnings.
I would like to why I still get "Already running" warning and what is the mechanism of the automation mode.

**Problem-relevant YAML-configuration entries:**
```yaml
- id: '1602421903084'
  alias: temperature_humidity_sensor_01_sleep
  description: ''
  trigger:
  - platform: mqtt
    topic: temperature_humidity_sensor_01/sensor/living_room_humidity_01/state
  condition: []
  action:
  - delay: 00:00:05
  - service: mqtt.publish
    data:
      topic: temperature_humidity_sensor_01/deep_sleep
      payload: 'on'
  - delay: 00:00:01
  mode: single
```
```yaml
esphome:
  name: temperature_humidity_sensor_01
  platform: ESP8266
  board: esp12e

wifi:
  ssid: "xxx"
  password: "xxx"
  manual_ip:
    static_ip: 192.168.86.10
    gateway: 192.168.86.1
    subnet: 255.255.255.0
  fast_connect: true

captive_portal:

# Enable logging
logger:
  level: DEBUG

ota:

sensor:
  - platform: dht
    pin: GPIO4
    temperature:
      name: "Living Room Temperature 01"
    humidity:
      name: "Living Room Humidity 01"
    update_interval: 10s
    model: AM2302

deep_sleep:
  run_duration: 10s
  sleep_duration: 5min
  id: deep_sleep_01

mqtt:
  broker: 192.168.86.xxx
  username: xxx
  password: xxx
  birth_message:
  will_message:
```

**Logs (if applicable):**

```
2020-10-11 21:29:47 DEBUG (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] Loaded automation automation.temperature_humidity_sensor_01_sleep with state True from state  storage last state <state automation.temperature_humidity_sensor_01_sleep=on; last_triggered=2020-10-11T13:29:39.952192+00:00, mode=single, current=0, id=1602421903084, friendly_name=temperature_humidity_sensor_01_sleep @ 2020-10-11T21:29:39.952341+08:00>
2020-10-11 21:29:47 INFO (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Running automation actions
2020-10-11 21:29:47 INFO (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Executing step delay 0:00:05
2020-10-11 21:29:47 INFO (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] Initialized trigger temperature_humidity_sensor_01_sleep
2020-10-11 21:29:52 INFO (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Executing step call service
2020-10-11 21:29:52 INFO (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Executing step delay 0:00:01
2020-10-11 21:33:24 INFO (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Running automation actions
2020-10-11 21:33:24 INFO (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Executing step delay 0:00:05
2020-10-11 21:33:25 WARNING (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Already running
2020-10-11 21:33:29 INFO (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Executing step call service
2020-10-11 21:33:29 INFO (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Executing step delay 0:00:01
2020-10-11 21:38:28 INFO (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Running automation actions
2020-10-11 21:38:28 INFO (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Executing step delay 0:00:05
2020-10-11 21:38:30 WARNING (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Already running
2020-10-11 21:38:33 INFO (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Executing step call service
2020-10-11 21:38:33 INFO (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Executing step delay 0:00:01
2020-10-11 21:43:31 INFO (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Running automation actions
2020-10-11 21:43:31 INFO (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Executing step delay 0:00:05
2020-10-11 21:43:35 WARNING (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Already running
2020-10-11 21:43:36 INFO (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Executing step call service
2020-10-11 21:43:36 INFO (MainThread) [homeassistant.components.automation.temperature_humidity_sensor_01_sleep] temperature_humidity_sensor_01_sleep: Executing step delay 0:00:01

```

glmnet, 2020-10-11:
You seem to be looking at home assistant logs, not ESPHome logs

hktkzyx, 2020-10-11:
> You seem to be looking at home assistant logs, not ESPHome logs

It is homeassistant log

closed 2020-10-12 by hktkzyx