esphome/issues #1546: [automatic_baseline_calibration] is an invalid option for [sensor.mhz19]

WouterJN, 2020-10-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

PIP on OSX

**ESP (ESP32/ESP8266, Board/Sonoff):**

D1 Mini

**ESPHome version (latest production, beta, dev branch)**

Latest (1.14.6
**Affected component:**

sensor.mhz19

**Description of problem:**
I get the error "[automatic_baseline_calibration] is an invalid option for [sensor.mhz19]". Pls see the print screen below.
[image]

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: co2_livingroom
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "SSID"
  password: "password"

# Enable logging
logger:
# Disable UART Logging
  baud_rate: 0

# Enable Home Assistant API
api:

ota:
  password: "password"

uart:
  rx_pin: RX
  tx_pin: TX
  baud_rate: 9600

sensor:
  - platform: mhz19
    co2:
      name: "MH-Z19 livingroom CO2 Value"
    temperature:
      name: "MH-Z19 livingroom Temperature"
    automatic_baseline_calibration: false  
    update_interval: 60s

```

**Logs (if applicable):**

N/A.

**Additional information and things you've tried:**

Update ESPHome to latest release.

rradar, 2020-10-19:
> ESPHome version (latest production, beta, dev branch)
> 
> Latest (1.14.6

Latest right now is 1.15.x. Could you confirm with the latest release?

WouterJN, 2020-10-23:
This can be closed. I installed the Dev branch and this issue has been gone.

closed 2020-10-23 by WouterJN