esphome/issues #1547: Can't use integration after HA and ESPhome update

mrtimelord, 2020-10-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Home Assistant Core: 0.116.2 ( Hass.io )
Home Assistant Supervisor: 247

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP32

**ESPHome version (latest production, beta, dev branch)** Add-on version: 1.15.2

**Affected component:** Integration

**Description of problem:**
After updating both Home Assistant and esphome I am unable to add a device through the integration option. I enter either the IP of the device or the .local address but nothing happens, I get an infinte loading circle. I've tried restarting both HA and the Esp module multiple times. I am able to access the web interface of the module just fine and I can look at the logs of the device from the supervisor screen ( in the esphome addon )

**Problem-relevant YAML-configuration entries:**
Here's the entire yaml for the module
```yaml
substitutions:
  esphome_name: btsensornode
esphome:
  name: ${esphome_name}
  platform: ESP32
  board: mhetesp32devkit
 
wifi:
  ssid: '*'
  password: '*'
  
#  Optional manual IP
  manual_ip:
    static_ip: 10.28.131.15
    gateway: 10.28.131.254
    subnet: 255.255.255.0
    dns1: 10.28.131.254
    dns2: 1.1.1.1
  
# Enable logging
logger:
  level: VERBOSE

web_server:
  port: 80
  
 #Enable Home Assistant API

api:
  reboot_timeout: 60min
  password: "*"
  #port: 6055

ota:
  password: "*"
  
 #Enable Bluetooth scanning for this ESP32
esp32_ble_tracker:

sensor:
  - platform: xiaomi_lywsdcgq
    mac_address: 58:2D:34:32:19:59
    temperature:  
      name: "S1 Temp"
    humidity:
      name: "S1 Umi"
    battery_level:
      name: "S1 Bat"
      
  - platform: xiaomi_lywsdcgq
    mac_address: 58:2D:34:36:88:09
    temperature:  
      name: "S2 Temp"
    humidity:
      name: "S2 Umi"
    battery_level:
      name: "S2 Bat"
      
  - platform: uptime
    name: "${esphome_name} Uptime"
    filters:
      - lambda: return x / 3600.0;
    unit_of_measurement: "h"
    accuracy_decimals: 2
  - platform: wifi_signal
    name: "${esphome_name} WiFi"
    update_interval: 60s

time:
  - platform: homeassistant
    on_time:
      - seconds: 0
        minutes: 0
        hours: 4
        days_of_week: MON-SUN
        then:
           - switch.toggle: btreset_switch

switch:
  - platform: restart
    name: "${esphome_name}-restart"
    id: btreset_switch

```

**Logs (if applicable):**

The logs on the device appear to be normal
```

```

**Additional information and things you've tried:**
Tried rolling back the ESPhome version to 1.15.0 with no success, tried changing IPs, ports and hostnames with no success, it simply refuses to connect with the integration after the update of HA and Esphome

eboston, 2020-10-14:
I just ran into a problem like this.  When I use the manual_ip option, the device doesn't connect to the network.  But when I don't use it (dynamic ip), it connects fine.  I have to physically reflash the device after OTA flash if manual_ip was used.  I am using a Shelly 1.

mrtimelord, 2020-10-14:
My device connects to the network and I'm able to access the web server just fine, I can even use the "logs" button under the add on in hassio, I just can't use the integration. I am stuck to using mqtt for now...

bangom, 2020-10-26:
> My device connects to the network and I'm able to access the web server just fine, I can even use the "logs" button under the add on in hassio, I just can't use the integration. I am stuck to using mqtt for now...

Same here, after adding new Device under ESPhome addon in Hassio no new device integration (and its entities) is automatically visible in hassio. Can connect / ping / see logs of new  ESPhome device, but it is not integrated into hassio. Old devices integration are visible and working.

**Workaround**: Manual integration of new esphome device into hassio works (Hassio->Configuration->Integrations->Setup a new integration->ESPhome... Host=IP or FQDN of new esphome device).

Home Assistant: 0.116.4
Operating System: HassOS 4.15
ESPhome: dev

kiwipaulrob, 2020-11-14:
I cured my connection problems by compiling for arduino 1.0.3, i.e.

`esphome:
  name: ble_lounge
  platform: ESP32
  board: wemos_d1_mini32
  arduino_version: 1.0.3`

closed 2021-03-28 by stale[bot]