esphome/issues #1553: Multiple HX711 load cells, measuring just one

Luc3as, 2020-10-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on Docker supervised
**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini
**ESPHome version (latest production, beta, dev branch)**

latest production 1.15.2
**Affected component:**

https://esphome.io/components/sensor/hx711.html?highlight=hx711

**Description of problem:**
I tried to hook up 2 of HX711 load cell amplifier boards, Each have own Clock and Data pins assigned D3 and D4, D5 and D6 ( I tried another pins as well, and I also tried to use common clock pin) 
everythings starts as normal, no sensor initialisation error, however, when I hook up the load cells, just one sensor is changing its values, another is randomly generating measurements no matter if load changes. I tried to swap the boards, and load cells, even Wemos itself, still getting into the same problem, Only one board is reading correct data. I tried to lower the gain. When I connect only one HX711, and tried both of them, they work as supposed. Could't there be some "locking" or interrupt issue when using multiple instances of sensor component ? I was thinking about duplicating the sensor to custom component and then run as native HX711 and one as custom component, however after looking into code for few hours it is beyond my actual skill. 

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: "smart_vaha"
  upper_devicename: "Smart váha"

esphome:
  name: $devicename
  platform: ESP8266
  board: d1_mini

wifi:
  networks:
  - ssid: "Luc3as-IoT"
    password: !secret Luc3as_IOT_WIFI_PASS
  manual_ip:
    static_ip: 192.168.3.123
    gateway: 192.168.3.1
    subnet: 255.255.255.0 
    dns1: 192.168.3.1

# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${upper_devicename} Hotspot
    password: !secret esphome_fallback_ap_pass

captive_portal:

# Enable logging
logger:
  #level: debug

# Enable Home Assistant API
api:
  password: !secret esphome_api_pass

ota:
  password: !secret esphome_ota_pass

text_sensor:
  - platform: template
    name: ${upper_devicename} uptime
    lambda: |-
      uint32_t dur = id(uptime_s).state;
      int dys = 0;
      int hrs = 0;
      int mnts = 0;
      if (dur > 86399) {
        dys = trunc(dur / 86400);
        dur = dur - (dys * 86400);
      }
      if (dur > 3599) {
        hrs = trunc(dur / 3600);
        dur = dur - (hrs * 3600);
      }
      if (dur > 59) {
        mnts = trunc(dur / 60);
        dur = dur - (mnts * 60);
      }
      char buffer[17];
      sprintf(buffer, "%ud %02uh %02um %02us", dys, hrs, mnts, dur);
      return {buffer};
    icon: mdi:clock-start
    update_interval: 30s
  
sensor:
  - platform: uptime
    id: uptime_s
    update_interval: 30s
  - platform: wifi_signal
    name: ${upper_devicename} WiFi Signal
    update_interval: 60s  
    
  - platform: hx711
    name: "Váha v posteli Majka"
    id: vaha_majka
    dout_pin: D4
    clk_pin: D3
    gain: 64
    update_interval: 3s   
    filters:
      - calibrate_linear:
          - -28862  -> 0
          - -257679  -> 78
      - sliding_window_moving_average:
          window_size: 5
          send_every: 10      
    unit_of_measurement: kg    

  - platform: hx711
    name: "Váha v posteli Lukáš"
    id: vaha_lukas
    dout_pin: D6
    clk_pin: D5
    gain: 64
    update_interval: 3s   
    filters:
      - calibrate_linear:
          - -28862  -> 0
          - -257679  -> 78
      - sliding_window_moving_average:
          window_size: 5
          send_every: 10      
    unit_of_measurement: kg    

switch:
  - platform: restart
    name: ${upper_devicename} reštart
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

closed 2020-10-17 by Luc3as

jschaeke, 2021-05-07:
@Luc3as Did you manage to get it working?

Luc3as, 2021-05-07:
sure thing, I replaced the HX711 , and used separate pins for each board. works like supposed. 
here is config 
```yaml
sensor:
  - platform: hx711
    id: vaha_lukas
    dout_pin: D6
    clk_pin: D5
    gain: 128
    update_interval: 1s   
    filters:
      - calibrate_linear:
          - -88000 -> 0
          - -345000 -> 78
      - sliding_window_moving_average:
          window_size: 5
          send_every: 10      
    unit_of_measurement: kg  
    on_value:
      then:
        - sensor.template.publish:
            id: vaha_lukas_template
            state: !lambda '
                if (id(vaha_lukas).state < 0 ) {
                  return 0.0;
                } else {
                  return x;
                }'

  - platform: template
    name: "Váha v posteli Lukáš"
    id: vaha_lukas_template
    unit_of_measurement: "kg"
    icon: "mdi:scale"

  - platform: hx711
    id: vaha_majka
    dout_pin: D4
    clk_pin: D3
    gain: 128
    update_interval: 1s   
    filters:
      - calibrate_linear:
          - -487000  -> 0
          - -760500 -> 78
      - sliding_window_moving_average:
          window_size: 5
          send_every: 10      
    unit_of_measurement: kg    
    on_value:
      then:
        - sensor.template.publish:
            id: vaha_majka_template
            state: !lambda '
                if (id(vaha_majka).state < 0 ) {
                  return 0.0;
                } else {
                  return x;
                }'

  - platform: template
    name: "Váha v posteli Majka"
    id: vaha_majka_template
    unit_of_measurement: "kg"
    icon: "mdi:scale"

binary_sensor:
  - platform: template
    name: "Lukáš v posteli"
    id: lukas_v_posteli
    lambda: |-
      if (id(vaha_lukas).state > ${spustac_vaha_lukas}) {
        return true;
      } else {
      return false;
      }
    device_class: "moving"
    
  - platform: template
    name: "Majka v posteli"
    id: majka_v_posteli
    lambda: |-
      if (id(vaha_majka).state > ${spustac_vaha_majka}) {
        return true;
      } else {
      return false;
      }
    device_class: "moving"
    
  - platform: template
    name: "Obsadená postel"
    lambda: |-
      if (id(lukas_v_posteli).state || id(majka_v_posteli).state) {
        return true;
      } else {
      return false;
      }
    device_class: "occupancy"

```

jschaeke, 2021-05-07:
Ok good to know.. I wasn't sure it was supported.