esphome/issues #1556: ESPHome update to ver: 1.15.2   ESP32   dallas.sensor:   Found no sensors!

danilindo2121, 2020-10-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** 

arch | armv7l
-- | --
chassis | embedded
dev | false
docker | true
docker_version | 19.03.11
hassio | true
host_os | HassOS 4.13
installation_type | Home Assistant OS
os_name | Linux
os_version | 4.19.127-v7l
python_version | 3.8.3
supervisor | 247
timezone | America/Santo_Domingo
version | 0.114.4
virtualenv | false

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp-wrover-kit

**ESPHome version (latest production, beta, dev branch)**

 ESPHome: 1.15.2  and   ESPHome dev 1.16.0
**Affected component:**

https://esphome.io/components/sensor/dallas.html

**Description of problem:**
After update to ver ESPHome: 1.15.2  and   ESPHome dev 1.16.0
[W][dallas.sensor:077]:   Found no sensors!

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: luces_gvp
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "GVP_IoT_1"
  password: "Danilindo2309"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Luces Gvp Fallback Hotspot"
    password: "POczoX5jllqF"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

dallas:
  - pin: GPIO23
   # update_interval: 15s

sensor:
  - platform: dallas
    address: 0x9902049245EBE728  
    name: "Oficina"
    id: tempetatura_oficina  
    
    
  - platform: dallas
    address: 0xAC02099245227328
    name: "Sala"
    id: temperatura_sala  
  

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO12
      mode: INPUT_PULLUP
      inverted: false
    id: button_1
    on_press:
      then:
        - light.toggle: light_1
        
  - platform: gpio
    pin:
      number: GPIO13
      mode: INPUT_PULLUP
      inverted: false
    id: button_2
    on_press:
      then:
        - light.toggle: light_2
        
  - platform: gpio
    pin:
      number: GPIO14
    name: "PIR Baño"
    device_class: occupancy
#    filters:
#      - delayed_off: 30s
      
        
  - platform: gpio
    pin:
      number: GPIO27
      mode: INPUT_PULLUP
      inverted: false
    id: button_4
    on_press:
      then:
        - light.toggle: light_4
    
        
  - platform: gpio
    pin:
      number: GPIO26
    name: "PIR Oficina"
    id: pir_oficina
    device_class: occupancy
    filters:
      - delayed_off: 60s
      
    
  - platform: gpio
    pin:
      number: GPIO25
    name: "PIR Sala"
    device_class: occupancy 
    filters:
      - delayed_off: 60s    
    
  - platform: gpio
    pin:
      number: GPIO33
    name: "PIR Taller"
    device_class: motion    
    
output:
  - platform: gpio
    id: relay_1
    pin: GPIO2
    
  - platform: gpio
    id: relay_2
    pin: GPIO15
    
  - platform: gpio
    id: relay_3
    pin: GPIO4
    
  - platform: gpio
    id: relay_4
    pin: GPIO5
    
    
light:
  - platform: binary
    name: "Luz Oficina"
    id: light_1
    output: relay_1

  - platform: binary
    name: "Luz Sala"
    id: light_2
    output: relay_2
    
  - platform: binary
    name: "Luz Baño"
    id: light_3
    output: relay_3

  - platform: binary
    name: "Luz Taller"
    id: light_4
    output: relay_4
    

```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/luces_gvp.yaml...
INFO Starting log output from luces_gvp.local using esphome API
INFO Connecting to luces_gvp.local:6053 (192.168.0.14)
INFO Successfully connected to luces_gvp.local
[18:16:20][I][app:105]: ESPHome version 1.15.2 compiled on Oct 15 2020, 16:57:10
[18:16:20][C][wifi:443]: WiFi:
[18:16:20][C][wifi:303]:   SSID: 

[redacted]
[18:16:20][C][wifi:304]:   IP Address: 192.168.0.14
[18:16:20][C][wifi:306]:   BSSID: 

[redacted]
[18:16:20][C][wifi:307]:   Hostname: 'luces_gvp'
[18:16:20][C][wifi:311]:   Signal strength: -51 dB ▂▄▆█
[18:16:20][C][wifi:315]:   Channel: 6
[18:16:20][C][wifi:316]:   Subnet: 255.255.255.0
[18:16:20][C][wifi:317]:   Gateway: 192.168.0.1
[18:16:20][C][wifi:318]:   DNS1: 8.8.8.8
[18:16:20][C][wifi:319]:   DNS2: 8.8.4.4
[18:16:20][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'button_1'
[18:16:20][C][gpio.binary_sensor:016]:   Pin: GPIO12 (Mode: INPUT_PULLUP)
[18:16:20][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'button_2'
[18:16:20][C][gpio.binary_sensor:016]:   Pin: GPIO13 (Mode: INPUT_PULLUP)
[18:16:20][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'PIR Baño'
[18:16:20][C][gpio.binary_sensor:015]:   Device Class: 'occupancy'
[18:16:20][C][gpio.binary_sensor:016]:   Pin: GPIO14 (Mode: INPUT)
[18:16:20][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'button_4'
[18:16:20][C][gpio.binary_sensor:016]:   Pin: GPIO27 (Mode: INPUT_PULLUP)
[18:16:20][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'PIR Oficina'
[18:16:20][C][gpio.binary_sensor:015]:   Device Class: 'occupancy'
[18:16:20][C][gpio.binary_sensor:016]:   Pin: GPIO26 (Mode: INPUT)
[18:16:20][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'PIR Sala'
[18:16:20][C][gpio.binary_sensor:015]:   Device Class: 'occupancy'
[18:16:20][C][gpio.binary_sensor:016]:   Pin: GPIO25 (Mode: INPUT)
[18:16:20][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'PIR Taller'
[18:16:20][C][gpio.binary_sensor:015]:   Device Class: 'motion'
[18:16:20][C][gpio.binary_sensor:016]:   Pin: GPIO33 (Mode: INPUT)
[18:16:20][C][gpio.output:010]: GPIO Binary Output:
[18:16:20][C][gpio.output:011]:   Pin: GPIO2 (Mode: OUTPUT)
[18:16:20][C][gpio.output:010]: GPIO Binary Output:
[18:16:20][C][gpio.output:011]:   Pin: GPIO15 (Mode: OUTPUT)
[18:16:20][C][gpio.output:010]: GPIO Binary Output:
[18:16:20][C][gpio.output:011]:   Pin: GPIO4 (Mode: OUTPUT)
[18:16:20][C][gpio.output:010]: GPIO Binary Output:
[18:16:20][C][gpio.output:011]:   Pin: GPIO5 (Mode: OUTPUT)
[18:16:20][C][logger:185]: Logger:
[18:16:20][C][logger:186]:   Level: DEBUG
[18:16:20][C][logger:187]:   Log Baud Rate: 115200
[18:16:20][C][logger:188]:   Hardware UART: UART0
[18:16:20][C][light:177]: Light 'Luz Oficina'
[18:16:20][C][light:177]: Light 'Luz Sala'
[18:16:20][C][light:177]: Light 'Luz Baño'
[18:16:20][C][light:177]: Light 'Luz Taller'
[18:16:20][C][dallas.sensor:072]: DallasComponent:
[18:16:20][C][dallas.sensor:073]:   Pin: GPIO23 (Mode: INPUT)
[18:16:21][C][dallas.sensor:074]:   Update Interval: 60.0s
[18:16:21][W][dallas.sensor:077]:   Found no sensors!
[18:16:21][C][dallas.sensor:087]:   Device 'Oficina'
[18:16:21][C][dallas.sensor:087]:     Unit of Measurement: '°C'
[18:16:21][C][dallas.sensor:087]:     Accuracy Decimals: 1
[18:16:21][C][dallas.sensor:087]:     Icon: 'mdi:thermometer'
[18:16:21][C][dallas.sensor:095]:     Address: 0x9902049245EBE728
[18:16:21][C][dallas.sensor:096]:     Resolution: 12
[18:16:21][C][dallas.sensor:087]:   Device 'Sala'
[18:16:21][C][dallas.sensor:087]:     Unit of Measurement: '°C'
[18:16:21][C][dallas.sensor:087]:     Accuracy Decimals: 1
[18:16:21][C][dallas.sensor:087]:     Icon: 'mdi:thermometer'
[18:16:21][C][dallas.sensor:095]:     Address: 0xAC02099245227328
[18:16:21][C][dallas.sensor:096]:     Resolution: 12
[18:16:21][C][captive_portal:169]: Captive Portal:
[18:16:21][C][ota:029]: Over-The-Air Updates:
[18:16:21][C][ota:030]:   Address: luces_gvp.local:3232
[18:16:21][C][api:095]: API Server:
[18:16:21][C][api:096]:   Address: luces_gvp.local:6053
[18:16:24][W][dallas.sensor:146]: 'Oficina' - Scratch pad checksum invalid!
[18:16:24][D][sensor:092]: 'Oficina': Sending state nan °C with 1 decimals of accuracy
[18:16:24][W][dallas.sensor:146]: 'Sala' - Scratch pad checksum invalid!
[18:16:24][D][sensor:092]: 'Sala': Sending state nan °C with 1 decimals of accuracy

```

**Additional information and things you've tried:**

I try changing the GPIO pin but noting.
On ESP8266  have no problem, just with ESP32

randybb, 2020-10-15:
All my devices are always on DEV branch and I never had an issue (i have seen only issues with 1.14.4, but never in dev branch) with Dallas temp sensors, even in two wire configuration (data and ground) just with a strong pull-up.

Kotiproffa, 2020-10-19:
Did you make HW changes on the ESP that does not work? Sometimes the dallas temperature sensors break down when "recycled" too many times. 

Does the same sensors work for the ESP8266 that does not for ESP32?

danilindo2121, 2020-10-19:
Solved!
I don't know why ... but all it  started when I updated to the latest version of ESPHome.
The two ds18b20 sensors connected in series with an ESP32 stopped working.
I did a test with a new ESP32 and a ds18b20 connected to 5vdc and it didn't work either, but when I tried connecting the ds18b20 with 3.3vdc they started working fine
Thanks everyone for your help!
[image]

closed 2020-10-19 by rradar