esphome/issues #1557: uart.write doesn't seem to work?

coogle, 2020-10-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

macOS latest pip package

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP-01

**ESPHome version (latest production, beta, dev branch)**

v1.14.4

**Affected component:**

UART bus (https://esphome.io/components/uart.html)

**Description of problem:**

I have one of these boards: https://smile.amazon.com/gp/product/B07WS7D9K1/ref=ppx_yo_dt_b_asin_image_o05_s00?ie=UTF8&psc=1

Here is the docs for it (in Chinese): http://www.lctech-inc.com/cpzx/1/94.html

The way the board works is there is a MCU that controls the relays and it communicates via UART with the ESP-01. There is a breakout header that lets you communicate with the secondary MCU directly and based on what I could figure out looking at the docs I determined sending these strings turns on/off the relays:

```
0xA0 01 01 A2 - CH1 On
0xA0 01 00 A1 - CH1 Off
0xA0 02 01 A3 - CH2 On
0xA0 02 00 A2 - CH2 Off
```

This is done using 115200 N/8/1 and I am able to successfully turn on/off the relays from my PC using a USB FTDI.

I tried to setup ESPHome with some template switches to mimic this behavior, but it isn't working. I can see in the logs that it acts like its sending the UART string, but nothing happens on the device. 

**Problem-relevant YAML-configuration entries:**
```yaml
logger:
  baud_rate: 0

uart:
  baud_rate: 115200
  tx_pin: TX

switch:
  - platform: template
    name: "${pretty_name} CH 1"
    turn_on_action:
      - logger.log: "Sending ON sequence for CH1..."
      - uart.write: [0xA0, 0x01, 0x01, 0xA2]
    turn_off_action:
      - logger.log: "Sending OFF sequence for CH2..."
      - uart.write: [0xA0, 0x01, 0x00, 0xA1]
    optimistic: true

  - platform: template
    name: "${pretty_name} CH 2"
    turn_on_action:
      - logger.log: "Sending ON sequence for CH2..."
      - uart.write: [0xA0, 0x02, 0x01, 0xA3]
    turn_off_action:
      - logger.log: "Sending OFF sequence CH2..."
      - uart.write: [0xA0, 0x02, 0x00, 0xA2]
    optimistic: true
```

**Logs (if applicable):**

**Additional information and things you've tried:**

coogle, 2020-10-17:
Turns out the issue was the device itself, not ESPHome. I was able to use this configuration to make it work on another board.

closed 2020-10-17 by coogle

aladin2000, 2021-03-19:
Hi
I have the same PB as yours 

I have two boards. None of theme are able to  make the relays Switched

Do you Know what  wasvthe problem ?