esphome/issues #1558: 1.15.2 doesn't transition LEDs as HA requests

cooljimy84, 2020-10-16:
1.13.6
transistions work

1.15.2
15 second transistion goes full after 4
10 after 3
30 after 7
60 after 15 seconds

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

arch | x86_64
-- | --
dev | false
docker | false
hassio | false
installation_type | Home Assistant Core
os_name | Linux
os_version | 4.15.0-118-generic
python_version | 3.8.0
timezone | Europe/London
version | 0.116.2
virtualenv | true

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESPNodeMCU V3
**ESPHome version (latest production, beta, dev branch)**

1.15.2
**Affected component:**

transition on or off of WS2812 and SK6812 from HA service

**Description of problem:**
After upgrading from ESPHome 1.13.6 to 1.15.2 i have found that the transitions doesn't quite count right, so a transition of 60 seconds really just goes full brightness after 15 seconds, no fade/ramp up

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp7
  platform: ESP8266
  board: nodemcuv2

# Window lights

wifi:
  ssid: 'WIFI'
  password: 'WIFI'

mqtt:
  broker: '192.168.1.113'
  username: 'user'
  password: 'pass'

# Enable logging
logger:
  level: INFO

ota:
  password: 'pass'

switch:
  - platform: restart
    name: "ESP7 Restart"

sensor:
  - platform: wifi_signal
    name: "WiFi Signal ESP7"
    update_interval: 120s
[... 245 lines cut ...]
[08:10:14][D][sensor:092]: 'ESP7 ldr': Sending state 86.91406 lux with 2 decimals of accuracy
[08:10:39][D][sensor:092]: 'WiFi Signal ESP7': Sending state -60.00000 dB with 0 decimals of accuracy
[08:10:44][D][adc:056]: 'ESP7 ldr': Got voltage=0.44V
[08:10:44][D][sensor:092]: 'ESP7 ldr': Sending state 87.10938 lux with 2 decimals of accuracy
[08:10:44][D][light:264]: 'ESP7 LED' Setting:
[08:10:44][D][light:273]:   State: ON
[08:10:44][D][light:277]:   Brightness: 100%
[08:10:44][D][light:286]:   Red=100%, Green=100%, Blue=100%
[08:10:44][D][light:289]:   White Value: 100%
[08:10:44][D][light:303]:   Transition Length: 60.0s
[08:11:14][D][adc:056]: 'ESP7 ldr': Got voltage=0.40V
[08:11:14][D][sensor:092]: 'ESP7 ldr': Sending state 80.85938 lux with 2 decimals of accuracy

```

**Additional information and things you've tried:**

The following are me setting the trasition time in HA to diffrent things and then testing how long it takes
10 seconds transition goes full brightness after about 3 seconds (no fade/ramp, just straight to full after 3 seconds) 
15 second transition goes full brightness after about 4 seconds (no fade/ramp, just straight to full after 4 seconds)
30 seconds transition goes full brightness after about 7 seconds (no fade/ramp, just straight to full after 7 seconds) 
60 seconds transition goes full brightness after about 15 seconds (no fade/ramp, just straight to full after 15 seconds) 

esphome:dev same results
1.14.0 same results

DieKatzchen, 2020-11-21:
I've also been seeing this behavior. If I set a transition time on the ESPHome, it'll fade in correctly. If I set it in HA, the ESPHome log looks Identical, but the behavior is different.

cooljimy84, 2020-12-15:
Just pulled 1.15.3 and retested. Fade up/down works now (rather than just coming straight on)
However the timings seem off. If i set a fade off or on for 15 seconds, it's completed within 5 seconds.

cooljimy84, 2021-02-10:
Just pulled 1.60.0 and transitions still turn on turn straight away.
turning off, is still not correct on timings. if i set 30 seconds, they fade off over 5 seconds

Also tested with 1.17.0-dev
same as above.

I can try using the API rather than MQTT see if that causes the same issue ?
for now i'm downgrading to 1.13.6

cooljimy84, 2021-05-05:
just pulled 1.17 and the transitions via MQTT still don't time correctly.
Turn on with a transition time of 10 seconds turns on to full brightness after 1 second.
Turn off with a transition time of 10 seconds turns the lights off after 2 seconds.

cooljimy84, 2021-05-05:
If i put in the API rather than MQTT in the config then i get building strobe, so the lights turn on at the colour i said, then they start strobeing and then after they strobe so fast the leds appear on, they turn off.
Fading off again takes 2 seconds when i set 10 seconds.

cooljimy84, 2021-05-19:
Just tried 1.18.0 and still the same behaviour with API and MQTT. I have even tried setting a the time to 10.5, same behaviour.

Not sure what else i can do to help figure out the issue.

closed 2021-08-11 by jesserockz