esphome/issues #1560: Running on single core chip, but application is built with dual core support

ahmetem, 2020-10-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Windows 10, esphomeflasher

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP-WROOM-32U
**ESPHome version (latest production, beta, dev branch)**

Current version: 1.15.2
**Affected component:**

**Description of problem:**
I try to flash the firmware  but the board rebooting.i

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[0;31mE (531) cpu_start: Running on single core chip, but application is built with dual core support.[0m
[0;31mE (540) cpu_start: Please enable CONFIG_FREERTOS_UNICORE option in menuconfig.[0m
abort() was called at PC 0x40082e4b on core 0
```

**Additional information and things you've tried:**

i try many esp32 platform but no one work.

Guyohms, 2020-10-18:
Could you post your YAML config ?

ahmetem, 2020-10-18:
Esphomeflasher work perfect. But when i flash device and reboot. Say esp32 have 1 core but aplication 2 core for. İ think my esp32 only 1 core but esphome work 2 core

esphome:
  name: qqqq
  platform: ESP32
  board: esp-wrove-kit

wifi:
  ssid: "k_cihaz"
  password: "ccxccc"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Qqqq Fallback Hotspot"
    password: "Tff6enLVHspB"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

Guyohms, 2020-10-18:
Can it be that : 
```yaml
board: esp-wrove-kit
```
should be : 
```yaml
board: esp-wrover-kit
```
I guess that would not pass code validation and would not compile also. Might be a typo.

By the way, you can insert your YAML by typing:

> \```yaml
> Your YAML code
> \```

It will make it easier to read.

ahmetem, 2020-10-18:
My copy mistake. İ paste yaml code

ahmetem, 2020-10-19:
i update my log. and 
I do not understand. The processor must be 2-core, but when updating, 1-core information comes. And then gives an error because esphome application 2-core but esp 1-core.

rradar, 2020-10-19:
> I do not understand. The processor must be 2-core, but when updating, 1-core information comes

Could you paste a photo from your esp32? Do you have a second one to cross check if your yaml works on that?

ahmetem, 2020-10-20:
Yaml file is correct. İ checked. 
[image]
[image]

AivarasAukselis, 2020-10-24:
Hmm, your board does not look like this:
[image]

Perhaps try another board configuration?

ahmetem, 2020-10-24:
I try many board but still not working

sliwma, 2020-12-21:
I have the same issue with ESP32-WROOM-32D.

dckiller51, 2021-01-30:
Hello, Look here at the person getting around the problem.[https://www.reddit.com/r/Esphome/comments/j11ayw/working_configuration_for_xiaomi_mi_desk_lamp_1s/](url)

sliwma, 2021-01-30:
Hey, I have this config, works except Bluetooth - created another issue.
https://github.com/esphome/issues/issues/1731

mmakaay, 2021-03-18:
Using the @pauln platform package together with espressif 32 1.11.0 works for my ESP32-WROOM-32D project too.

**For documentation purposes:**

I also managed to compile a platform package for version 1.0.5 using [esp32-arduino-lib-builder](https://github.com/espressif/esp32-arduino-lib-builder), which makes it possible to use espressif32@3.1.0. All I changed to get a working version was the `CONFIG_FREERTOS_UNICORE=y` in sdkconfig, and a hack in `esp-idf/components/esp32/system_api.c`, which disables the MAC CRC check. This might be the same kind of hack that @pauln applied to his release:

```c++
esp_err_t esp_efuse_mac_get_default(uint8_t* mac)
{
    uint8_t efuse_crc;
    esp_efuse_read_field_blob(ESP_EFUSE_MAC_FACTORY, mac, 48);
    esp_efuse_read_field_blob(ESP_EFUSE_MAC_FACTORY_CRC, &efuse_crc, 8);
    uint8_t calc_crc = esp_crc8(mac, 6);

    /*
     * Hack mmakaay: MAC CRC fails on my device
    if (efuse_crc != calc_crc) {
         // Small range of MAC addresses are accepted even if CRC is invalid.
         // These addresses are reserved for Espressif internal use.
        uint32_t mac_high = ((uint32_t)mac[0] << 8) | mac[1];
        if ((mac_high & 0xFFFF) == 0x18fe) {
            uint32_t mac_low = ((uint32_t)mac[2] << 24) | ((uint32_t)mac[3] << 16) | ((uint32_t)mac[4] << 8) | mac[5];
            if ((mac_low >= 0x346a85c7) && (mac_low <= 0x346a85f8)) {
                return ESP_OK;
            }
        } else {
            ESP_LOGE(TAG, "Base MAC address from BLK0 of EFUSE CRC error, efuse_crc = 0x%02x; calc_crc = 0x%02x", efuse_crc, calc_crc);
            abort();
        }
    }
    */
    return ESP_OK;
}
```

Using this I managed to get a ESP32-WROOM-32D up and running. I would be nice to have pre-built updated libraries for this, so we can use some more up to date code in an easy manner. I'm not sure if my build is the one to publish for this. I was mainly hacking on that 1.0.5 build to see if upgrading things would fix a bug that I was working on (it didn't). I'm in no way an expert in this field (yet ;-)).

syssi, 2021-03-18:
@mmakaay The automatic builds of Jason2866 (used for Tasmota) are a good starting point. Take a look at https://github.com/Jason2866/esp32-arduino-lib-builder/raw/framework-arduinoespressif32/framework-arduinoespressif32-release_v3.3-solo1-bd65eb8d1.tar.gz. 

```
  platformio_options:
    platform: espressif32@3.0.0
    platform_packages: |-4
          framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/raw/framework-arduinoespressif32/framework-arduinoespressif32-release_v3.3-solo1-bd65eb8d1.tar.gz
```

5tan, 2021-06-21:
@syssi I get 404 with your workaround

5tan, 2021-06-21:
@dckiller51 
With this config:
```
esphome:
  name: test
  platform: ESP32
  board: esp32doit-devkit-v1
  platformio_options:
    platform: espressif32@1.11.0
    platform_packages: |-4
          framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.4
```
I frequently get reboots with following error:
```
[17:36:29]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[17:36:29]Core 0 register dump:
[17:36:29]PC      : 0x400e328e  PS      : 0x00060930  A0      : 0x800e3326  A1      : 0x3ffb1be0  
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d aeolus_uv_lamp_esp32 -t idedata
WARNING Decoded 0x400e328e: AsyncEventSourceClient::_runQueue() at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
```
on ESP32-SOLO-1

syssi, 2021-06-21:
@5tan The repo/build has moved to:

```
  platformio_options:
    platform: espressif32@3.0.0
    platform_packages: tasmota/framework-arduinoespressif32 @ 3.10006.210420
```

5tan, 2021-06-22:
@syssi 

With this config I get some warnings regarding Hash package:
```
$ esphome test.yaml compile && esphome test.yaml run 
INFO Reading configuration test.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d abc
Processing abc (board: esp32dev; framework: arduino; platform: espressif32@3.0.0)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210420 (1.0.6) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
Library Manager: AsyncTCP-esphome @ 1.2.2 has been installed!
Library Manager: Installing ESPAsyncWebServer-esphome @ 1.2.7
Library Manager: ESPAsyncWebServer-esphome @ 1.2.7 has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing ESPAsyncTCP-esphome
Library Manager: ESPAsyncTCP-esphome @ 1.2.3 has been installed!
Library Manager: Installing Hash
Library Manager: Warning! Could not install dependency {'name': 'Hash', 'platforms': ['espressif8266']} for package 'ESPAsyncWebServer-esphome'
Library Manager: Installing Hash
Warning! Could not find the package with 'Hash' requirements for your system 'linux_x86_64'
Library Manager: Installing ArduinoJson-esphomelib @ 5.13.3
Library Manager: ArduinoJson-esphomelib @ 5.13.3 has been installed!
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <AsyncTCP-esphome> 1.2.2
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling .pioenvs/abc/src/esphome/components/json/json_util.cpp.o
Compiling .pioenvs/abc/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/abc/src/esphome/components/ota/ota_component.cpp.o
Compiling .pioenvs/abc/src/esphome/components/switch/automation.cpp.o
...
```
Also crash on boot up:
```
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[12:55:43]ets Jun  8 2016 00:22:57
[12:55:43]
[12:55:43]rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
[12:55:43]configsip: 0, SPIWP:0xee
[12:55:43]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:55:43]mode:DIO, clock div:2
[12:55:43]load:0x3fff0018,len:4
[12:55:43]load:0x3fff001c,len:160
[12:55:43]ho 0 tail 12 room 4
[12:55:43]load:0x40078000,len:9928
[12:55:43]ho 0 tail 12 room 4
[12:55:43]load:0x40080400,len:5664
[12:55:43]entry 0x40080678
[12:55:43]\xff[I][logger:170]: Log initialized
[12:55:43][C][ota:366]: There have been 3 suspected unsuccessful boot attempts.
[12:55:43][I][app:029]: Running through setup()...
[12:55:43][C][wifi:033]: Setting up WiFi...
[12:55:43][D][wifi:324]: Starting scan...
[... 13 lines cut ...]
[12:55:48]E (5262) task_wdt: CPU 0: loopTask
[12:55:48]E (5262) task_wdt: Aborting.
[12:55:48]abort() was called at PC 0x40146710 on core 0
[12:55:48]
[12:55:48]ELF file SHA256: 0000000000000000
[12:55:48]
[12:55:48]Backtrace: 0x40087b9b:0x3ffbfa50 0x40087da5:0x3ffbfa70 0x40146710:0x3ffbfa90 0x40086666:0x3ffbfab0 0x400d59f2:0x3ffb1d90 0x400d6735:0x3ffb1dd0 0x4015a5cd:0x3ffb1df0 0x4015a661:0x3ffb1e10 0x400d83b9:0x3ffb1e30 0x400d9fe5:0x3ffb1e80 0x400e462e:0x3ffb1fb0 0x40088d46:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d abc -t idedata
WARNING Decoded 0x40087b9b: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:644
WARNING Decoded 0x40087da5: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:644
WARNING Decoded 0x40146710: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
WARNING Decoded 0x40086666: _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
WARNING Decoded 0x400d59f2: esphome::wifi::WiFiComponent::check_connecting_finished() at /home/z00lwix/test/abc/src/esphome/components/wifi/wifi_component.cpp:576
WARNING Decoded 0x400d6735: esphome::wifi::WiFiComponent::loop() at /home/z00lwix/test/abc/src/esphome/components/wifi/wifi_component.cpp:576
WARNING Decoded 0x4015a5cd: esphome::Component::call_loop() at /home/z00lwix/test/abc/src/esphome/core/component.cpp:156
WARNING Decoded 0x4015a661: esphome::Component::call() at /home/z00lwix/test/abc/src/esphome/core/component.cpp:156
WARNING Decoded 0x400d83b9: esphome::Application::setup() at /home/z00lwix/test/abc/src/esphome/core/application.cpp:50 (discriminator 2)
WARNING Decoded 0x400d9fe5: setup() at /home/z00lwix/test/abc/test.yaml:46
WARNING Decoded 0x400e462e: loopTask(void*) at /home/z00lwix/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:18
WARNING Decoded 0x40088d46: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:403
[12:55:49]
[12:55:49]Rebooting...
```

syssi, 2021-06-22:
@5tan Could you provide your yaml? My setup is stable if bluetooth isn't used.

5tan, 2021-06-22:
my yaml is pretty minimal:
```yaml
esphome:
  name: abc
  platform: ESP32
  board: esp32dev
  platformio_options:
    platform: espressif32@3.0.0
    platform_packages: tasmota/framework-arduinoespressif32 @ 3.10006.210420

wifi:
  ssid: ...
  password: ...

logger:
  level: DEBUG

ota:

web_server:
  port: 80
  auth:
    username: user
    password: pass

globals:
  - id: my_global
    type: bool
    restore_value: no
    initial_value: "false"

switch:
  - platform: template
    id: my_switch
    name: my_switch
    lambda: |-
      return id(my_global);
    turn_on_action:
      - lambda: |-
          id(my_global) = true;
    turn_off_action:
      - lambda: |-
          id(my_global) = false;
```
I get Guru Meditation Error:
```
[15:52:38]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[15:52:38]Core 0 register dump:
[15:52:38]PC      : 0x400dc960  PS      : 0x00060030  A0      : 0x800dc9fe  A1      : 0x3ffb1d40  
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d abc -t idedata
WARNING Decoded 0x400dc960: AsyncEventSourceClient::_runQueue() at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
[15:52:39]A2      : 0x3ffd1fc4  A3      : 0x77735436  A4      : 0x00000000  A5      : 0x3ffd0b7e  
[15:52:39]A6      : 0x0000000a  A7      : 0x1b2e4646  A8      : 0x00000000  A9      : 0x3ffb1d20  
[15:52:39]A10     : 0x00000000  A11     : 0x00000057  A12     : 0x00000057  A13     : 0x3ffd0b10  
[15:52:39]A14     : 0x00000000  A15     : 0x5678006d  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
[15:52:39]EXCVADDR: 0x00000008  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[15:52:39]
[15:52:39]ELF file SHA256: 0000000000000000
[15:52:39]
[15:52:39]Backtrace: 0x400dc960:0x3ffb1d40 0x400dc9fb:0x3ffb1d60 0x400dca1b:0x3ffb1d80 0x400dca71:0x3ffb1da0 0x400d3746:0x3ffb1dd0 0x400d204d:0x3ffb1df0 0x400d2195:0x3ffb1e20 0x400d8bdd:0x3ffb1e40 0x400d8c0a:0x3ffb1e70 0x400d3095:0x3ffb1ed0 0x400d370d:0x3ffb1ef0 0x40080e76:0x3ffb1f10 0x400d74e9:0x3ffb1f60 0x400d98aa:0x3ffb1f90 0x400e463d:0x3ffb1fb0 0x40088d46:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400dc960: AsyncEventSourceClient::_runQueue() at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400dc9fb: AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*) at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400dca1b: AsyncEventSourceClient::write(char const*, unsigned int) at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400dca71: AsyncEventSource::send(char const*, char const*, unsigned int, unsigned int) at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400d3746: std::_Function_handler<void (int, char const*, char const*), esphome::web_server::WebServer::setup()::{lambda(int, char const*, char const*)#2}>::_M_invoke(std::_Any_data const&, int&&, char const*&&, int&&) at /home/z00lwix/test/abc/src/esphome/components/web_server/web_server.cpp:68
 (inlined by) _M_invoke at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
WARNING Decoded 0x400d204d: std::function<void (int, char const*, char const*)>::operator()(int, char const*, char const*) const at /home/z00lwix/test/abc/src/esphome/components/logger/logger.cpp:175
 (inlined by) esphome::CallbackManager<void (int, char const*, char const*)>::call(int, char const*, char const*) at /home/z00lwix/test/abc/src/esphome/core/helpers.h:212
 (inlined by) esphome::logger::Logger::log_message_(int, char const*, int) at /home/z00lwix/test/abc/src/esphome/components/logger/logger.cpp:113
WARNING Decoded 0x400d2195: esphome::logger::Logger::log_vprintf_(int, char const*, int, char const*, __va_list_tag) at /home/z00lwix/test/abc/src/esphome/components/logger/logger.cpp:175
WARNING Decoded 0x400d8bdd: esphome::esp_log_vprintf_(int, char const*, int, char const*, __va_list_tag) at /home/z00lwix/test/abc/src/esphome/core/log.cpp:32
WARNING Decoded 0x400d8c0a: esphome::esp_log_printf_(int, char const*, int, char const*, ...) at /home/z00lwix/test/abc/src/esphome/core/log.cpp:14
WARNING Decoded 0x400d3095: esphome::switch_::Switch::toggle() at /home/z00lwix/test/abc/src/esphome/components/switch/switch.cpp:53
WARNING Decoded 0x400d370d: std::_Function_handler<void (), esphome::web_server::WebServer::handle_switch_request(AsyncWebServerRequest*, esphome::web_server::UrlMatch)::{lambda()#1}>::_M_invoke(std::_Any_data const&) at /home/z00lwix/test/abc/src/esphome/components/web_server/web_server.cpp:68
 (inlined by) _M_invoke at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
WARNING Decoded 0x40080e76: std::function<void ()>::operator()() const at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_vector.h:932
 (inlined by) esphome::Scheduler::call() at /home/z00lwix/test/abc/src/esphome/core/scheduler.cpp:158
WARNING Decoded 0x400d74e9: esphome::Application::loop() at /home/z00lwix/test/abc/src/esphome/core/application.cpp:70
[... cut ...]
WARNING Decoded 0x40088d46: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:403
```
Almost instantly, when I start switching the switch using REST :see_no_evil: 
```
while [ true ] ; do time curl -u user:pass http://192.168.0.10/switch/my_switch/toggle -i -X POST ; sleep 1 ; done
```
https://user-images.githubusercontent.com/15819543/122938461-0e53ba00-d373-11eb-8caf-4b537ef122d0.mp4

PS the very same yaml (except of board/platformio_options settings) works perfect with ESP-WROOM-32!

5tan, 2021-06-22:
Even with minimal config like:
```yaml
esphome:
  name: abc
  platform: ESP32
  board: esp32dev
  platformio_options:
    platform: espressif32@3.0.0
    platform_packages: tasmota/framework-arduinoespressif32 @ 3.10006.210420

wifi:
  ssid: ...
  password: ...

logger:
  level: DEBUG

ota:
```
I got:
```
[16:32:59][I][wifi:194]: WiFi Connecting to 'CGA2121_R99Vk7A'...
[16:33:02]E (5235) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[16:33:02]E (5235) task_wdt:  - IDLE0 (CPU 0)
[16:33:02]E (5235) task_wdt: Tasks currently running:
[16:33:02]E (5235) task_wdt: CPU 0: loopTask
[16:33:02]E (5235) task_wdt: Aborting.
[16:33:02]abort() was called at PC 0x4013b21c on core 0
[16:33:02]
[16:33:02]ELF file SHA256: 0000000000000000
[16:33:02]
[16:33:02]Backtrace: 0x40087b9b:0x3ffbf9f0 0x40087da5:0x3ffbfa10 0x4013b21c:0x3ffbfa30 0x40086666:0x3ffbfa50 0x4008ada8:0x3ffb1e40 0x400d97fb:0x3ffb1e60 0x400d5445:0x3ffb1e80 0x400d65bd:0x3ffb1ed0 0x400da2ee:0x3ffb1fb0 0x40088d46:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d abc -t idedata
WARNING Decoded 0x40087b9b: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:644
WARNING Decoded 0x40087da5: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:644
WARNING Decoded 0x4013b21c: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
WARNING Decoded 0x40086666: _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
WARNING Decoded 0x4008ada8: _frxt_dispatch at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:435
WARNING Decoded 0x400d97fb: __yield at /home/z00lwix/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-misc.c:172
WARNING Decoded 0x400d5445: esphome::Application::setup() at /home/z00lwix/test/abc/src/esphome/core/application.cpp:55
WARNING Decoded 0x400d65bd: setup() at /home/z00lwix/test/abc/src/main.cpp:74
WARNING Decoded 0x400da2ee: loopTask(void*) at /home/z00lwix/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:18
WARNING Decoded 0x40088d46: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:403
```

syssi, 2021-06-22:
Which esphome version do you use?

5tan, 2021-06-22:
`Version: 1.18.0`

dckiller51, 2021-06-22:
> `Version: 1.18.0`
I just tested version 1.19.2 on my lamp. No problem here is my yaml.
```
esphome:
  name: mi_desk_lamp_1s
  platform: ESP32
  board: esp32doit-devkit-v1
  platformio_options:
    platform: espressif32@3.0.0
    platform_packages: tasmota/framework-arduinoespressif32 @ 3.10006.210420

wifi:
  ssid: !secret ssid
  password: !secret wpa2
  fast_connect: true

captive_portal:

logger:

api:

ota:
```

5tan, 2021-06-25:
regarding "Guru Meditation Error" from above, I have opened separate ticket: https://github.com/esphome/issues/issues/2191