esphome/issues #1562: SPIFFs deprecated

PierreScerri, 2020-10-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home assistantOS 0.116.4
**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff dual
**ESPHome version (latest production, beta, dev branch)**

1.15.2
**Affected component:**

**Description of problem:**
During compilation, I get a warning about SPIFFs being deprecated.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
n/a
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

rradar, 2020-10-19:
> **Logs (if applicable):**
> ```
> PASTE DEBUG LOG HERE
> 
> ```

Could you try to fill that one @PierreScerri? Thank you!

shirou93, 2020-10-29:
same issue when compile bin file.

sreknob, 2020-11-03:
Yup, seeing this waring in the logs as well.
From the [ESP8266 core docs on filesystem](https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html), it appears that they have depreciated SPIFFS and suggest using [LittleFS](https://github.com/littlefs-project/littlefs) as a replacement. which they say is faster and better in a few regards (and actively maintained).
However, they go on to say in the docs:

> SPIFFS is the original filesystem and is ideal for space and RAM constrained applications that utilize many small files and care about static and dynamic wear levelling and don’t need true directory support. Filesystem overhead on the flash is minimal as well.
> LittleFS is recently added and focuses on higher performance and directory support, but has higher filesystem and per-file overhead (4K minimum vs. SPIFFS’ 256 byte minimum file allocation unit).
> They share a compatible API but have incompatible on-flash implementations, so it is important to choose one or the per project as attempting to mount a SPIFFS volume under LittleFS may result in a format operation and definitely will not preserve any files, and vice-versa.

So staying with SPIFFS may make the most sense for ESPhome given the limited implementation of the filesystem.

closed 2021-03-18 by stale[bot]

compujunk, 2021-06-06:
I have the same issue

neo799, 2021-06-08:
Same issue here

Karma4Good, 2021-06-11:
I do as well - which led me to this thread - but [the most recent message with relevant content](https://github.com/esphome/issues/issues/1562#issuecomment-721386552) says that it's OK. I'm taking that to mean "let it be and don't worry about the warning".