esphome/issues #1563: millis() and time component not working on ESP8266?

ebaschiera, 2020-10-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP8266 NodeMCU 30pin

**ESPHome version (latest production, beta, dev branch):** v1.15.2 (latest production)

**Affected component:** https://esphome.io/components/time.html

**Description of problem:**

Hi, I am trying to implement a simple blinking effect on a led strip. In other words: in certain conditions (too high energy consumption at home), all leds should repeatedly  turn on and off. I would like to decide the blinking frequency, so I tried to use millis() and the Time component to read the current time and decide the next action, based on the blinking interval.

It looks like millis() and the time component are not working correctly. To simplify the debugging process, I added a log statement that prints millis() and time.now() values: ``ESP_LOGD("ERM", "time: %f - millis: %f", time, millis());`` But they are always zero.

Why they are always zero? Am I missing something?

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: energy_display
  platform: ESP8266
  board: esp01_1m
  on_boot:
    priority: -100
    then:
      - light.turn_on:
          id: led8riga
          effect: "Energy Monitor"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_pwd_energy_display

ota:
  password: !secret api_pwd_energy_display
  
time:
  - platform: homeassistant
    id: homeassistant_time
    timezone: Europe/Rome

light:
  - platform: fastled_clockless
    chipset: WS2811
    pin: GPIO2
    num_leds: 8
    rgb_order: GRB
    name: 8-led energy monitor
    id: led8riga
    effects:
    - addressable_lambda:
        name: "Energy Monitor"
        update_interval: 100ms
        lambda: |-
          const int alert_threshold = -3000; //W
          const int update_interval = 1000; //ms
          const int led_strip_scale = 3000; //W - "all leds on mean 3000W"
          int leds_on = 0;
          auto time = id(homeassistant_time).now().timestamp;
          ESP_LOGD("ERM", "time: %f - millis: %f", time, millis());
          double energy_export_value = (id(energy_export).state);
          //double energy_export_value = (id(energy_import_test).state);
          if (isnan(energy_export_value)) {
            //ESP_LOGD("ERM", "Waiting for energy_export_value");
            it.all() = ESPColor::BLACK;
            return;
          }
          //ESP_LOGD("ERM", "energy_export_value: %f", (energy_export_value));
          if (energy_export_value >= 0) {
            leds_on = ceil((it.size() * energy_export_value) / led_strip_scale);
            it.all() = ESPColor::BLACK;
            it.range(0, leds_on) = ESPColor(0, 255, 0);
          } else {
            leds_on = ceil((it.size() * (energy_export_value * -1)) / led_strip_scale);
            it.all() = ESPColor::BLACK;
            it.range(0, leds_on) = ESPColor(255, 0, 0);
            if (energy_export_value <= alert_threshold) {
              //blink lights!
              static bool is_on = true;
              uint32_t now = millis();
              static uint32_t last_update = now;
              //ESP_LOGD("ERM", "now: %f", millis());
              if (now - last_update < update_interval) {
                //ESP_LOGD("ERM", "intervengo");
                last_update = now;
                if (is_on) {
                  it.all() = ESPColor::BLACK;
                  is_on = false;
                } else {
                  //they are already turned on
                  //it.all() = ESPColor::BLACK;
                  //it.range(0, leds_on) = ESPColor(255, 0, 0);
                  is_on = true;
                }
              }
            }
          }

          
sensor:
  - platform: homeassistant
    id: energy_import_test
    entity_id: sensor.energy_import_test
    internal: true
  - platform: homeassistant
    id: energy_export
    entity_id: sensor.shelly_energy_export
    internal: true

```

**Logs (if applicable):**

```
INFO Starting log output from energy_display.local using esphome API
INFO Connecting to energy_display.local:6053 (192.168.0.134)
INFO Successfully connected to energy_display.local
[20:43:55][I][app:105]: ESPHome version 1.15.2 compiled on Oct 18 2020, 20:43:13
[20:43:55][C][wifi:443]: WiFi:
[20:43:55][C][wifi:303]:   SSID: [redacted]
[20:43:55][C][wifi:304]:   IP Address: 192.168.0.134
[20:43:55][C][wifi:306]:   BSSID: [redacted]
[20:43:55][C][wifi:307]:   Hostname: 'energy_display'
[20:43:55][C][wifi:311]:   Signal strength: -74 dB ▂▄▆█
[20:43:55][C][wifi:315]:   Channel: 1
[20:43:55][C][wifi:316]:   Subnet: 255.255.255.0
[20:43:55][C][wifi:317]:   Gateway: 192.168.0.1
[20:43:55][C][wifi:318]:   DNS1: 192.168.0.127
[20:43:55][C][wifi:319]:   DNS2: 1.1.1.1
[20:43:55][D][ERM:239]: time: 0.000000 - millis: 0.000000
[20:43:55][C][fastled:019]: FastLED light:
[20:43:55][C][fastled:020]:   Num LEDs: 8
[20:43:55][C][fastled:021]:   Max refresh rate: 400
[20:43:55][C][logger:185]: Logger:
[20:43:55][C][logger:186]:   Level: DEBUG
[20:43:55][C][logger:187]:   Log Baud Rate: 115200
[20:43:55][C][logger:188]:   Hardware UART: UART0
[20:43:55][C][light:177]: Light '8-led energy monitor'
[20:43:55][C][light:179]:   Default Transition Length: 1.0s
[20:43:55][C][light:180]:   Gamma Correct: 2.80
[20:43:55][C][homeassistant.time:010]: Home Assistant Time:
[20:43:55][C][homeassistant.time:011]:   Timezone: 'CET-1CEST-2,M3.4.0/2,M10.5.0/3'
[20:43:55][C][ota:029]: Over-The-Air Updates:
[20:43:55][C][ota:030]:   Address: energy_display.local:8266
[20:43:55][C][ota:032]:   Using Password.
[20:43:55][D][ERM:239]: time: 0.000000 - millis: 0.000000
[20:43:55][C][api:095]: API Server:
[20:43:55][C][api:096]:   Address: energy_display.local:6053
[20:43:55][C][homeassistant.sensor:024]: Homeassistant Sensor 'energy_import_test'
[20:43:55][C][homeassistant.sensor:024]:   Unit of Measurement: ''
[20:43:55][C][homeassistant.sensor:024]:   Accuracy Decimals: 1
[20:43:55][C][homeassistant.sensor:025]:   Entity ID: 'sensor.energy_import_test'
[20:43:55][C][homeassistant.sensor:024]: Homeassistant Sensor 'energy_export'
[20:43:55][C][homeassistant.sensor:024]:   Unit of Measurement: ''
[20:43:55][C][homeassistant.sensor:024]:   Accuracy Decimals: 1
[20:43:55][C][homeassistant.sensor:025]:   Entity ID: 'sensor.shelly_energy_export'
[20:43:55][D][api.connection:604]: Client 'Home Assistant 0.116.2 (192.168.0.127)' connected successfully!
[20:43:55][D][ERM:239]: time: 0.000000 - millis: 0.000000
[20:43:55][D][time:040]: Synchronized time: Sun Oct 18 20:43:55 2020
[20:43:55][D][ERM:239]: time: 0.000000 - millis: 0.000000
[20:43:55][D][homeassistant.sensor:019]: 'sensor.energy_import_test': Got state -2520.00
[20:43:55][D][sensor:092]: 'energy_import_test': Sending state -2520.00000  with 1 decimals of accuracy
[20:43:55][D][homeassistant.sensor:019]: 'sensor.shelly_energy_export': Got state -1211.00
[20:43:55][D][sensor:092]: 'energy_export': Sending state -1211.00000  with 1 decimals of accuracy
[20:43:55][D][ERM:239]: time: 0.000000 - millis: 0.000000
[20:43:55][D][ERM:239]: time: 0.000000 - millis: 0.000000
[20:43:55][D][ERM:239]: time: 0.000000 - millis: 0.000000
[20:43:55][D][ERM:239]: time: 0.000000 - millis: 0.000000
[20:43:56][D][ERM:239]: time: 0.000000 - millis: 0.000000
[20:43:56][D][ERM:239]: time: 0.000000 - millis: 0.000000
[20:43:56][D][ERM:239]: time: 0.000000 - millis: 0.000000
[20:43:56][D][ERM:239]: time: 0.000000 - millis: 0.000000
[20:43:56][D][ERM:239]: time: 0.000000 - millis: 0.000000
[20:43:56][D][ERM:239]: time: 0.000000 - millis: 0.000000
[20:43:56][D][ERM:239]: time: 0.000000 - millis: 0.000000

```

**Additional information and things you've tried:**

Thank you!

glmnet, 2020-10-19:
Can't tell from the top of my head about the time but for millis you should use %d not %f. 

Also %f surely isn't gonna work for time either

ebaschiera, 2020-10-19:
Haha, yeah, I was definitely missing something! %d was the answer.

Sorry for my spamming!

Thanks!

closed 2020-10-19 by ebaschiera