esphome/issues #1564: HMC5883L - Update interval below 1s not working

kozoke, 2020-10-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**
 D1 Mini.

**ESPHome version (latest production, beta, dev branch)**
v1.15.2

**Affected component:**
https://github.com/esphome/esphome/tree/dev/esphome/components/hmc5883l

**Description of problem:**
I'm trying to use an update_interval in milliseconds (less that 1 second). Everything works fine at 1 second. 

Per config/time docs I should be able to use 0ms to run each loop. But i'm getting this error:  '''    interval_hz = 1.0/interval_sec
TypeError: unsupported operand type(s) for /: 'float' and 'NoneType''''

Discord Help/EuroTrash suggested:
But whatever that code is, it's making an incorrect assumption, causing division by zero.
Loop speed is actually.. 6ms or something, I think, though, so 1ms would prevent that code from breaking and still run pretty much at the same speed.
Although the component is trying to calculate a Hz value for some reason, so giving it 1ms would be way off.
It's doing a comparison against the hz value to select a data rate which is a bunch of ranges in an enum. It's just trying to do a size comparison, and returning a default, but 0ms will cause division by zero.

Without modification that code will keep failing. Can open an issue, but that won't help you right now. A relatively easy way to fix it for you right now would be to copy the sources of the component to a custom_components/ dir in your configuration directory and do a small modification

Suggested Fix:
Remove lines 84-88 so it always returns the fastest data rate....in sensor.py

I'm a bit of a noob so very much appreciate everything Otto, eurotrash and everyone are doing. So awesome.
And thanks for looking at this bug.

**Problem-relevant YAML-configuration entries:**
```yaml
#hmc5883l magnomitor
  - platform: hmc5883l
    address: 0x1E
    field_strength_x:
      name: "watermeter X"
      icon: mdi:water-pump
    field_strength_y:
      name: "watermeter y"
      icon: mdi:water-pump
    field_strength_z:
      name: "watermeter Z"
      icon: mdi:water-pump
#    heading:
#      name: "HMC5883L Heading"
    oversampling: 1x
    range: 130uT
    #update_interval: 1s
    update_interval: 100ms
```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/watermeter.yaml...
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 11, in <module>
    load_entry_point('esphome', 'console_scripts', 'esphome')()
  File "/opt/esphome/esphome/__main__.py", line 554, in main
    return run_esphome(sys.argv)
  File "/opt/esphome/esphome/__main__.py", line 541, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/opt/esphome/esphome/__main__.py", line 307, in command_run
    exit_code = write_cpp(config)
  File "/opt/esphome/esphome/__main__.py", line 122, in write_cpp
    generate_cpp_contents(config)
  File "/opt/esphome/esphome/__main__.py", line 134, in generate_cpp_contents
    CORE.flush_tasks()
  File "/opt/esphome/esphome/core.py", line 653, in flush_tasks
    next(task)
  File "/opt/esphome/esphome/core.py", line 438, in _wrapper_generator
    for x in var:
  File "/opt/esphome/esphome/core.py", line 434, in _wrapper_generator
    var = gen.send(var)
  File "/opt/esphome/esphome/components/hmc5883l/sensor.py", line 98, in to_code
    cg.add(var.set_datarate(auto_data_rate(config)))
  File "/opt/esphome/esphome/components/hmc5883l/sensor.py", line 85, in auto_data_rate
    interval_hz = 1.0/interval_sec
TypeError: unsupported operand type(s) for /: 'float' and 'NoneType'

```

**Additional information and things you've tried:**
I Tried 999ms, 501MS, 1 MS, 0.5s - none worked.

baldisos, 2021-01-10:
This affects the readout from my QMC5883L too. I can't set the update_interval less than 1s because of this issue, but i actually need to read out more frequently to get good enough data.

chrismyers2000, 2021-02-03:
I too am having this issue. After digging through the spec sheet it seems that this sensor is capable of 75hz update rate. After looking through the library for this sensor in esphome it appears that faster than 1hz update rates are supported....but I keep getting the divide by zero error. 

From the library:

enum HMC5883LDatarate {
  HMC5883L_DATARATE_0_75_HZ = 0b000,
  HMC5883L_DATARATE_1_5_HZ = 0b001,
  HMC5883L_DATARATE_3_0_HZ = 0b010,
  HMC5883L_DATARATE_7_5_HZ = 0b011,
  HMC5883L_DATARATE_15_0_HZ = 0b100,
  HMC5883L_DATARATE_30_0_HZ = 0b101,
  HMC5883L_DATARATE_75_0_HZ = 0b110,

How do we actually use these higher rates? My coding foo is very limited.

baldisos, 2021-06-03:
Can this be resolved in future? I still need to work around this, which isn't quite nice.

kozoke, 2021-06-05:
Agreed. Still looking for a solution or help to make the coding change.