esphome/issues #1565: light.turn_on exhibits odd behavior when setting brightness to 0

jbcodemonkey, 2020-10-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP8266 NODEMCU**

**ESPHome version (latest production, beta, dev branch)**

1.15.2
**Affected component:**

**Description of problem:**
Using a rotary sensor value to set a light brightness on_value works perfectly for all sensor values > 0.
When issuing the light.turn_on command w brightness set to 0, the light is momentarily set to 100% before turning off.
Expected behavior: setting a light brightness to 0 should turn the light off.

**Problem-relevant YAML-configuration entries:**
```yaml
light:
    #rgb led
  - platform: rgb
    id: test1rgb
    name: "test1.rgb"
    red: output_component1
    green: output_component2
    blue: output_component3

# output for rgb led
output:
  - platform: esp8266_pwm
    id: output_component1
    pin: D1
  - platform: esp8266_pwm
    id: output_component2
    pin: D2
  - platform: esp8266_pwm
    id: output_component3
    pin: D3

sensor:
  - platform: rotary_encoder
    name: "test1.rotary"
    id: test1rotary
    pin_a: D6
    pin_b: D7
    min_value: 0
    max_value: 12
    resolution: 1
    on_value:
      then:
        - light.turn_on:
            id: test1rgb
            brightness: !lambda "return x/12;"
            transition_length: 0ms
```

**Logs (if applicable):**

```
[13:51:39][D][sensor:092]: 'test1.rotary': Sending state 1.00000 steps with 0 decimals of accuracy
[13:51:39][D][light:264]: 'test1.rgb' Setting:
[13:51:39][D][light:273]:   State: ON
[13:51:39][D][light:277]:   Brightness: 8%
[13:51:42][D][sensor:092]: 'test1.rotary': Sending state 2.00000 steps with 0 decimals of accuracy
[13:51:42][D][light:264]: 'test1.rgb' Setting:
[13:51:42][D][light:277]:   Brightness: 17%
[13:51:44][D][sensor:092]: 'test1.rotary': Sending state 3.00000 steps with 0 decimals of accuracy
[13:51:44][D][light:264]: 'test1.rgb' Setting:
[13:51:44][D][light:277]:   Brightness: 25%
[13:51:46][D][sensor:092]: 'test1.rotary': Sending state 2.00000 steps with 0 decimals of accuracy
[13:51:46][D][light:264]: 'test1.rgb' Setting:
[13:51:46][D][light:277]:   Brightness: 17%
[13:51:47][D][sensor:092]: 'test1.rotary': Sending state 1.00000 steps with 0 decimals of accuracy
[13:51:47][D][light:264]: 'test1.rgb' Setting:
[13:51:47][D][light:277]:   Brightness: 8%
[13:51:48][D][sensor:092]: 'test1.rotary': Sending state 0.00000 steps with 0 decimals of accuracy
[13:51:48][D][light:264]: 'test1.rgb' Setting:
[13:51:48][D][light:277]:   Brightness: 100%
```

**Additional information and things you've tried:**
When setting transition_length to > 200ms, the momentary full brightness can be observed.
Setting transition_length to 0ms hides the problem visually, but logs still show the light being set to 100%.

[EuroTrash on discord suggested the problem code was likely here](https://github.com/esphome/esphome/blob/5f619e6f0135e27d3a7081408d5d73c53c6d673d/esphome/components/light/light_color_values.h#L164).

jesserockz, 2020-10-19:
I have just reproduced the same using `dim_relative`. When it gets to 0%, the light flickers up then back down to 0% and the state in the logs says 100% as above.

Possibly related to #1523

faxpapier, 2020-11-08:
I have somewhat the same issue. I made the following:

`
output:
  - platform: gpio
    pin: D8
    id: ledpin
  - platform: esp8266_pwm
    pin: D1
    id: ledred
  - platform: esp8266_pwm
    pin: D2
    id: ledgreen
  - platform: esp8266_pwm
    pin: D3
    id: ledblue

light:
  - platform: rgb
    name: "Waste light"
    id: wastelight
    red: ledred
    green: ledgreen
    blue: ledblue
    
    effects:
      - lambda:
          name: Pulsing_red
          update_interval: 10s
          lambda: |-
            static int state = 0;
            auto call = id(wastelight).make_call();
            call.set_transition_length(2000);
            //set all leds to 0
            
            if (state ==0 ) {
              call.set_red(1.0);
              call.set_green(0);
              call.set_blue(0);
            } else if (state ==1 ) {
              call.set_red(0);
              call.set_green(0);
              call.set_blue(0);
            }
            call.perform();
            state += 1;
            if (state == 2) {
              state =0;
            }
`

When calling the effect, the red light turns on and stays on, and all other colors are off. (if I turn the colors on manualy the colors work as axpected) but the output from the logging is the following:

'
[12:23:24][D][light:265]: 'Waste light' Setting:
[12:23:24][D][light:287]:   Red=100%, Green=100%, Blue=100%
[12:23:24][D][light:304]:   Transition Length: 2.0s
[12:23:34][D][light:265]: 'Waste light' Setting:
[12:23:34][D][light:287]:   Red=100%, Green=0%, Blue=0%
[12:23:34][D][light:304]:   Transition Length: 2.0s
[12:23:44][D][light:265]: 'Waste light' Setting:
[12:23:44][D][light:287]:   Red=100%, Green=100%, Blue=100%
[12:23:44][D][light:304]:   Transition Length: 2.0s'

schlion, 2020-11-11:
I made a workaround, kind of

You can get the brightness you light `id(kitchen_light)->remote_values.get_brightness()`.
If your light is in a transition, this will give you the value it is transitioning to get the current value use `id(kitchen_light)->current_values.get_brightness()`

I made a workaround for IR-remote controlled dimming:
```YAML
#if is off, but brightness is low (<0.11), turn on to 0.1 brightness on IR-code
      - if:
          condition: 
            and:
              - lambda: 'return x.command == 0x3AC5;' #remote controll code for "brighter"
              - lambda: 'return id(kitchen_light)->remote_values.get_brightness() <= 0.11;' #0.11 since 0.1 does not trigger ??
              - light.is_off: kitchen_light
          then:
            - light.turn_on:
                id: kitchen_light
                brightness: 0.1
# if light is on, increase brightness on IR-code
      - if:
          condition: 
            and:
              - lambda: 'return x.command == 0x3AC5;' #remote controll code for "brighter"
              - light.is_on: kitchen_light
          then:
            - light.dim_relative:
                id: kitchen_light
                relative_brightness: 0.1

#-------------------------------------------------------
# if light is darker than 0.1 turn off on IR-code
      - if:
          condition: 
            and:
              - lambda: 'return x.command == 0xBA45;' #IR-code for darker
              - lambda: 'return id(kitchen_light)->remote_values.get_brightness() <= 0.11;' #0.11 since 0.1 does not trigger ??
          then:
            - light.turn_off:
                id: kitchen_light
                
# if light is bright enough to dim without reaching 0, dim it down on IR-code
      - if:
          condition: 
            and:
              - lambda: 'return x.command == 0xBA45;'
              - lambda: 'return id(kitchen_light)->remote_values.get_brightness() > 0.11;' #0.11 since 0.1 does not trigger ??
              - light.is_on: kitchen_light
          then:
            - light.dim_relative:
                id: kitchen_light
                relative_brightness: -0.1
```

schlion, 2020-11-11:
This commit might be the source of our problems https://github.com/esphome/esphome-core/pull/526/commits/1ab373430baf688dff1a74383d2bb761f655c82a . Back in [esphome-core](https://github.com/esphome/esphome-core) someone requested that lights turn off, if the brightness is set to 0 (https://github.com/esphome/issues/issues/78).

This code was changed but in spirit it survived [_The Great Merge_](https://github.com/esphome/esphome/pull/504) and lives at https://github.com/esphome/esphome/blob/c9055f2aefdc99c0e5c3dec6246e53efdffecaff/esphome/components/light/light_color_values.h#L160

When normalizing RGB components, if brightness 0 is detected, the light is turned off and brightness is set to `1.0f`, 0 brightness is not allowed. While I can understand that the state should be set to off, if brightness reaches 0, this causes issues down the line.

The easy fix would be to just leave the brightness on 0. But since @OttoWinter wrote `// reset brightness to 100% (0% brightness is not allowed)`, I would very much like to hear his opinion on this. Since changing this might have unforeseen side effect.

Stuart-Phillips, 2021-02-03:
I am having exactly the same issue as described above. It is very frustrating to see the light brighten when I want it to turn down to zero brightness (or off), and also not desired for it to come back on to 100% if switched back on via the HA interface.

I'm not at home right now, so I can't test, but I'm going to try commenting out or editing line 168 here: https://github.com/esphome/esphome/blob/c9055f2aefdc99c0e5c3dec6246e53efdffecaff/esphome/components/light/light_color_values.h#L160 to see what happens. I'm still learning with C and ESPHome, but if I have success and no noticeable side effects I'll share my findings. 

I too would be interested to hear why 0% brightness is not allowed, and why it must be set to 100% @OttoWinter. Or maybe it was a typo and it should have been set to 1% when switched off?

trvrnrth, 2021-03-25:
I ran into this recently as well and came up with basically the same workaround as posted above. In case it helps anyone else here's the config I ended up with for a rotary encoder with switch:
```
binary_sensor:
  - platform: gpio
    id: rotary_encoder_switch
    pin:
      number: GPIO0
      inverted: True
    on_release:
      then:
        - light.toggle: light_id
    filters:
      - delayed_on_off: 10ms

sensor:
  - platform: rotary_encoder
    id: rotary_encoder_main
    pin_a: GPIO12
    pin_b: GPIO14
    on_clockwise:
      - light.dim_relative:
          id: light_id
          relative_brightness: 5%
          transition_length: 0.1s
    on_anticlockwise:
      - if:
          condition:
            lambda: |-
              return roundf(id(light_id)->current_values.get_brightness() * 100.0f) > 5;
          then:
            - light.dim_relative:
                id: light_id
                relative_brightness: -5%
                transition_length: 0.1s
          else:
            - light.turn_off:
                id: light_id
                transition_length: 0.1s
```

closed 2021-09-19 by OttoWinter