esphome/issues #1568: RF433 Transmitter not working

tymzang, 2020-10-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**ESPHome version (latest production, beta, dev branch)**

1.15.2
**Affected component:**

**Description of problem:**
I got easily the RF code from receiver, but when applied these codes to transmitter and nothing happen. Tried to figure out many weeks but no hope. When I press button "RF ON Button", I can see ESPHome sending code, but my bulb still OFF...

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: gh_rf_hub
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "***"
    password: "***"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

remote_receiver:
  pin: D2
  dump:
    - rc_switch
  # Settings to optimize recognition of RF devices
  tolerance: 50%
  filter: 250us
  idle: 4ms
  buffer_size: 2kb
  
remote_transmitter:
  pin: D1
  # RF uses a 100% carrier signal
  carrier_duty_percent: 100%
  
switch:
  - platform: template
    name: RF ON Button
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '00101101000110010110110110010011'
          protocol: 2

```

**Logs (if applicable):**

```
[21:37:41][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=8 data='000000000000000000000000000000000000000'
[21:37:41][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=8 data='000000000000000000000000000000000000000'
[21:37:41][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=8 data='000000000000000000000000000000000000000'
[21:37:41][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=8 data='000000000000000000000000000000000000000'
[21:37:41][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=8 data='000000000000000000000000000000000000000'
[21:37:41][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=8 data='000000000000000000000000000000000000000'
[21:37:41][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=8 data='000000000000000000000000000000000000000'
[21:37:42][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=2 data='00101101000110010110110110010011'
[21:37:42][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=2 data='00101101000110010110110110010011'
[21:37:42][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=2 data='00101101000110010110110110010011'
[21:37:42][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=2 data='00101101000110010110110110010011'
[21:37:42][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=2 data='00101101000110010110110110010011'
[21:37:42][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=2 data='00101101000110010110110110010011'
[21:37:42][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=2 data='00101101000110010110110110010011'
[21:37:42][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=2 data='0010110100011001011011011001001'
[21:37:50][D][switch:021]: 'RF ON Button' Turning ON.
[21:37:51][D][remote_transmitter:066]: Sending remote code...

```

**Additional information and things you've tried:**

glmnet, 2020-10-24:
Try setting up another node as receiver. Then compare what it receives with the transmitter and with ESPHome

davida72, 2021-01-08:
I appreciate this is an old thread but did you get this working? I have the same issue in that I can easily read the data but don't have much success when sending it. On occasion it works, but not reliably. I've noticed that there are settings to tweak, such as pulse_length, sync, zero, one, inverted, repeat, but I don't know how I can establish what those should be other than by trial and error.

I've tried this on two D1 Minis. I'll try it on an ESP32 to see if that makes a difference.

closed 2021-05-23 by stale[bot]