esphome/issues #1575: SSD1306 lines issue

ShonP40, 2020-10-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io (the same issue is also happening with pip3 installations)

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU ESP-32S (the same issue is also happening on the WeMos D1 Mini (ESP8266))

**ESPHome version (latest production, beta, dev branch)**

v1.15.3 (was happening on v1.15.2 as well)

**Affected component:**

SSD1306

**Description of problem:**
Any text displayed will make the SSD1306 display lines above and below it

[image]

**Problem-relevant YAML-configuration entries:**
```yaml
i2c:
  sda: 25
  scl: 26
  id: bus_a

display:
  - platform: ssd1306_i2c
    model: "SH1106 128x64"
    reset_pin: D0
    i2c_id: bus_a
    address: 0x3C
    lambda: |-
      // Print "Shons control unit" in top center.
      it.printf(64, 0, id(font1), TextAlign::TOP_CENTER, "Shons control unit");

      // Print time in HH:MM format
      it.strftime(0, 60, id(font2), TextAlign::BASELINE_LEFT, "%H:%M", id(esptime).now());

      // Print inside temperature (from homeassistant sensor)
      if (id(inside_temperature).has_state()) {
        it.printf(127, 23, id(font3), TextAlign::TOP_RIGHT , "%.1f°", id(inside_temperature).state);
      }

      // Print outside temperature (from homeassistant sensor)
      if (id(outside_temperature).has_state()) {
        it.printf(127, 60, id(font3), TextAlign::BASELINE_RIGHT , "%.1f°", id(outside_temperature).state);
      }
```

**Logs (if applicable):**

```
[C][ssd1306_i2c:023]: I2C SSD1306
[C][ssd1306_i2c:023]:   Rotations: 0 °
[C][ssd1306_i2c:023]:   Dimensions: 128px x 64px
[C][ssd1306_i2c:024]:   Address: 0x3C
[C][ssd1306_i2c:025]:   Model: SH1106 128x64
[C][ssd1306_i2c:027]:   External VCC: NO
[C][ssd1306_i2c:028]:   Update Interval: 1.0s
```

**Additional information and things you've tried:**

I used this example: https://esphome.io/cookbook/display_time_temp_oled.html
Simpler examples create the same issue.

This Arduino example worked as expected (no lines): https://paste.bombsite.be/iwoduneqob.cpp

ShonP40, 2021-01-06:
Full log:

```
[22:41:26][I][app:105]: ESPHome version 1.15.3 compiled on Dec 29 2020, 16:17:23
[22:41:26][C][wifi:443]: WiFi:
[22:41:26][C][wifi:303]:   SSID: [redacted]
[22:41:26][C][wifi:304]:   IP Address: 192.168.0.209
[22:41:26][C][wifi:306]:   BSSID: [redacted]
[22:41:26][C][wifi:307]:   Hostname: 'control_unit'
[22:41:26][C][wifi:311]:   Signal strength: -36 dB ▂▄▆█
[22:41:26][C][wifi:315]:   Channel: 1
[22:41:26][C][wifi:316]:   Subnet: 255.255.255.0
[22:41:26][C][wifi:317]:   Gateway: 192.168.0.1
[22:41:26][C][wifi:318]:   DNS1: 0.0.0.0
[22:41:26][C][wifi:319]:   DNS2: 0.0.0.0
[22:41:26][C][i2c:028]: I2C Bus:
[22:41:26][C][i2c:029]:   SDA Pin: GPIO1
[22:41:26][C][i2c:030]:   SCL Pin: GPIO3
[22:41:26][C][i2c:031]:   Frequency: 50000 Hz
[22:41:26][I][i2c:033]: Scanning i2c bus for active devices...
[22:41:26][I][i2c:040]: Found i2c device at address 0x3C
[22:41:26][C][template.switch:058]: Template Switch 'Backlight switch'
[22:41:26][C][template.switch:058]:   Assumed State: YES
[22:41:26][C][template.switch:059]:   Restore State: NO
[22:41:26][C][template.switch:060]:   Optimistic: YES
[22:41:26][C][switch.gpio:042]: GPIO Switch 'Flashlight'
[22:41:26][C][switch.gpio:042]:   Icon: 'mdi:flashlight'
[22:41:26][C][switch.gpio:043]:   Pin: GPIO4 (Mode: OUTPUT)
[22:41:26][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[22:41:26][C][logger:185]: Logger:
[22:41:26][C][logger:186]:   Level: DEBUG
[22:41:26][C][logger:187]:   Log Baud Rate: 115200
[22:41:26][C][logger:188]:   Hardware UART: UART0
[22:41:26][C][homeassistant.text_sensor:011]: Homeassistant Text Sensor 'alarm_clock_time'
[22:41:26][C][homeassistant.text_sensor:012]:   Entity ID: 'sensor.alarm_clock_time'
[22:41:26][C][homeassistant.text_sensor:011]: Homeassistant Text Sensor 'alarm_clock_status'
[22:41:27][C][homeassistant.text_sensor:012]:   Entity ID: 'input_boolean.alarm_clock_status'
[22:41:27][C][esp32_camera:043]: ESP32 Camera:
[22:41:27][C][esp32_camera:044]:   Name: Control Unit
[22:41:27][C][esp32_camera:045]:   Board Has PSRAM: YES
[22:41:27][C][esp32_camera:047]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[22:41:27][C][esp32_camera:048]:   VSYNC Pin: 25
[22:41:27][C][esp32_camera:049]:   HREF Pin: 23
[22:41:27][C][esp32_camera:050]:   Pixel Clock Pin: 22
[22:41:27][C][esp32_camera:051]:   External Clock: Pin:0 Frequency:20000000
[22:41:27][C][esp32_camera:052]:   I2C Pins: SDA:26 SCL:27
[22:41:27][C][esp32_camera:053]:   Reset Pin: -1
[22:41:27][C][esp32_camera:086]:   Resolution: 1600x1200 (UXGA)
[22:41:27][C][esp32_camera:099]:   JPEG Quality: 10
[22:41:27][C][esp32_camera:101]:   Contrast: 0
[22:41:27][C][esp32_camera:102]:   Brightness: 0
[22:41:27][C][esp32_camera:103]:   Saturation: 0
[22:41:27][C][esp32_camera:104]:   Vertical Flip: OFF
[22:41:27][C][esp32_camera:105]:   Horizontal Mirror: OFF
[22:41:27][C][esp32_camera:122]:   Test Pattern: NO
[22:41:27][C][homeassistant.time:010]: Home Assistant Time:
[22:41:27][C][homeassistant.time:011]:   Timezone: 'IST-2IDT-3,M3.4.5/2,M10.5.0/2'
[22:41:27][C][ssd1306_i2c:023]: I2C SSD1306
[22:41:27][C][ssd1306_i2c:023]:   Rotations: 0 °
[22:41:27][C][ssd1306_i2c:023]:   Dimensions: 128px x 64px
[22:41:27][C][ssd1306_i2c:024]:   Address: 0x3C
[22:41:27][C][ssd1306_i2c:025]:   Model: SH1106 128x64
[22:41:27][C][ssd1306_i2c:027]:   External VCC: NO
[22:41:27][C][ssd1306_i2c:028]:   Update Interval: 1.0s
[22:41:27][C][web_server:131]: Web Server:
[22:41:27][C][web_server:132]:   Address: 192.168.0.209:80
[22:41:27][C][ota:029]: Over-The-Air Updates:
[22:41:27][C][ota:030]:   Address: 192.168.0.209:3232
[22:41:27][C][api:095]: API Server:
[22:41:27][C][api:096]:   Address: 192.168.0.209:6053
[22:41:27][C][homeassistant.sensor:024]: Homeassistant Sensor 'inside_temperature'
[22:41:27][C][homeassistant.sensor:024]:   Unit of Measurement: ''
[22:41:27][C][homeassistant.sensor:024]:   Accuracy Decimals: 1
[22:41:27][C][homeassistant.sensor:025]:   Entity ID: 'sensor.temperature_sensor'
[22:41:27][C][homeassistant.sensor:024]: Homeassistant Sensor 'inside_humidity'
[22:41:27][C][homeassistant.sensor:024]:   Unit of Measurement: ''
[22:41:27][C][homeassistant.sensor:024]:   Accuracy Decimals: 1
[22:41:27][C][homeassistant.sensor:025]:   Entity ID: 'sensor.humidity_sensor'
```

ShonP40, 2021-01-06:
Also, since my last comment on this issue, I switched to the ESP32-CAM
Here's my new config:

```
esphome:
  name: control_unit
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "[redacted]"
  password: "[redacted]"
  manual_ip:
    static_ip: 192.168.0.209
    gateway: 192.168.0.1
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

web_server:
  port: 80

time:
  - platform: homeassistant
    id: esptime
    
sensor:
  - platform: homeassistant
    id: inside_temperature
    entity_id: sensor.temperature_sensor
    internal: true

  - platform: homeassistant
    id: inside_humidity
    entity_id: sensor.humidity_sensor
    internal: true

text_sensor:
  - platform: homeassistant
    id: alarm_clock_time
    entity_id: sensor.alarm_clock_time
    internal: true

  - platform: homeassistant
    id: alarm_clock_status
    entity_id: input_boolean.alarm_clock_status
    internal: true

i2c:
  sda: 1
  scl: 3
  id: bus_b

display:
  - platform: ssd1306_i2c
    model: "SH1106 128x64"
    id: "screen"
    i2c_id: bus_b
    address: 0x3C
    lambda: |-
      // Date
      it.strftime(64, 0, id(font1), TextAlign::TOP_CENTER, "%d-%m-%Y", id(esptime).now());

      // Clock
      it.strftime(0, 50, id(font2), TextAlign::BASELINE_LEFT, "%H:%M", id(esptime).now());

      // Temperature
      if (id(inside_temperature).has_state()) {
        it.printf(127, 23, id(font3), TextAlign::TOP_RIGHT , "%.1f°", id(inside_temperature).state);
      }

      // Humidity
      if (id(inside_humidity).has_state()) {
        it.printf(127, 60, id(font3), TextAlign::BASELINE_RIGHT , "%.1f%%", id(inside_humidity).state);
      }

      // Alarm clock
      if (id(alarm_clock_status).state == "on") {
        std::string val = id(alarm_clock_time).state;
        it.printf(0, 60, id(font4), TextAlign::BASELINE_LEFT, "Alarm %s", val.c_str());
      } else {
        it.printf(0, 60, id(font4), TextAlign::BASELINE_LEFT, "No Alarm");
      }

switch:
  - platform: template
    name: "Backlight switch"
    id: "backlight_switch"
    assumed_state: true
    optimistic: true
    turn_on_action:
      then:
        - lambda: |-
            id(screen).set_brightness(1);
            id(screen).fill(COLOR_ON);
    turn_off_action:
      then:
        - lambda: |-
            id(screen).set_brightness(0);
            id(screen).fill(COLOR_OFF);

  - platform: gpio
    pin: 4
    name: "Flashlight"
    icon: mdi:flashlight

font:
  - file: 'fonts/slkscr.ttf'
    id: font1
    size: 8

  - file: 'fonts/BebasNeue-Regular.ttf'
    id: font2
    size: 48

  - file: 'fonts/arial.ttf'
    id: font3
    size: 14

  - file: 'fonts/slkscr.ttf'
    id: font4
    size: 8
  
esp32_camera:
  name: Control Unit
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  resolution: 1600x1200
  jpeg_quality: 10
  vertical_flip: false
  horizontal_mirror: false
  max_framerate: 15Hz
```

mh255, 2021-01-18:
Hi @ShonP40,
i had the same problem with some of my SH1106 displays, reducing the britghtness solved the issue for me (as far as i can tell it is 100% per default).
```
...
display:
  - platform: ssd1306_i2c
    model: "SH1106 128x64"
    id: "screen"
    i2c_id: bus_b
    address: 0x3C
    brightness: 30%
    lambda: |-
...
```
Best wishes,
Michael :)

ShonP40, 2021-01-19:
Thank you!
But unfortunately I had to lower the screen brightness to 26% to get rid of all the lines.

This is more of a workaround than a solution, but the screen is still visible under the lighting conditions in my room, so I’ll close this issue for now.

closed 2021-01-19 by ShonP40