esphome/issues #1579: fastled E1.31 doesn't work with more than 340 leds

j2mc, 2020-10-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32

**ESPHome version (latest production, beta, dev branch)**
1.15.2

**Affected component:**
fastled_clockless

**Description of problem:**
I have a single output with 399 leds, the built in effects work with all 399 however E1.31 ignores all leds over 340, which is the third universe.  All the leds in the first and second universe work great, but I'm unable to control anything in the third.  Tested with both xLights and Vixen, both show the same results.

**Problem-relevant YAML-configuration entries:**
```yaml
e131:
  method: unicast

light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: GPIO4
    num_leds: 399
    rgb_order: GRB
    name: "Garage Roof"
    effects:
      - addressable_rainbow:
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - addressable_flicker:
      - e131:
          universe: 1
          channels: RGB

```

**Logs (if applicable):**

```
Forgot to copy the log, but when e131 is enabled it shows it attaching to universe 1, 2 and 3

```

**Additional information and things you've tried:**

docteurzoidberg, 2021-03-17:
same problems leds stops working at the 4th universe

docteurzoidberg, 2021-03-17:
maybe this line of code is the problem:

int output_end = std::min(it->size(), std::min(output_offset + get_lights_per_universe(), packet.count - 1));

(in components/e131/e131_addressable_light_effect.cpp)

3 working universes x 170 leds = 510. maximum packet size / max channels , there must be a mistake in the min()

logs on verbose shows: [23:01:53][V][e131_addressable_light_effect:060]: Applying data for 'e131' on 4 universe, for 510-510.

log code: ESP_LOGV(TAG, "Applying data for '%s' on %d universe, for %d-%d.", get_name().c_str(), universe, output_offset,
           output_end);

so seems we can't have more than 510 leds with e1.31 in esphome at the moment

docteurzoidberg, 2021-03-17:
i will post a pr. found the bug. the line is meant to be:

 int output_end = std::min(it->size(), std::min(output_offset + get_lights_per_universe(), output_offset + packet.count - 1));

(missing output offset)

docteurzoidberg, 2021-03-17:
result here with more than 510 leds:
https://youtu.be/_kfibNDFscw

closed 2021-03-19 by jesserockz