esphome/issues #1580: Xiaomi LYWSD02 BLE reporting battery, but there is no way to get data

sermayoral, 2020-10-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant Supervised 0.116.4 on Debian

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

1,15,3 (stable)

**Affected component:**

https://esphome.io/components/sensor/xiaomi_ble.html#lywsd02

**Description of problem:**
I have observed in the ESPHome log that this device is reporting battery level, but there is no information about it in the documentation.

**Problem-relevant YAML-configuration entries:**
```yaml
# Enable Bluetooth scanning for this ESP32
esp32_ble_tracker:

# Especifica los sensores de temperatura a escanear
sensor:
  - platform: xiaomi_lywsd02
    mac_address: 3F:59:C8:63:38:99
    temperature:
      name: "Temperatura 4"
    humidity:
      name: "Humedad 4"
```

**Logs (if applicable):**

```
[10:59:19][D][xiaomi_ble:300]: Got Xiaomi LYWSD02 (3F:59:C8:63:38:99):
[10:59:19][D][xiaomi_ble:306]:   Humidity: 70.0%
[10:59:19][D][sensor:092]: 'Humedad 4': Sending state 70.00000 % with 1 decimals of accuracy
[10:59:33][D][xiaomi_ble:300]: Got Xiaomi LYWSD02 (3F:59:C8:63:38:99):
[10:59:33][D][xiaomi_ble:309]:   Battery Level: 26%
```

**Additional information and things you've tried:**

rradar, 2020-10-25:
Did you try to add something like? Maybe it's just missing in the documentation:
```
    battery_level:
      name: "LYWSD02 Battery Level"
```

sermayoral, 2020-10-25:
@rradar no, i didn't. I will try it! If it works, i will prepare a PR to the doc.

sermayoral, 2020-10-25:
It does not work

[image]

Alex9779, 2020-10-26:
I think I can add that option to that sensor tonight but I can't test it... @sermayoral are you able to test a fix before it is in master or dev?

sermayoral, 2020-10-26:
> 
> 
> I think I can add that option to that sensor tonight but I can't test it... @sermayoral are you able to test a fix before it is in master or dev?

Yes, I can

sermayoral, 2020-10-26:
Great. I looks good! If I can help you, tell me what should I do.

I am going to prepare the documentation.

Alex9779, 2020-10-26:
Good...

> If I can help you, tell me what should I do.
Not really, it compiles and it works for you, so I can create the PR as soon as you have the docs ready, but that's also not a very hard change...
So just tell me, then I create my PR, then you yours for the docs, and then we can cross link them by mentioning them in each description...

sermayoral, 2020-10-26:
Docs is ready in https://github.com/sermayoral/esphome-docs/tree/lywsd02_battery

:-)

Alex9779, 2020-10-26:
I just copied it too from CGG1 😃

closed 2020-10-26 by jesserockz