esphome/issues #1584: Mitsubishi AC not working with ESPHome

mattqm, 2020-10-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

Generic ESP8266 12F

**ESPHome version (latest production, beta, dev branch)**

1.15.3

**Affected component:**

https://esphome.io/components/climate/ir_climate.html

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
remote_transmitter:
  pin: GPIO14
  carrier_duty_percent: 50%

climate:
  - platform: mitsubishi
    name: "Living Room AC"

```

**Logs (if applicable):**

```
[19:39:38][D][remote_transmitter:066]: Sending remote code...
[19:39:38][D][climate:262]: 'Living Room AC' - Sending state:
[19:39:38][D][climate:265]:   Mode: COOL
[19:39:38][D][climate:282]:   Target Temperature: 21.00°C
[19:39:50][D][climate:010]: 'Living Room AC' - Setting
[19:39:50][D][climate:014]:   Mode: HEAT

```

**Additional information and things you've tried:**

I have tested my hardware out with my own Arduino code using the HeatpumpIR library and it's all working as expected - i.e. the IR codes turn the unit on/off and set the temp, etc...  But with ESPHome I can see the command being issued in the log, but nothing happens.

My key bit of Arduino code that works looks like this:
```
IRSenderBitBang irSender(14);
HeatpumpIR *heatpumpIR = new MitsubishiFDHeatpumpIR();
...
   heatpumpIR->send(irSender, power, mode, fan, temp, vdir, hdir);
```

I've tried to compare the heatpumpIR code to the ESPHome approach  - they go about it quite differently, but as far as I can tell they seem to do the same thing.

Is there any extra debugging I can turn on to try to work out what is not working.

As an aside - maybe it is worth wrapping HeatpumpIR in ESPHome as it supports many different units out of the box.

rob-deutsch, 2020-10-28:
> I've tried to compare the heatpumpIR code to the ESPHome approach - they go about it quite differently, but as far as I can tell they seem to do the same thing.

You seem to be right. heatpumpIR supports many different Mitsubishi aircons, but `MitsubishiFDHeatpumpIR` is the same as ESPHome's `mitsubishi`.

I don't know why it isn't working for you.

> As an aside - maybe it is worth wrapping HeatpumpIR in ESPHome as it supports many different units out of the box.

I am actually working on this now. I have implemented a `mitsubishi_heavy` (Mitsubishi Heavy Industries) component that uses heatpumpIR to generate the codes. I just need some feedback/input/direction on the most desirable way to generalise this approach and support the dozens of heatpumps that heatpumpIR supports.

https://github.com/rob-deutsch/esphome/tree/mitsubishi_heavy/esphome/components/mitsubishi_heavy

mattqm, 2020-10-28:
I still haven't quite got my head around how ESPHome constructs the C++, but could you do something like:

User selects A/C platform, e.g. Hitachi, Hisense, MitsubishiFD, MitsubishiHeavy, etc and the GPIO pin.  Then you include the corresponding header file and object declaration, e.g.: 

`#include <MitsubishiHeatpumpIR.h>
IRSenderBitBang irSender(14);
HeatpumpIR *heatpumpIR = new MitsubishiFDHeatpumpIR();`

Having said that it highlights a difference in approach.  The current implementation is using 'remote_transmitter' whereas HeatpumpIR uses the GPIO directly.  Makes me think I should try out 100% duty cycle on the current implementation - nope just tried that - no difference.  I can see the IR is working, but I haven't got a rig to compare the signals generated.  I moved the LED right up to the unit, still no response.

rob-deutsch, 2020-10-28:
I was thinking about it some more - and I have an idea of something you can investigate.

I noticed you are using an ESP8266. It's possible that GPIO14 doesn't support this particular function.

You are using `IRSenderBitBang` in the pure Arduino code. This would work on just about any pin. I think that ESPHome uses timers. I wonder if arduino-heatpumpir's code would actually work if you used the `IRSenderESP8266` which also uses timers.

According to to the [arduino-heatpumpir README.md](arduino-heatpumpir README.md):

> Note that depending on your board, certain GPIO's might not be available. For example on NodeMCU, the usable GPIO's are D1 (GPIO5), D2 (GPIO4), D6 (GPIO12), D7 (GPIO13) and D3 (GPIO0).

I would try your ESPHome code on GPIO5/4/12/13/0. And also see if arduino-heatpumpir's code works using `IRSenderESP8266 irSender(14)`, because I suspect maybe it doesn't?

mattqm, 2020-10-29:
@rob-deutsch thanks for your help.  I've moved to GPIO12 and it's working now.  Tested on both a NodeCPU and raw ESP8266.  I think I was using BitBang since that's what they use for ESP8266 in their simple example and didn't think to question it.

rob-deutsch, 2020-10-29:
Fantastic, I'm glad it worked!

Can I suggest closing this issue?

mattqm, 2020-10-29:
It would be nice to have this documented somewhere to save someone else some grief in the future.  Do we expect this pin behaviour would apply to all IR transmitter use and climate platforms or just mitsubishi?  If it is just mitsubishi we can probably leave it as a fringe case.  I do note that one of the examples on the climate page uses GPIO14.

lightzhuk, 2020-11-21:
Good afternoon. I have a similar problem. I am using an Avatto S06 flashed IR remote control
example of my code (
`status_led:
  pin: GPIO4

remote_receiver:
  pin:
    number: GPIO5
    inverted: true
  dump: all

remote_transmitter:
  pin:
    number: GPIO14
    inverted: false
  carrier_duty_percent: 50%
  id: "s06_transmitter"

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO13
      inverted: true
    internal: true
    name: "s06_button"
    id: "s06_button"
    
climate:
  - platform: mitsubishi
    name: "s06_mitsubishi_ac"
    transmitter_id: "s06_transmitter"
    id: "s06_mitsubishi_ac"
    visual:
      min_temperature: 18 ° C
      max_temperature: 30 ° C
      temperature_step: 1 ° C
      
switch:
  - platform: template
    name: Raw Code Power Button ON
    turn_on_action:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [5973, -7465, 584, -3356, 583, -3380, 560, -3380, 560, -3354, 586,
                -3379, 561, -1457, 561, -1458, 560, -3389, 559, -1484, 534, -1457,
                 561, -1458, 559, -1458, 558, -1459, 560, -3358, 582, -3381, 559,
                 -1465, 560, -1436, 582, -3380, 560, -3380, 561, -1458, 559, -1483,
                 535, -3380, 560, -1460, 557, -1492, 533, -3354, 587, -1458, 560,
                 -1483, 534, -3356, 584, -3381, 560, -1460, 556, -3381, 560, -3391,
                 557, -1459, 558, -3379, 562, -1459, 558, -3380, 560, -1460, 558,
                 -3380, 561, -1483, 534, -1492, 534, -3379, 561, -1483, 534, -3356,
                 584, -1484, 533, -3381, 560, -1460, 558, -3382, 558, -3385, 558, -7455, 587]
                 
  - platform: template
    name: Raw Code Power Button Off
    turn_on_action:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [5977, -7466, 583, -3383, 557, -3383, 557, -3381, 558, -3385, 556,
                -3384, 556, -1461, 556, -1487, 531, -3391, 557, -1461, 556, -1460,
                558, -1461, 557, -1461, 557, -1460, 558, -3381, 559, -3381, 559,
                -1469, 557, -1458, 559, -3379, 562, -3358, 582, -3381, 559, -1460,
                558, -3379, 560, -1459, 559, -1466, 559, -3381, 559, -1483, 536,
                -1457, 561, -1457, 560, -3380, 560, -1458, 560, -3356, 584, -3387,
                560, -1482, 537, -3379, 560, -1484, 534, -3381, 559, -1460, 559,
                -3378, 560, -1484, 534, -1467, 559, -3380, 559, -1460, 558, -3380,
                560, -1484, 535, -3382, 557, -1484, 534, -3380, 560, -3360, 583, -7457, 609] `)
I only managed to record the command from the old remote control and then reproduce this command from the flashed remote control.
None of the components have stuck. Do I understand correctly that the mitsubishi_heavy platform may appear in the future?
Thank you very much for your work.

closed 2021-03-21 by mattqm

Ranchuking, 2021-04-16:
Hello, I also use esp8266f.
My code is:
`remote_transmitter:
  - id: ir_mitsubishi
    pin: GPIO04
    carrier_duty_percent: 50%

climate:
  - platform: mitsubishi
    name: "Living Room AC"
    transmitter_id: ir_mitsubishi
    supports_cool: True
    supports_heat: False
    visual:
      min_temperature: 22 °C
      max_temperature: 30 °C
      temperature_step: 1 °C`

I can turn on the AC and change the temperature but can't power off the AC.